arendst/Tasmota #2980: Error While Compiling

Caswell1000, 2018-06-13:
Hi,

I had problems with the previous version when flashing, so I uploaded all the latest versions and followed in the Wiki, I have this error:

```
Arduino: 1.8.5 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, nodemcu, 26 MHz, 40MHz, DOUT, 1M (no SPIFFS), 2, v1.4 Higher Bandwidth, Disabled, None, Only Sketch, 512000"

In file included from c:\users\caswe\downloads\arduino\arduino-1.8.5-windows\arduino-1.8.5\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\algorithm:60:0,

                 from C:\Users\caswe\Downloads\arduino\arduino-1.8.5-windows\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/Arduino.h:254,

                 from sketch\sonoff.ino.cpp:1:

c:\users\caswe\downloads\arduino\arduino-1.8.5-windows\arduino-1.8.5\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\utility:68:28: fatal error: bits/c++config.h: No such file or directory

 #include <bits/c++config.h>

                            ^

compilation terminated.

exit status 1
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [n/a] Hardware used : 
- [n/a] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

ascillato, 2018-06-13:
the file _sonoff.ino.cpp_ does not exist in tasmota.

Please delete your folder, download again Tasmota, and try again

ascillato, 2018-06-13:
I'm closing it. Please, reopen if your issue still persist. Thanks :+1:

Caswell1000, 2018-06-13:
Hi,

The Error does persist, below is a more verbose compile output: 
```
Arduino: 1.8.5 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, nodemcu, 26 MHz, 40MHz, DOUT, 1M (no SPIFFS), 2, v1.4 Higher Bandwidth, Disabled, None, Only Sketch, 512000"

C:\Users\caswe\Downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\arduino-builder -dump-prefs -logger=machine -hardware C:\Users\caswe\Downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\hardware -hardware C:\Users\caswe\Downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\portable\packages -tools C:\Users\caswe\Downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\tools-builder -tools C:\Users\caswe\Downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\hardware\tools\avr -tools C:\Users\caswe\Downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\portable\packages -built-in-libraries C:\Users\caswe\Downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\libraries -libraries C:\Users\caswe\Downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\portable\sketchbook\libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,FlashSize=1M0,led=2,LwIPVariant=Prebuilt,Debug=Disabled,DebugLevel=None____,FlashErase=none,UploadSpeed=512000 -ide-version=10805 -build-path C:\Users\caswe\AppData\Local\Temp\arduino_build_263242 -warnings=none -build-cache C:\Users\caswe\AppData\Local\Temp\arduino_cache_322354 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\caswe\Downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=C:\Users\caswe\Downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\portable\packages\esp8266\tools\esptool\0.4.13 -prefs=runtime.tools.mkspiffs.path=C:\Users\caswe\Downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\portable\packages\esp8266\tools\mkspiffs\0.2.0 -verbose C:\Users\caswe\Downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\portable\sketchbook\sonoff\sonoff.ino
C:\Users\caswe\Downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\arduino-builder -compile -logger=machine -hardware C:\Users\caswe\Downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\hardware -hardware C:\Users\caswe\Downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\portable\packages -tools C:\Users\caswe\Downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\tools-builder -tools C:\Users\caswe\Downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\hardware\tools\avr -tools C:\Users\caswe\Downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\portable\packages -built-in-libraries C:\Users\caswe\Downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\libraries -libraries C:\Users\caswe\Downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\portable\sketchbook\libraries -fqbn=esp8266:esp8266:generic:CpuFrequency=80,ResetMethod=nodemcu,CrystalFreq=26,FlashFreq=40,FlashMode=dout,FlashSize=1M0,led=2,LwIPVariant=Prebuilt,Debug=Disabled,DebugLevel=None____,FlashErase=none,UploadSpeed=512000 -ide-version=10805 -build-path C:\Users\caswe\AppData\Local\Temp\arduino_build_263242 -warnings=none -build-cache C:\Users\caswe\AppData\Local\Temp\arduino_cache_322354 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\caswe\Downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=C:\Users\caswe\Downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\portable\packages\esp8266\tools\esptool\0.4.13 -prefs=runtime.tools.mkspiffs.path=C:\Users\caswe\Downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\portable\packages\esp8266\tools\mkspiffs\0.2.0 -verbose C:\Users\caswe\Downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\portable\sketchbook\sonoff\sonoff.ino
Using board 'generic' from platform in folder: C:\Users\caswe\Downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.1
Using core 'esp8266' from platform in folder: C:\Users\caswe\Downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.1
Detecting libraries used...
"C:\Users\caswe\Downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\caswe\Downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/include" "-IC:\Users\caswe\Downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/lwip/include" "-IC:\Users\caswe\Downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.1/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\Users\caswe\AppData\Local\Temp\arduino_build_263242/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC   -DARDUINO=10805 -DARDUINO_ESP8266_GENERIC -DARDUINO_ARCH_ESP8266  [... cut ...]
                 from C:\Users\caswe\AppData\Local\Temp\arduino_build_263242\sketch\sonoff.ino.cpp:1:

c:\users\caswe\downloads\arduido download\arduino-1.8.5-windows\arduino-1.8.5\portable\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\utility:68:28: fatal error: bits/c++config.h: No such file or directory

 #include <bits/c++config.h>

                            ^

compilation terminated.

exit status 1
Error compiling for board Generic ESP8266 Module.
```

andrethomas, 2018-06-13:
Are you running arduino.exe from inside the portable folder and then opening the sketch?

I have a pre-configured portable version available at https://thehackbox.org/downloads/arduino_esp8266_tasmota.7z if you want to use that as a quick start.

Extract and run the arduino.exe from inside that folder and open the sonoff sketch from within arduino - it is located under portable/sketchbook/

ascillato, 2018-06-14:
Hi, If the issue persist just reopen this issue

ascillato, 2018-06-14:
Seems to be an arduino issue.

Have you tried the @andrethomas solution?

Have you tried to uninstall and install it again?

ascillato, 2018-06-14:
Hi, seems that your arduino is not finding the libraries. If you already install everything needed, please try to change the name of your folder **"arduido download"** to something without spaces in it. Sometime spaces make problems

Caswell1000, 2018-06-14:
The solution by @andrethomas works. Thank You......

closed 2018-06-14 by Caswell1000