arendst/Tasmota #2981: Toggle count and MQTT

pakgio, 2018-06-13:
hello. 
I'm a real noob. I'm using tasmota on sonoff and they are working ok with homeassistant.

I need to toggle an existing relay using the sonoff (it gives a pulse to the relay that toggle its status on/off).
Do you think is possible to comunicate to HA trough MQTT the count of pulse?
(the idea is to associate an ON status to a even number of pulse and a OFF status to a odd number of pulse)
How to do that?

Frogmore42, 2018-06-13:
This isn't supposed to be a forum.  This is supposed to be a place where issues with the Tasmota software are reported and resolved.  Your question is more appropriate for one of the many forums about Sonoff devices and or esp8266.

pakgio, 2018-06-13:
well ok. sorry.

ascillato, 2018-06-13:
If you want to count something use rules and commands VAR and publish. Please see wiki. Thanks

ascillato, 2018-06-13:
I'm closing it. Please, reopen if your find any issue related to this but in that case please provide all the information of the troubleshooting template. Thanks +1

closed 2018-06-13 by ascillato2