arendst/Tasmota #2983: PZEM004T

tcpipchip, 2018-06-13:
Hi

I see a define to
#define USE_PZEM004T (serial)

But where in software i enable it to use and get the data and publish on MQTT ? And in wich SONOFF MODEL i select ?
#define MODULE                 SONOFF_XXXX

Thank you again!

Miguel

Frogmore42, 2018-06-13:
Looking at the code, you have to enable the code with the #define and you have to select the pins that will be RX and TX and then hook it up correctly.  I am sure a quick internet search will show you lots of examples of that.
In Tasmota, it is easy to use the web ui to set the module type and the definitions of the pins (tasmota will remember these even if you flash new fw).  The module you select should match what you have.

Since you provided none of the required information and apparently didn't follow the instructions in the template, it is impossible to provide any more help, without making wild guesses.

vMeph, 2018-06-13:
[image]

tcpipchip, 2018-06-13:
Thank you so much!!!!!!

closed 2018-06-13 by tcpipchip