arendst/Tasmota #2984: Best way to implement SPI drivers - specifically because they need CS pins

andrethomas, 2018-06-13:
**Have you look for this feature in other issues and in the wiki?**
Yes

**Is your feature request related to a problem? Please describe.**
Support for SPI drivers is hinted at since there is already a #ifdef USE_SPI in the code but I'm not clear on how to proceed with CS implementation.

**Describe the solution you'd like**
Description from lead developers on best practice route to follow for implementing SPI drivers.

**Describe alternatives you've considered**
Adding additional outputs via I2C using something like MCP23008/MCP23017 to make provision for additional IO pins, or making it possible to use existing GPIO to use CS select but then one would have to find a clean way of implementing this also... and yes, I know adding expansion using the above mentioned chips have been discussed and frowned upon but we need to find a way to expand GPIO in an efficient and low-cost way.

**Additional context**
The only solution I currently have for sensors is to run the SPI code on a pro mini 3.3V and I wrote an I2C driver which pulls the values from the pro mini 328 as a JSON object and pushes it out over mqqt, or to the web interface I guess (although I have not needed to do the last part yet)

Jason2866, 2018-06-13:
I am not a programmer.... Maybe this helps you 
@arendst released the branch Display. With this branch a TFT SPI  Display can be connected.

andrethomas, 2018-06-13:
Excellent find, thanks... will follow that one :)

closed 2018-06-13 by andrethomas