arendst/Tasmota #2985: Confused about latest available binary

matrix200, 2018-06-13:
If I understand correctly, several releases have been done since 5.1.4
Yet the OTA and manual check shows that the latest binaries are still 5.1.4
Is there going to be a set of binaries for 6.0 released at some point?

Also I think its wrong, that OTA invoked by using GUI or console, doesn't compare the latest available binary to the currently installed version.
Because of that my device just flashed over the same version which surely isn't healthy for the flash endurance, not to mention caused two power cycles which were totally unnecessary.

reloxx13, 2018-06-14:
Dev build is at 6.0.0 and not released now.

latest release:: https://github.com/arendst/Sonoff-Tasmota/releases/latest

matrix200, 2018-06-14:
Thanks, I think I will wait until Theo Arends decides it is stable enough and builds the binaries.
Prefer not to run code that hasn't been fully tested on the main switch connecting all my computers to the mains.

closed 2018-06-14 by matrix200