arendst/Tasmota #2987: interlock bug or intended behaviour?

biotecov, 2018-06-13:
Using tasmota 6.0.0a on electrodragon and on a d1-mini with two relays attached to control a roller shutter. I am trying the interlock mode (setoption14 on), a nice feature.

When I use power1 on, relay2 switches off if previously on, and conversely power2 on switches off relay1. So far so good, it is the intended behaviour as explained in the docs.

However, with relay1 already off, and relay2 on, the command power1 off relaxes relay2 (as a result both relays get off). The other way is the same (power2 off relaxes relay1). In other words, turning off any relay already off results in all relays being turned off.

Is this a bug or a feature? If it is a feature, it would be nice to have it documented shomehow.

Regards.

Frogmore42, 2018-06-14:
Seems like it is by design.  Interlock is to prevent both motors from being ON at the same time.  Having the ability to have both OFF is a good thing, otherwise you would have to depend on limit switches to stop the motors and you would not be able to have the shutters at anything other than completely closed or open.

biotecov, 2018-06-14:
Thanks for your comment. Agreed, but that's not the point I wanted to rise. Consider that relay1 is off and relay2 is on, then to have both off, issuing power2 off would do the job. The surprise came when I saw that power1 off (remark that relay1 is already off) also turns relay2 off. This is probably the most convenient behaviour for shutters since any powerX off will turn everything off, regardless of wich was on, so I am happy with it once aware, but the point is that if this is by design, it is expected to be stable, but if it is just by chance/error, it could change at any time and mess up previous setups. A confirmation from the author that this is the intended behaviour would be useful.

ascillato, 2018-06-21:
Hi

It is by design. Only one relay can turned on and the rest will be off. And also all can be off. 

If this solves your question, please close this issue. Thanks :+1:

closed 2018-06-27 by ascillato2