arendst/Tasmota #2988: DHT22 not working when Restart/Reset... OK at power up

klingon888, 2018-06-13:
Wemos D1 mini flashed with Tasmota Version 5.14.0-2_4_1, I have a DHT22 configured as Generic module connected to D4 GPIO2 : 02 AM2301 as per the Wiki. It works great at power up but whenever I do a Restart on the web portal or pressing the Reset button on the Wemos, I no longer see the DHT22 on the web portal.  If I powercycle the wemos or disconnect & reconnect just the +ve/-ve to the DHT22, it works again.  Is this a bug or something I missed?

klingon888, 2018-06-13:
Thought it may be the internal pull-up resistor on GPIO2. Then tried with GPIO0 (also with pull-up) and GPIO5 and it works fine, even after Restart. Only problem when connected to GPIO2 as per wiki.  Maybe its just my Wemos board. Closing issue but keeping comment in case someone else come across some problem.

closed 2018-06-13 by klingon888

JollyJames, 2018-08-19:
I am also finding that GPIO2 in 6.1.1c shows the same behaviour, power on = working, restart = nan for both temp and humidity. Using identical AM2302 on GPIO2 and GPIO4 configuring with AM2301 and SI7021 shows the same behaviour.. removing the sensor and then adding it whilst powered also prompts correct operation. Web restart always fails GPIO2 connected sensor. This should work, feels like timing issue or lock-up with sensor from restart... 100% repeatable.

salas81, 2018-10-07:
I have the same problem. DHT is not working at boot up, then I remove it and re-plug it and starts working. Any hint?

hofmeisterchrigu, 2019-02-06:
I also recognized that problem. The Temp. and Hum. information is not available, even after a reset by reset-button on Wemos. But it works after powering off/on the device.

Quite strange but ok for me.