arendst/Tasmota #2989: Json Issues not allowing compiling

Music4lity, 2018-06-14:
I'm not much of a coder, I'm just trying to make this work. This issue is keeping the code from compiling in the Arduino IDE. I've tried this on two different computers and both give this output:

`C:\Users\sebas\Desktop\Sonoff-Tasmota-5.9.0\sonoff\xdrv_domoticz.ino: In function 'boolean DomoticzMqttData(char*, uint16_t, char*, uint16_t)':

xdrv_domoticz:141: error: 'StaticJsonBuffer' was not declared in this scope

     StaticJsonBuffer<400> jsonBuf;

     ^

C:\Users\sebas\Desktop\Sonoff-Tasmota-5.9.0\sonoff\xdrv_domoticz.ino:141:5: note: suggested alternative:

In file included from C:\Program Files (x86)\Arduino\portable\sketchbook\libraries\ArduinoJson\src/ArduinoJson/StaticJsonDocument.hpp:8:0,

                 from C:\Program Files (x86)\Arduino\portable\sketchbook\libraries\ArduinoJson\src/ArduinoJson.hpp:15,

                 from C:\Program Files (x86)\Arduino\portable\sketchbook\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from C:\Users\sebas\Desktop\Sonoff-Tasmota-5.9.0\sonoff\sonoff.ino:49:

C:\Program Files (x86)\Arduino\portable\sketchbook\libraries\ArduinoJson\src/ArduinoJson/Memory/StaticJsonBuffer.hpp:110:7: note:   'ArduinoJson::Internals::StaticJsonBuffer'

 class StaticJsonBuffer : public Internals::StaticJsonBufferBase {

       ^

xdrv_domoticz:141: error: 'jsonBuf' was not declared in this scope

     StaticJsonBuffer<400> jsonBuf;

                           ^

C:\Users\sebas\Desktop\Sonoff-Tasmota-5.9.0\sonoff\xdrv_ir_send.ino: In function 'boolean IrSendCommand(char*, uint16_t, char*, uint16_t, int16_t)':

xdrv_ir_send:294: error: 'StaticJsonBuffer' was not declared in this scope

       StaticJsonBuffer<128> jsonBuf;

       ^

C:\Users\sebas\Desktop\Sonoff-Tasmota-5.9.0\sonoff\xdrv_ir_send.ino:294:7: note: suggested alternative:

In file included from C:\Program Files (x86)\Arduino\portable\sketchbook\libraries\ArduinoJson\src/ArduinoJson/StaticJsonDocument.hpp:8:0,

                 from C:\Program Files (x86)\Arduino\portable\sketchbook\libraries\ArduinoJson\src/ArduinoJson.hpp:15,

                 from C:\Program Files (x86)\Arduino\portable\sketchbook\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from C:\Users\sebas\Desktop\Sonoff-Tasmota-5.9.0\sonoff\sonoff.ino:49:

C:\Program Files (x86)\Arduino\portable\sketchbook\libraries\ArduinoJson\src/ArduinoJson/Memory/StaticJsonBuffer.hpp:110:7: note:   'ArduinoJson::Internals::StaticJsonBuffer'

 class StaticJsonBuffer : public Internals::StaticJsonBufferBase {

       ^

xdrv_ir_send:294: error: 'jsonBuf' was not declared in this scope

       StaticJsonBuffer<128> jsonBuf;

                             ^

C:\Users\sebas\Desktop\Sonoff-Tasmota-5.9.0\sonoff\xdrv_wemohue.ino: In function 'void HueLights(String*)':

xdrv_wemohue:554: error: 'StaticJsonBuffer' was not declared in this scope

       StaticJsonBuffer<400> jsonBuffer;

       ^

C:\Users\sebas\Desktop\Sonoff-Tasmota-5.9.0\sonoff\xdrv_wemohue.ino:554:7: note: suggested alternative:

In file included from C:\Program Files (x86)\Arduino\portable\sketchbook\libraries\ArduinoJson\src/ArduinoJson/StaticJsonDocument.hpp:8:0,

                 from C:\Program Files (x86)\Arduino\portable\sketchbook\libraries\ArduinoJson\src/ArduinoJson.hpp:15,

                 from C:\Program Files (x86)\Arduino\portable\sketchbook\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from C:\Users\sebas\Desktop\Sonoff-Tasmota-5.9.0\sonoff\sonoff.ino:49:

C:\Program Files (x86)\Arduino\portable\sketchbook\libraries\ArduinoJson\src/ArduinoJson/Memory/StaticJsonBuffer.hpp:110:7: note:   'ArduinoJson::Internals::StaticJsonBuffer'

 class StaticJsonBuffer : public Internals::StaticJsonBufferBase {

       ^

xdrv_wemohue:554: error: 'jsonBuffer' was not declared in this scope

       StaticJsonBuffer<400> jsonBuffer;

                             ^

exit status 1
'StaticJsonBuffer' was not declared in this scope`

Music4lity, 2018-06-14:
Fixed by downgrading form the Beta 6.0.1 version of ArduinoJson to 5.13.2

closed 2018-06-14 by Music4lity

aemtcprojects, 2019-05-06:
Music4lity many thanks for that. Who would've thought this through? Appreciated!