arendst/Tasmota #2990: Sonoff Basic randomly turning off after short periods

mrjemson, 2018-06-14:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x] Hardware used : Sonoff Basic
- [x] Provide the output of command ``status 0`` :
```
05:49:35 RSL: Received Topic /status, Data Size 1, Data 0
05:49:35 RSL: Group 0, Index 1, Command STATUS, Data 0
05:49:35 MQT: stat/sonoff-printlight/STATUS = {"Status":{"Module":1,"FriendlyName":"Sonoff-PrintLight","Topic":"sonoff-printlight","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
05:49:35 MQT: stat/sonoff-printlight/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://10.80.75.10:80/sonoff.ino.bin","Uptime":598,"Sleep":0,"BootCount":86,"SaveCount":435,"SaveAddress":"F9000"}}
05:49:35 MQT: stat/sonoff-printlight/STATUS2 = {"StatusFWR":{"Version":"5.10.0","BuildDateTime":"2018-03-08T22:00:52","Boot":6,"Core":"2_4_0_RC2","SDK":"2.1.0(7106d38)"}}
05:49:35 MQT: stat/sonoff-printlight/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":4,"SysLog":0,"LogHost":"10.80.75.11","LogPort":514,"SSId1":"HomeCtrl","SSId2":"Jemson-AP","TelePeriod":300,"SetOption":"00000009"}}
05:49:35 MQT: stat/sonoff-printlight/STATUS4 = {"StatusMEM":{"ProgramSize":510,"Free":428,"Heap":17,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
05:49:35 MQT: stat/sonoff-printlight/STATUS5 = {"StatusNET":{"Hostname":"sonoff-printlight-7175","IPAddress":"192.168.137.103","Gateway":"192.168.137.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.137.1","Mac":"68:C6:3A:A3:BC:07","Webserver":2,"WifiConfig":3}}
05:49:35 MQT: stat/sonoff-printlight/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.137.25","MqttPort":1883,"MqttClientMask":"DVES_X","MqttClient":"DVES_A3BC07","MqttUser":"HomeCtrl","MAX_PACKET_SIZE":512,"KEEPALIVE":15}}
05:49:35 MQT: stat/sonoff-printlight/STATUS7 = {"StatusTIM":{"UTC":"Thu Jun 14 04:49:35 2018","Local":"Thu Jun 14 05:49:35 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1}}
05:49:35 MQT: stat/sonoff-printlight/STATUS10 = {"StatusSNS":{"Time":"2018-06-14T05:49:35","Switch1":"ON"}}
05:49:35 MQT: stat/sonoff-printlight/STATUS11 = {"StatusSTS":{"Time":"2018-06-14T05:49:35","Uptime":598,"Vcc":3.486,"POWER":"ON","Wifi":{"AP":2,"SSId":"Jemson-AP","RSSI":100,"APMac":"80:2A:A8:54:5C:FA"}}}
```

Running Tasmota 5.10.0 and the device randomly turns off without any indication in the console as to why even with debug set to 4.
Time it stays on is completely random sometimes it can be 15-20 seconds other times it might be 15-20 minutes but never seems to stay on for longer than that.

See below console output, I have omitted 5mins worth of "checking connection" output in the middle:

`05:51:13 WIF: Checking connection...
05:51:13 WIF: Connected
05:51:22 MQT: tele/sonoff-printlight/STATE = {"Time":"2018-06-14T05:51:22","Uptime":598,"Vcc":3.482,"POWER":"ON","Wifi":{"AP":2,"SSId":"Jemson-AP","RSSI":100,"APMac":"80:2A:A8:54:5C:FA"}}
05:51:22 MQT: tele/sonoff-printlight/SENSOR = {"Time":"2018-06-14T05:51:22","Switch1":"ON"}
05:51:33 WIF: Checking connection...
05:51:33 WIF: Connected
...
05:55:33 WIF: Checking connection...
05:55:33 WIF: Connected
05:55:53 WIF: Checking connection...
05:55:53 WIF: Connected
05:56:12 MQT: stat/sonoff-printlight/RESULT = {"POWER":"OFF"}
05:56:12 MQT: stat/sonoff-printlight/POWER = OFF
05:56:12 CFG: Saved to flash at F8, Count 436, Bytes 1648
05:56:13 WIF: Checking connection...
05:56:13 WIF: Connected
05:56:22 MQT: tele/sonoff-printlight/STATE = {"Time":"2018-06-14T05:56:22","Uptime":598,"Vcc":3.484,"POWER":"OFF","Wifi":{"AP":2,"SSId":"Jemson-AP","RSSI":100,"APMac":"80:2A:A8:54:5C:FA"}}
05:56:22 MQT: tele/sonoff-printlight/SENSOR = {"Time":"2018-06-14T05:56:22","Switch1":"ON"}
05:56:33 WIF: Checking connection...
05:56:33 WIF: Connected`

Can't find any similar reports and device doesn't seem to lose connectivity or anything so don't believe the unit to be faulty. Out of ideas as to what else the issue could be.

Frogmore42, 2018-06-14:
Well, if it is checking connection and connecting repeatedly for an extended period of time, you need to fix that first.

You are also using a now very old version and should update to the newest release.

mrjemson, 2018-06-14:
Thanks for your 2 cents, but if you had seen a working device in operation at debug level 4 you would know it is not reconnecting repeatedly but rather checking and confirming the connection is still present per normal operation.

I didn't realise a build under 6 months old was considered "very old" and required an upgrade before assistance in troubleshooting would be provided but if given no option other that is what I will need to do.

Frogmore42, 2018-06-14:
You are running core 2.4.0RC2, which is old and known to be not good.  You also have relatively low heap, which is a known issue with 2.4.0.

So, yeah, what you are running is old.  Since you are having a problem, the first thing to do is update the firmware.  If it was working fine, there would be no need to update, unless you wanted some new feature that was implemented after you last updated.

Newer versions also provide more detail on commands, which is the issue you are seeing.

Frogmore42, 2018-06-14:
Core 2.4.0 is newer than 2.3.0.  Some people/devices have had issues with 2.4.0 and 2.4.1, but others have been fine.
https://github.com/arendst/Sonoff-Tasmota/wiki/Theo's-Tasmota-Tips#20180103---tasmota-is-ready-for-esp8266arduino-version-240
Look at Theo's tips to see his notes on the various versions of the core and the issues each of them has.

Most people who have had issues with newer stuff that have tried core 2.3.0 have seen their issue resolved by the older version.

mrjemson, 2018-06-14:
No change in my issue with 5.14.0 and core 2.3.0.
No change in logging output either.

````
00:21:09 LOG: SerialLog 2, WebLog 4, SysLog 0, LogHost 10.80.75.11, LogPort 514, TelePeriod 300
00:21:09 HTP: Configuration
00:21:09 CFG: Saved to flash at FA, Count 510, Bytes 2560
00:21:11 HTP: Main Menu
00:21:12 MQT: stat/sonoff-printlight/RESULT = {"POWER":"ON"}
00:21:12 MQT: stat/sonoff-printlight/POWER = ON
00:21:13 CFG: Saved to flash at F9, Count 511, Bytes 2560
00:21:13 HTP: Console
00:21:19 WIF: Checking connection...
00:21:19 WIF: Connected
00:21:29 MQT: tele/sonoff-printlight/STATE = {"Time":"2018-06-15T00:21:29","Uptime":"0T03:10:25","Vcc":3.196,"POWER":"ON","Wifi":{"AP":2,"SSId":"Jemson-AP","RSSI":100,"APMac":"80:2A:A8:54:5C:FA"}}
00:21:29 MQT: tele/sonoff-printlight/SENSOR = {"Time":"2018-06-15T00:21:29","Switch1":"ON"}
00:21:39 WIF: Checking connection...
00:21:39 WIF: Connected
00:21:42 HTP: Main Menu
00:21:59 WIF: Checking connection...
00:21:59 WIF: Connected
00:22:19 WIF: Checking connection...
00:22:19 WIF: Connected
00:22:39 WIF: Checking connection...
00:22:39 WIF: Connected
00:22:44 HTP: Console
00:22:55 HTP: Main Menu
00:23:00 MQT: stat/sonoff-printlight/RESULT = {"POWER":"OFF"}
00:23:00 MQT: stat/sonoff-printlight/POWER = OFF
00:23:00 WIF: Checking connection...
00:23:00 WIF: Connected
00:23:00 CFG: Saved to flash at F8, Count 512, Bytes 2560
00:23:03 HTP: Information
00:23:05 HTP: Main Menu
00:23:05 HTP: Console
00:23:20 WIF: Checking connection...
00:23:20 WIF: Connected

Frogmore42, 2018-06-15:
5.14.0b added the extra logging.

Frogmore42, 2018-06-15:
What other connections do you have?  It looks like there is a switch.  What kind and how is it attached?
It is possible that there is noise on the line that is causing it to switch.

mrjemson, 2018-06-15:
Switch on GPIO14, it is a basic push button switch.
That is a very valid point on the noise potential. I have disabled the switch now and turned on the light again. If it stays on that would be a giveaway that induction noise is the issue.
Given it only ever turns off and never on, this would back up the theory there is noise coming off the light itself, which is an LED with it own transformer in close proximity.

mrjemson, 2018-06-15:
Been on for over an hour now with the switch disabled. Looks like noise coming from the transformer.
I will look to relocate the sonoff or even try shielding the cable.
Really appreciate your help in diagnosing this!

closed 2018-06-15 by mrjemson