arendst/Tasmota #2994: Attempting to flash to so off Basic results in espcomm failure

Music4lity, 2018-06-14:
When attempting to flash Tasmota release 5.9.0 to the Sonoff Basic using Arduino IDE, the following output is generated. Using an FTDI232 board. Have tried reversing the connection for RX and TX with no avail. Only Port available is COM3.

Board: "Generic ESP8266 Module"
Flash Mode: "DOUT"
Flash Frequency: "40MHz"
CPU Frequency: "80MHz"
Flash Size: "1M (64K SPIFFS)"
Debug Port: "Disabled"
Debug Level: "None"
Reset Method: "ck"
Upload Speed: "115200"
Port: "COM3"

```Archiving built core (caching) in: C:\Users\sebas\AppData\Local\Temp\arduino_cache_588056\core\core_esp8266_esp8266_generic_CpuFrequency_80,FlashFreq_40,FlashMode_dout,UploadSpeed_115200,FlashSize_1M64,ResetMethod_ck,Debug_Disabled,DebugLevel_None_____f7025fb08a3574e99f1785948801a9c1.a
Sketch uses 481728 bytes (50%) of program storage space. Maximum is 958448 bytes.
Global variables use 45388 bytes (55%) of dynamic memory, leaving 36532 bytes for local variables.
Maximum is 81920 bytes.
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

Frogmore42, 2018-06-14:
Please search the issues list and read the wiki for help in how to do what you want.
You are making several mistakes all of which have been covered many times in other issues.
1. You are using a very old version
2. You have the wrong flash size/model
3. You have not provided any actionable information on how to help you

The error you are seeing is a generic "failure to communicate" error.  It can be caused by MANY things.  In particular the FTDI is known to not always provide enough power, but that is likely not the cause of this issue.

Getting the right COM port is important as is getting the RX and TX lines correct.  You can verify that by using a terminal program and making sure you can see something on the terminal when you power cycle the Sonoff.  I use Teraterm but there are others that can work.

You also have to put it into bootloader mode to get flashing to work, which you did not indicate you did.

Music4lity, 2018-06-14:
The reason that I am using an older version is due to the fact that this exact same issue has happened on the newest release.

The flash size is the same as was used in a succesful install howto on YouTube, which is why I am using it. If this is the wrong size, then what is the correct size?

Also, how can I use the terminal to test the connection?

Frogmore42, 2018-06-15:
If you are using Windows, you can use Device Manager to find which COM port is available.  Plug and unplug your USB serial converters from the PC and see which COM port shows up under Serial Ports in Device Manager.  This will tell you which one to use.  

Read up on how to use TeraTerm and get a copy of it and then configure it to use the serial port you identified.  When you remove and replace the power (low voltage 3V3, not the high voltage one) while it is still attached to the serial converter, you will see some activity if you have it hooked up correctly.

I don't reccommed using the FTDI to power the Sonoff.  It is not rated to provide enough power and many people struggle with it, but it works fine for some people.  Mine worked fine, but it actually doesn't provide 3V3, instead it outputs USB VBUS which is ~5V.  It did work okay for awhile, but I am pretty sure it led to the early death of at least two of my Sonoffs.

I recommend using Wemos D1 Mini until you are really familiar with the process for flashing.  While flashing a Sonoff is not really hard, there are several things that you have to do right and in the correct order or you get an unintelligible error, that basically says something went wrong.

closed 2018-06-24 by Music4lity