arendst/Tasmota #2995: Automatic power OFF

cirolaferrara, 2018-06-14:
I have a sonoff touch T1 US model 1 gang. Sometimes the light power off itself. I think that is a WIFI / MQTT related problem.
This is the log from console:
```
18:33:06 MQT: tele/sonoff-touch-cameretta/STATE = {"Time":"2018-06-14T18:33:06","Uptime":25,"Vcc":3.108,"POWER":"OFF","Wifi":{"AP":1,"SSId":"MYWIFI","RSSI":68,"APMac":"60:38:E0:B8:E4:71"}}
18:34:05 MQT: stat/sonoff-touch-cameretta/RESULT = {"POWER":"ON"}
18:34:05 MQT: stat/sonoff-touch-cameretta/POWER = ON
18:35:04 MQT: Attempting connection...
18:35:04 MQT: Connected
18:35:05 MQT: tele/sonoff-touch-cameretta/LWT = Online (retained)
18:35:05 MQT: cmnd/sonoff-touch-cameretta/POWER = 
18:35:05 MQT: stat/sonoff-touch-cameretta/RESULT = {"POWER":"OFF"}
18:35:05 MQT: stat/sonoff-touch-cameretta/POWER = OFF
18:35:11 MQT: stat/sonoff-touch-cameretta/RESULT = {"POWER":"ON"}
18:35:11 MQT: stat/sonoff-touch-cameretta/POWER = ON
18:36:49 CMD: PowerOnState
18:36:49 MQT: stat/sonoff-touch-cameretta/RESULT = {"PowerOnState":3}
```
What's wrong? Thanks

Frogmore42, 2018-06-14:
Provide all of the information that is listed in the template and it might be possible to help you.  Anything else would just be a guess and most likely that guess would be you have a retained message sent by your home automation system.  Search "retained message" on the issues.

You probably also have connectivity issues, providing the information requested might provide some hints as to why.

cirolaferrara, 2018-06-15:
I'm using Openhab 2.3 on x64 Linux Ubuntu. I don't know where to start to indagate: Tasmota firmware, Openhab, Mosquitto or something else. I turn off the light and after non predictable time, the light turns itself on.

_/etc/mosquitto/mosquitto.conf_
```
pid_file /var/run/mosquitto.pid
persistence false
log_dest file /var/log/mosquitto/mosquitto.log
allow_anonymous false
password_file /etc/mosquitto/passwd
```
_/var/lib/openhab2/openhab2/services/mqtt.cfg_
```
openhab.url=tcp://localhost:1883
openhab.user=myuser
openhab.pwd=mypwd
openhab.qos=1
openhab.retain=true
openhab.async=false
```

Frogmore42, 2018-06-15:
You have retain set to true for OpenHab, that will cause you lots of problems.  Search the issues for all the others that had issues with retain.  You need to change the setting and clear the messages, there are instructions for doing that in the other issues that reference retain.

cirolaferrara, 2018-06-15:
Thanks. I hope that this solve my issue.

_/var/lib/openhab2/openhab2/services/mqtt.cfg_

```
openhab.url=tcp://localhost:1883
openhab.user=myuser
openhab.pwd=mypwd
openhab.qos=1
#openhab.retain=true
openhab.retain=false
openhab.async=false
```

```
openhab@openhab:~$ sudo service mosquitto stop
openhab@openhab:~$ sudo rm /var/lib/mosquitto/mosquitto.db
openhab@openhab:~$ sudo service mosquitto start
openhab@openhab:~$ sudo systemctl restart openhab2
```

Frogmore42, 2018-06-15:
You will also need to delete the already retained messages.

cirolaferrara, 2018-06-16:
Deleted them with mosquitto.db. no?

Frogmore42, 2018-06-17:
Yes, you need to delete them from the mqtt broker.

cirolaferrara, 2018-06-19:
Ok thanks. Problem is now fixed

closed 2018-06-19 by cirolaferrara