arendst/Tasmota #2996: Support for H803WF or MagicHomeController for WS2812/SK6812

mjmagro, 2018-06-14:
Hi,

Does the Tasmota firmware already support H803WF or MagicHomeController version for WS2812/SK6812 please? Both these devices should contain the ESP chip and can support more easily pixel LEDs since they work in 5V directly (vs 3.3V which gives problems with these pixel strips)?

Devices:
H803WF:
https://www.aliexpress.com/item/led-wifi-controller-support-2048-pixels-full-DIY-WS2812-WS2811-UCS1903-DMX512-artnet-remote-control-Madrix/32861800586.html?spm=2114.search0204.3.48.31935a21CgFSwd&ws_ab_test=searchweb0_0,searchweb201602_3_10152_10151_10065_10344_10068_5722815_10342_10343_10340_5722915_10341_10698_5722615_10697_10696_10084_10083_10618_10307_10301_10303_5722715_5711215_10059_308_100031_10103_441_10624_10623_10622_5711315_5722515_10621_10620,searchweb201603_32,ppcSwitch_5_ppcChannel&algo_expid=2d02e428-4f58-4297-8cdc-4783cb9b222e-7&algo_pvid=2d02e428-4f58-4297-8cdc-4783cb9b222e&transAbTest=ae803_2&priceBeautifyAB=0

Magic Home:
https://www.ebay.com/itm/Nexlux-WiFi-led-controller-Music-for-WS2811-2812-2801-2813-sk6812-IC-LED-Strip/323296343789?hash=item4b45f6b6ed:m:m70salehAWTb2YznZ_5qfbg

Thanks & Regards,

Manuel

Frogmore42, 2018-06-14:
search the issues (like the template says to do) and you will find:
https://github.com/arendst/Sonoff-Tasmota/issues/1867
and others

Jason2866, 2018-06-15:
Hi @mjmagro 
If it has a ESP8266 or ESP8285 inside there is a good chance to get it working with TASMOTA.
WS2812 is already supported by TASMOTA.
So buy one give it a try and come back if you need help...

closed 2018-07-10 by ascillato2