arendst/Tasmota #2997: SONOFF

tcpipchip, 2018-06-14:
Sir 

I would like to use the SONOFF in a HOME AUTOMATION product with TASMOTA software, not the original software!

Will i have legal issues if i use it ? 

Or, if i use my personal esp8266 with TASMOTA software, will i have legal issues with TASMOTA ?

Thank you!

Frogmore42, 2018-06-15:
The license for Tasmota is GPL.  Speak to your legal counsel to understand what that really means for anything that you create on top of it.  I am not a lawyer and never even played one on TV, so anything I tell you will not be worth what you didn't pay me for it.  Generally GPL means you have to provide the source code for anything that create on top.  In the open source community that is not considered a problem, but rather a good thing.  Other people do not always agree with that assessment.

Frogmore42, 2018-06-15:
If you are just using it for home use and are not creating a product you are selling, you are fine.  That is one of the reasons for open source, share the goodness.