arendst/Tasmota #2998: error it took 39s help

arissaraiva, 2018-06-15:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x] Hardware used : sonoff basic
- [x] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

**(Please, remember to close the issue when the problem has been addressed)**

Hello everyone I´ve just got a sonoff and unfortunetly it´s a 1.6.0 firmware so I was unable to sonota, I´m not good soldering and my sonof came with the gpios holes filled, so I improvised holding jumpers while doing the flash, I´m using platformio on windows with a sonoff basic. First when I hold the button and the cables while conecting the usb I heard the usb beep lights on the ftdi and green light on sonoff (don´t know if it´s supposed to happened) after this I click upload and start to compiling but after 30 to 40 s it gives an error (don´t know where to find the log anyone could help me?) something like "error it took 39s " tried several ways not one seems to complete. Also didn´t uncomment any envioroment nor changed the platformio.ini from the folder, should I? On arduino IDE just stays compiling and never show nothing. Anyone could help me, sorry If i´m too noobie.

Jason2866, 2018-06-15:
Hello arissaraiva,

before you try to flash sonoff devices, go get a Wemos D1 mini and do your first steps here.
There are too many things that could be wrong in your setup! 
Nobody can you help in this stage with this informations.

arissaraiva, 2018-06-15:
Just needed some tips, I´ve already did somethings with esp in arduino ide, already flashed one sonoff via OTA, but never worked with platformio. Would like to know where I can find the logs to check what was my error? If it´s really necessary to uncoment the enviroment, since some videos tutorial didn´t say nothing but in the wiki says to uncomment, and when the flashing is happening the led lights from sonoff stays on?

Jason2866, 2018-06-15:
First FTDI adapters are a bad choice. FTDI delievers not enough current. It may work... In many cases it will fail. Second which COM Port do your adapter use?
Does compile work in general? Post some logs screenshots etc.
Again without concrete info no help is possible.
Just install platformio exactly as described in the wiki and tutorial 
https://github.com/arendst/Sonoff-Tasmota/wiki/Beginner-Guide---Create-your-own-Firmware-Build

ascillato, 2018-06-21:
Hi,

If your issue is solved, please close it. Thanks! :+1: