arendst/Tasmota #2999: Sonoff can't connect to WIFI / OpenWRT /DNSMasq

huramentzefix, 2018-06-15:
Hi,
I have seen that people all over the net have this problem. I am not sure if there is a problem with tasmota and DHCP or tasmota and openwrt or what it is.
I just would like to provide my logfiles to be evaluated by someone with further in depth knowledge.

Fri Jun 15 08:04:29 2018 daemon.info hostapd: wlan0-3: STA OTHER_MAC_ADDRESS WPA: group key handshake completed (RSN)
Fri Jun 15 08:04:29 2018 daemon.info hostapd: wlan0-3: STA OTHER_2_MAC_ADDRESS WPA: group key handshake completed (RSN)
Fri Jun 15 08:04:36 2018 daemon.info hostapd: wlan0-2: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: deauthenticated due to local deauth request
Fri Jun 15 08:04:36 2018 kern.info kernel: [110440.170000] eth0: link up (1000Mbps/Full duplex)
Fri Jun 15 08:04:36 2018 kern.info kernel: [110440.170000] br-lan: port 1(eth0) entered forwarding state
Fri Jun 15 08:04:36 2018 kern.info kernel: [110440.180000] br-lan: port 1(eth0) entered forwarding state
Fri Jun 15 08:04:36 2018 daemon.notice netifd: Network device 'eth0' link is up
Fri Jun 15 08:04:37 2018 kern.info kernel: [110440.670000] eth0: link down
Fri Jun 15 08:04:37 2018 daemon.notice netifd: Network device 'eth0' link is down
Fri Jun 15 08:04:37 2018 kern.info kernel: [110441.170000] br-lan: port 1(eth0) entered disabled state
Fri Jun 15 08:04:39 2018 daemon.info hostapd: wlan0-2: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: authenticated
Fri Jun 15 08:04:46 2018 daemon.info hostapd: wlan0-2: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: authenticated
Fri Jun 15 08:04:46 2018 daemon.info hostapd: wlan0-2: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: associated (aid 1)
Fri Jun 15 08:04:46 2018 daemon.info hostapd: wlan0-2: STA MY_SONOFF_MAC_ADDRESS WPA: pairwise key handshake completed (WPA)
Fri Jun 15 08:04:46 2018 daemon.info hostapd: wlan0-2: STA MY_SONOFF_MAC_ADDRESS WPA: group key handshake completed (WPA)
Fri Jun 15 08:04:46 2018 daemon.warn dnsmasq-dhcp[2855]: DHCP packet received on wlan0-2 which has no address
Fri Jun 15 08:04:48 2018 daemon.warn dnsmasq-dhcp[2855]: DHCP packet received on wlan0-2 which has no address

Fri Jun 15 08:04:56 2018 daemon.info hostapd: wlan0-2: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: disassociated
Fri Jun 15 08:04:56 2018 daemon.info hostapd: wlan0-2: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: disassociated
Fri Jun 15 08:04:56 2018 daemon.info hostapd: wlan0-2: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: disassociated
Fri Jun 15 08:04:56 2018 daemon.info hostapd: wlan0-2: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: disassociated
Fri Jun 15 08:04:56 2018 daemon.info hostapd: wlan0-2: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: disassociated
Fri Jun 15 08:04:56 2018 daemon.info hostapd: wlan0-2: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: disassociated
Fri Jun 15 08:04:56 2018 daemon.info hostapd: wlan0-2: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: disassociated

Fri Jun 15 08:04:57 2018 daemon.info hostapd: wlan0-2: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)

Fri Jun 15 08:04:59 2018 daemon.info hostapd: wlan0: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: authenticated

Fri Jun 15 08:05:02 2018 daemon.info hostapd: wlan0: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: authenticated

Fri Jun 15 08:05:09 2018 daemon.info hostapd: wlan0: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: authenticated
Fri Jun 15 08:05:09 2018 daemon.info hostapd: wlan0: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: associated (aid 1)
Fri Jun 15 08:05:09 2018 daemon.info hostapd: wlan0: STA MY_SONOFF_MAC_ADDRESS WPA: pairwise key handshake completed (RSN)
Fri Jun 15 08:05:09 2018 kern.info kernel: [110473.170000] eth0: link up (1000Mbps/Full duplex)
Fri Jun 15 08:05:09 2018 kern.info kernel: [110473.170000] br-lan: port 1(eth0) entered forwarding state
Fri Jun 15 08:05:09 2018 kern.info kernel: [110473.180000] br-lan: port 1(eth0) entered forwarding state
Fri Jun 15 08:05:09 2018 daemon.notice netifd: Network device 'eth0' link is up
Fri Jun 15 08:05:10 2018 kern.info kernel: [110473.670000] eth0: link down
Fri Jun 15 08:05:10 2018 daemon.notice netifd: Network device 'eth0' link is down
Fri Jun 15 08:05:10 2018 kern.info kernel: [110474.170000] br-lan: port 1(eth0) entered disabled state

Fri Jun 15 08:08:27 2018 daemon.info hostapd: wlan0: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: authenticated
Fri Jun 15 08:08:27 2018 kern.info kernel: [110671.170000] eth0: link up (1000Mbps/Full duplex)
Fri Jun 15 08:08:27 2018 kern.info kernel: [110671.170000] br-lan: port 1(eth0) entered forwarding state
Fri Jun 15 08:08:27 2018 kern.info kernel: [110671.180000] br-lan: port 1(eth0) entered forwarding state
Fri Jun 15 08:08:27 2018 daemon.notice netifd: Network device 'eth0' link is up
Fri Jun 15 08:08:28 2018 kern.info kernel: [110671.670000] eth0: link down
Fri Jun 15 08:08:28 2018 daemon.notice netifd: Network device 'eth0' link is down
Fri Jun 15 08:08:28 2018 kern.info kernel: [110672.170000] br-lan: port 1(eth0) entered disabled state
Fri Jun 15 08:08:30 2018 daemon.info hostapd: wlan0: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: authenticated
Fri Jun 15 08:08:30 2018 daemon.info hostapd: wlan0: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: associated (aid 1)
Fri Jun 15 08:08:30 2018 daemon.info hostapd: wlan0: STA MY_SONOFF_MAC_ADDRESS WPA: pairwise key handshake completed (RSN)
Fri Jun 15 08:08:30 2018 daemon.info dnsmasq-dhcp[2855]: 2882338817 available DHCP range: MY_IP_ADDRESS.201 -- MY_IP_ADDRESS.250
Fri Jun 15 08:08:30 2018 daemon.info dnsmasq-dhcp[2855]: 2882338817 client provides name: sonoff-0460
Fri Jun 15 08:08:30 2018 daemon.info dnsmasq-dhcp[2855]: 2882338817 DHCPDISCOVER(br-lan) MY_SONOFF_MAC_ADDRESS 
Fri Jun 15 08:08:30 2018 daemon.info dnsmasq-dhcp[2855]: 2882338817 tags: lan, br-lan
Fri Jun 15 08:08:30 2018 daemon.info dnsmasq-dhcp[2855]: 2882338817 DHCPOFFER(br-lan) MY_IP_ADDRESS.245 MY_SONOFF_MAC_ADDRESS 
Fri Jun 15 08:08:30 2018 daemon.info dnsmasq-dhcp[2855]: 2882338817 requested options: 1:netmask, 3:router, 28:broadcast, 6:dns-server, 
Fri Jun 15 08:08:30 2018 daemon.info dnsmasq-dhcp[2855]: 2882338817 requested options: 15:domain-name, 44:netbios-ns, 46:netbios-nodetype, 
Fri Jun 15 08:08:30 2018 daemon.info dnsmasq-dhcp[2855]: 2882338817 requested options: 47:netbios-scope, 31:router-discovery, 33:static-route, 
Fri Jun 15 08:08:30 2018 daemon.info dnsmasq-dhcp[2855]: 2882338817 requested options: 121:classless-static-route, 43:vendor-encap
Fri Jun 15 08:08:30 2018 daemon.info dnsmasq-dhcp[2855]: 2882338817 next server: MY_IP_ADDRESS.4
Fri Jun 15 08:08:30 2018 daemon.info dnsmasq-dhcp[2855]: 2882338817 sent size:  1 option: 53 message-type  2
Fri Jun 15 08:08:30 2018 daemon.info dnsmasq-dhcp[2855]: 2882338817 sent size:  4 option: 54 server-identifier  MY_IP_ADDRESS.4
[... 8 lines cut ...]

Fri Jun 15 08:08:44 2018 kern.info kernel: [110687.670000] eth0: link up (1000Mbps/Full duplex)
Fri Jun 15 08:08:44 2018 kern.info kernel: [110687.670000] br-lan: port 1(eth0) entered forwarding state
Fri Jun 15 08:08:44 2018 kern.info kernel: [110687.680000] br-lan: port 1(eth0) entered forwarding state
Fri Jun 15 08:08:44 2018 daemon.notice netifd: Network device 'eth0' link is up
Fri Jun 15 08:08:44 2018 kern.info kernel: [110688.170000] eth0: link down
Fri Jun 15 08:08:45 2018 daemon.notice netifd: Network device 'eth0' link is down
Fri Jun 15 08:08:45 2018 kern.info kernel: [110688.670000] br-lan: port 1(eth0) entered disabled state
Fri Jun 15 08:08:46 2018 daemon.info hostapd: wlan0: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: disassociated
Fri Jun 15 08:08:46 2018 daemon.info hostapd: wlan0: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: disassociated
Fri Jun 15 08:08:46 2018 daemon.info hostapd: wlan0: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: disassociated
Fri Jun 15 08:08:46 2018 daemon.info hostapd: wlan0: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: disassociated
Fri Jun 15 08:08:46 2018 daemon.info hostapd: wlan0: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: disassociated
Fri Jun 15 08:08:46 2018 daemon.info hostapd: wlan0: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: disassociated
Fri Jun 15 08:08:46 2018 daemon.info hostapd: wlan0: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: disassociated
Fri Jun 15 08:08:47 2018 daemon.info hostapd: wlan0: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)

Fri Jun 15 08:08:53 2018 daemon.info hostapd: wlan0-2: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: authenticated
Fri Jun 15 08:09:00 2018 kern.info kernel: [110704.170000] eth0: link up (1000Mbps/Full duplex)
[... cut ...] 4.170000] br-lan: port 1(eth0) entered forwarding state
Fri Jun 15 08:09:00 2018 kern.info kernel: [110704.180000] br-lan: port 1(eth0) entered forwarding state
Fri Jun 15 08:09:00 2018 daemon.notice netifd: Network device 'eth0' link is up
Fri Jun 15 08:09:01 2018 kern.info kernel: [110704.670000] eth0: link down
Fri Jun 15 08:09:01 2018 daemon.notice netifd: Network device 'eth0' link is down
Fri Jun 15 08:09:01 2018 kern.info kernel: [110705.170000] br-lan: port 1(eth0) entered disabled state

huramentzefix, 2018-06-15:
It will connect eventually after an hour or two.

Fri Jun 15 08:24:36 2018 daemon.info hostapd: wlan0-2: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: deauthenticated due to local deauth request

Fri Jun 15 08:23:48 2018 daemon.info hostapd: wlan0: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)

Fri Jun 15 08:15:36 2018 kern.info kernel: [111100.170000] eth0: link up (1000Mbps/Full duplex)
Fri Jun 15 08:15:36 2018 kern.info kernel: [111100.170000] br-lan: port 1(eth0) entered forwarding state
Fri Jun 15 08:15:36 2018 kern.info kernel: [111100.180000] br-lan: port 1(eth0) entered forwarding state
Fri Jun 15 08:15:36 2018 daemon.notice netifd: Network device 'eth0' link is up
Fri Jun 15 08:15:37 2018 kern.info kernel: [111100.670000] eth0: link down
Fri Jun 15 08:15:37 2018 daemon.notice netifd: Network device 'eth0' link is down
Fri Jun 15 08:15:37 2018 kern.info kernel: [111101.170000] br-lan: port 1(eth0) entered disabled state
Fri Jun 15 08:15:53 2018 kern.info kernel: [111116.670000] eth0: link up (1000Mbps/Full duplex)
Fri Jun 15 08:15:53 2018 kern.info kernel: [111116.670000] br-lan: port 1(eth0) entered forwarding state
Fri Jun 15 08:15:53 2018 kern.info kernel: [111116.680000] br-lan: port 1(eth0) entered forwarding state
Fri Jun 15 08:15:53 2018 daemon.notice netifd: Network device 'eth0' link is up
Fri Jun 15 08:15:53 2018 kern.info kernel: [111117.170000] eth0: link down
Fri Jun 15 08:15:54 2018 daemon.notice netifd: Network device 'eth0' link is down
Fri Jun 15 08:15:54 2018 kern.info kernel: [111117.670000] br-lan: port 1(eth0) entered disabled state
Fri Jun 15 08:15:58 2018 daemon.info hostapd: wlan0: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: authenticated
Fri Jun 15 08:16:05 2018 daemon.info hostapd: wlan0: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: authenticated
Fri Jun 15 08:16:05 2018 daemon.info hostapd: wlan0: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: associated (aid 1)
Fri Jun 15 08:16:05 2018 daemon.info hostapd: wlan0: STA MY_SONOFF_MAC_ADDRESS WPA: pairwise key handshake completed (RSN)
Fri Jun 15 08:16:09 2018 kern.info kernel: [111133.170000] eth0: link up (1000Mbps/Full duplex)
Fri Jun 15 08:16:09 2018 kern.info kernel: [111133.170000] br-lan: port 1(eth0) entered forwarding state
Fri Jun 15 08:16:09 2018 kern.info kernel: [111133.180000] br-lan: port 1(eth0) entered forwarding state
Fri Jun 15 08:16:09 2018 daemon.notice netifd: Network device 'eth0' link is up
Fri Jun 15 08:16:10 2018 kern.info kernel: [111133.670000] eth0: link down
Fri Jun 15 08:16:10 2018 daemon.notice netifd: Network device 'eth0' link is down
Fri Jun 15 08:16:10 2018 kern.info kernel: [111134.170000] br-lan: port 1(eth0) entered disabled state
Fri Jun 15 08:16:18 2018 daemon.info hostapd: wlan0: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: disassociated
Fri Jun 15 08:16:18 2018 daemon.info hostapd: wlan0: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: disassociated
Fri Jun 15 08:16:18 2018 daemon.info hostapd: wlan0: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: disassociated
Fri Jun 15 08:16:18 2018 daemon.info hostapd: wlan0: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: disassociated
Fri Jun 15 08:16:18 2018 daemon.info hostapd: wlan0: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: disassociated
Fri Jun 15 08:16:18 2018 daemon.info hostapd: wlan0: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: disassociated
Fri Jun 15 08:16:18 2018 daemon.info hostapd: wlan0: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: disassociated
Fri Jun 15 08:16:19 2018 daemon.info hostapd: wlan0: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Fri Jun 15 08:16:21 2018 daemon.info hostapd: wlan0-2: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: authenticated
Fri Jun 15 08:16:25 2018 daemon.info hostapd: wlan0-2: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: authenticated
Fri Jun 15 08:16:25 2018 daemon.info hostapd: wlan0-2: STA MY_SONOFF_MAC_ADDRESS IEEE 802.11: associated (aid 1)
Fri Jun 15 08:16:26 2018 kern.info kernel: [111149.670000] eth0: link up (1000Mbps/Full duplex)
Fri Jun 15 08:16:26 2018 kern.info kernel: [111149.670000] br-lan: port 1(eth0) entered forwarding state
Fri Jun 15 08:16:26 2018 kern.info kernel: [111149.680000] br-lan: port 1(eth0) entered forwarding state
Fri Jun 15 08:16:26 2018 daemon.notice netifd: Network device 'eth0' link is up
Fri Jun 15 08:16:26 2018 daemon.info hostapd: wlan0-2: STA MY_SONOFF_MAC_ADDRESS WPA: pairwise key handshake completed (WPA)
Fri Jun 15 08:16:26 2018 daemon.info hostapd: wlan0-2: STA MY_SONOFF_MAC_ADDRESS WPA: group key handshake completed (WPA)
Fri Jun 15 08:16:26 2018 kern.info kernel: [111150.170000] eth0: link down
Fri Jun 15 08:16:27 2018 daemon.notice netifd: Network device 'eth0' link is down
Fri Jun 15 08:16:27 2018 kern.info kernel: [111150.670000] br-lan: port 1(eth0) entered disabled state
Fri Jun 15 08:16:42 2018 kern.info kernel: [111166.170000] eth0: link up (1000Mbps/Full duplex)
Fri Jun 15 08:16:42 2018 kern.info kernel: [111166.170000] br-lan: port 1(eth0) entered forwarding state
Fri Jun 15 08:16:42 2018 kern.info kernel: [111166.180000] br-lan: port 1(eth0) entered forwarding state
Fri Jun 15 08:16:42 2018 daemon.notice netifd: Network device 'eth0' link is up
Fri Jun 15 08:16:43 2018 kern.info kernel: [111166.670000] eth0: link down
Fri Jun 15 08:16:43 2018 daemon.notice netifd: Network device 'eth0' link is down
Fri Jun 15 08:16:43 2018 kern.info kernel: [111167.170000] br-lan: port 1(eth0) entered disabled state

[... cut ...]
Fri Jun 15 08:26:03 2018 kern.info kernel: [111727.170000] br-lan: port 1(eth0) entered forwarding state
Fri Jun 15 08:26:03 2018 kern.info kernel: [111727.180000] br-lan: port 1(eth0) entered forwarding state
Fri Jun 15 08:26:03 2018 daemon.notice netifd: Network device 'eth0' link is up
Fri Jun 15 08:26:04 2018 kern.info kernel: [111727.670000] eth0: link down
Fri Jun 15 08:26:04 2018 daemon.notice netifd: Network device 'eth0' link is down
Fri Jun 15 08:26:04 2018 kern.info kernel: [111728.170000] br-lan: port 1(eth0) entered disabled state

Jason2866, 2018-06-15:
It is not a Tasmota issue. The problem is OpenWRT or the Esp wifi library. Both are not under the control of Tasmota. Plz post your issue there and close this thread.

huramentzefix, 2018-06-15:
I have read this statement many times that this is not a problem caused by tasmota.
And just as many times  I have read about other people having this same Issue.

By no means I am an expert and questioning the capabilities of any developer.
I am just putting 1+1 together.
My sonoff devices with the itead software manages to connect in less than 10 seconds to the same OpenWRT router with the same Esp chip.

As soon as I install tasmota without altering the wifi library they have this behaviour.

1+1=2? Am I wrong? I am not an expert, nor a developer. It seems that I suck in forensics and math? I really don't get it, sorry for my ignorance and stupidity. Please explain to me where I make the mistake in putting things together. I like to learn and progress.

znanev, 2018-06-15:
@huramentzefix There is something strange going on in your network. An Ethernet port (*eth0*) is going up/down several times a minute. What is connected to this port - it this your upstream connection, or is it a port on the LAN bridge?

It might be that your router is dropping some packets. Look for DHCPDISCOVER in your log - if this gets to the router, there should be a corresponding DHCPOFFER packet, effectively serving an IP from your network to your Sonoff device.

Which version of the firmware do you use? There are some firmware releases with Core version 2.4.1, which makes WiFi unstable in some devices and networks combinations.

huramentzefix, 2018-06-15:
Thank you I appreciate your help.

might be a physical defect when I jtaged the device.
Just changed the WAN IP to different one from the LAN so that NAT works. (the router in front of that has NAT)
and I uninstalled wshaper. no change though.

I will have to find a new device that runs openwrt, mosquitto,  and has usb.
Any suggestions?

should I close this and open a new case if it persists when I have it up and running on my side?

znanev, 2018-06-15:
If you have a Raspberry Pi - give that a go. It is more than enough to run a home automation system on it. Surely it can't substitute a router, but its power consumption is negligible so you can leave it running 24/7. As a bonus your router will not be loaded with additional tasks.

huramentzefix, 2018-06-16:
@znanev Thank you, I thought about an arduino before. I just haven't found out how to make it run stable without having the card go bad due to r/w.
I was experimenting with an A70H with A23 cpu. Just struggling with u-boot a bit.
Was plaing on running a ubuntu on it for the server.

I guess I get myself a PI. is there any good documentation for having it running with reliable memory or a second memory card and have automatic mirroring each day?

@znanev kernel 3.18.23

Can anything be changed if I tweak the RF bridge?
How about changing the bridge to NO RF Calibration?
that reduces init time from 20ms to 2ms. has anyone tried if there is room for improovments?

Addresses for the 8Mb flash:

blank.bin 0xFB000 Initializes the RF_CAL parameter area.

blank.bin 0x7E000
Stores the default RF parameter values; the BIN has to be downloaded into ﬂash at least once. If the RF_CAL parameter area is initialized, this BIN has to be downloaded too. 

esp_init_data_default.bin 0xFC000
Stores the default RF parameter values; the BIN has to be downloaded into ﬂash at least once. If the RF_CAL parameter area is initialized, this BIN has to be downloaded too.

what is the correct procedure for flashing? a lot of people are under the impression that it is enough to download tasmota.bin to 0x00000 only and no other files.
that is what I have done in the past.

But just after my research yesterday and today I can see that there have to be  some configuration files for the RF bridge in the flash?
Depending on flash size on diferrent addresses.

I was writing a 1MBblank each time before I flashed tasmota. so there is for sure no config file for the bridge in my flash.
right now I am trying to flash the two times blank.bin (not 1MB blank) and the esp_init_data_default.bin together with tasmota.

I also flashed the bin here blank.bin 0xFE000 Initializes the ﬂash user parameter area.
but I guess that is not needed. when I read the com port after flashing I just have this matrix-movie stuff running over the screen.

will try again with just the two bins for the rf and the rf config file.

can someone give me correct instructions for configuration and flashing please?
why haven't I read anything or very little about these rf configs that seem to be needed?

huramentzefix, 2018-06-16:
I have solved it!! thanks to you guys. I would like to share this with you to see if that helps the other hundreds of users that have the same issue, or if it was just me who didn't know to upload the esp config file.bin and the blank.bin onto the chip. together with tasmota???!!!

It seems to me that in my case, just the config file for the ESP wifi bridge needs to be flashed onto the flash.
I had never done that before.

But now it is reproducable, from hitting "restart" or typing "Restart 1" in the console, I have 8 second till I get a ping again from the sonoff.
From plugging the power plug in till I get a ping it's 18 seconds now!
Blazingly fast wifi connection, no more restarts of the device. version 5.14 Haha I am so happy!

what did it for me is:
1. download the expressif software package from the expresif webpage.
2. open ESP8266 DOWNLOAD TOOL V3.6.4
3. load 3 files into ESP8266 DOWNLOAD TOOL V3.6.4, blank.bin @ 0xFE000,  esp_init_data_default.bin @ 0xFC000, sonoffv5.14.0-classic.bin @ 0x00000  (this works only for 8Mbit flash)

4. DOWNLOAD TOOL V3.6.4 settings in SPIDownload: CrystalFrequ 26M, SPI SPEED 40MHz, SPI MODE DOUT, FLASH SIZE 8Mbit (the adresses mentioned in point (3.) are only valid for the 8Mbit (1MByte) chip, select DoNotChgBin, COMX, BAUD 115200

Then in RFConfig TxTargetPowerConfig all original, LowPowerMode all original, CrystalFreq 26Mhz, TOUT PinConf select TOUT_ADC_EN VDD:3.3V, RFInit mode FullRFCal in RFInit

5. DO NOT select these files for downloading yet

6. disconnect mains, never connect both at the same time! UART dongle can be connected while mains are connected to read serial data but not together with the 3.3V, connect to UART USB dongle and connect 3.3V by holding the button on the sonoff down. while holding the button push start (no files selected) and It will show you crystal frequency and memory size ...

7. disconnect 3.3V, select files for downloading: blank.bin@0xFE000, esp_init_data_default.bin@0xFC000 and sonoffv5.14.0-classic.bin@0x00000

8. like (6.) just with the 3 files selected. when it says "FINISH" you will hopefully be as happy as I am.

9. if your sonoff doesn't work the first time when you plug it into power, unplug and plug it in again. sometimes it doesn't work the first time after flashing.
have serial connected and write in windows terminal "type com6" (do not connect 3.3V, only TX, RX & GND) so that you can see what is going on.

happy days!

Kielkopf, 2018-06-17:
@znanev Tasmota Version | 6.0.0a -- 2018.06.17 16:01:35 -- 2_3_0/1.5.3(aec24ac9)
works fine
Andreas

huramentzefix, 2018-06-17:
mine works like a sharm as well.

you flashed with ESP8266 DOWNLOAD TOOL V3.6.4,
blank.bin @ 0xFE000,
esp_init_data_default.bin @ 0xFC000

and 5.14, 5.12 & 5-11 are working for you?
I had 5.14 running all night, o restarts, no lost signal ...
Never before I had a sonoff running so smooth!

Just fiddling with a 32Mb flash chips which are not being recognised yet.
any hints there?

huramentzefix, 2018-06-17:
just wondering if an OTA update will overwrite 0xFE000 & 0xFC000 and how to write protect them or tell the ota update not tu use that memory or flash it together again with or after the ota update process to those adresses?

znanev, 2018-06-17:
@huramentzefix Wow, just managed to catch up with your responses.

It seems that you have greatly over-complicated the flashing process :) It is quite easy to just use the Python-based [esptool](https://github.com/espressif/esptool) for this, like:

(erase SPI chip flash):
```esptool --port /dev/ttyUSB0 erase_flash```

(upload compiled Tasmota binary):
```esptool --port /dev/ttyUSB0 write_flash -fs 1MB -fm dout 0x0 sonoff.bin```

- _/dev/ttyUSB0_ is the port where your USB to serial adapter is connected (change according to your environment)
- _sonoff.bin_ is the compiled binary file of the Tasmota firmware

This way you won't have to fiddle with ESP8266 download tool and different memory regions. The esptool hides all complexities from the end-user - give it a go! :)

It is also possible to use a 32 Mbit flash chip - no problems at all. I changed the original SPI chips in all my Sonoff devices with 32 Mbit Winbond chips, but this means that I have to compile the firmware myself when I need a new binary. Have a look at this Wiki page:

[How to Expand Sonoff Basic to 4MB Flash](https://github.com/arendst/Sonoff-Tasmota/wiki/How-to-Expand-Sonoff-Basic-to-4MB-Flash)

and especially the section [Flashing](https://github.com/arendst/Sonoff-Tasmota/wiki/How-to-Expand-Sonoff-Basic-to-4MB-Flash#flashing)

Hope this helps!

znanev, 2018-06-17:
@Kielkopf - I'm glad that it worked!

@huramentzefix  - give version v6.0.0 a go too (just get the latest code from branch _development_) - it should work with your Sonoffs too.

znanev, 2018-06-18:
@huramentzefix You don't need to upload blank.bin and esp_init_data_dafault.bin yourself. This is only required if you use Espressif's flashing tool. When you use the Arduino SDK, which is the case with Tasmota, the output of the build is just a single binary file (sonoff.bin by default). The Arduino SDK takes care to fill blank.bin and esp_init_data_dafault sections in the final bin. So you really don't have to specify any other files or memory addresses when you flash the sonoff.bin file:

```esptool --port /dev/ttyUSB0 write_flash -fs 1MB -fm dout 0x0 sonoff.bin```

This works irrespective of SPI chip flash size. Memory outside the 8Mbit boundary won't be used if you use the default build to compile a sonoff.bin file.

Read again the Flashing section of the Wiki page for expanding the memory to 4MB. You need to modify your platformio.ini file as instructed, also you have to download ```esp8266.flash.4m1m.ld``` from here: https://github.com/platformio/platformio-pkg-ldscripts and put it in the directory where platformio.ini file is.

closed 2018-07-10 by ascillato2