arendst/Tasmota #3001: sonoff 4ch pro relay on on reboot

Smandurlo, 2018-06-15:
Hey there,
first of all thanks for the incredible firmware. I have a weird problem on reboot or power on after power lost.

First time I updated the 4ch pro from original firmware with sonota. The relay 2 switched on after every reboot or when disconnected from power. Setting up into console PowerOnState 0 (which should force all the relay to OFF after a reboot) didn't change it. Tried with PowerOnState 3, same result: relay 2 on.

So, I tried with a reset: reset 1. I was not able to find the sonoff in any way. So, I started to work with Arduino IDE. I was able to flash 2 sonoff basic and 1 sonoff pow r2. No problem. Then the 4CH pro. Again ok.

Configured as 4ch pro, reboot and now the relay turning on on reboot was the relay number 3. Again: poweronstate 0, poweronstate 3... nothing, still turning on on reboot.

After some fiddling with the relays, number 4 started to turn on on reboot.

I disabled mqtt just in case it was something connected with it. No luck.

So, I am lost. Now it turn on with relay number 1.

This is the console log on start up:

00:00:00 Progetto sonoff Sonoff (Topic sonoff, Riserva DVES_9F53D5, GroupTopic sonoffs) Versione 5.14.0-2_3_0
00:00:00 WIF: Connessione ad AP1 XXXX in modalità 11N come sonoff-5077...
00:00:04 RSL: RESULT = {"POWER1":"ON"}
00:00:04 RSL: POWER1 = ON
00:00:04 RSL: RESULT = {"POWER2":"OFF"}
00:00:04 RSL: POWER2 = OFF
00:00:05 WIF: Connesso
00:00:05 DNS: Inizializzato
00:00:05 HTP: Web server attivo su sonoff-5077.local con indirizzo IP 192.168.xxx.xxx
00:00:06 RSL: INFO1 = {"Module":"Sonoff 4CH Pro","Version":"5.14.0","FallbackTopic":"DVES_9F53D5","GroupTopic":"sonoffs"}
00:00:06 RSL: INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-5077","IPAddress":"192.168.xxx.xxx"}
00:00:06 RSL: INFO3 = {"RestartReason":"Software/System restart"}
13:38:18 RSL: RESULT = {"POWER1":"ON"}
13:38:18 RSL: POWER1 = ON
13:38:18 RSL: RESULT = {"POWER2":"OFF"}
13:38:18 RSL: POWER2 = OFF
13:38:18 RSL: RESULT = {"POWER3":"OFF"}
13:38:18 RSL: POWER3 = OFF
13:38:18 RSL: RESULT = {"POWER4":"OFF"}
13:38:18 RSL: POWER4 = OFF
13:38:25 RSL: STATE = {"Time":"2018-06-15T13:38:25","Uptime":"0T00:00:14","Vcc":3.126,"POWER1":"ON","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"xxxx","RSSI":100,"APMac":"xxxxxxxxxxxxx"}}

Smandurlo, 2018-06-15:
update: now it keeps continue to switch from 1 to 2 and viceversa. If the relay 2 is ON -> reboot -> relay numer 2 is OFF, relay numer 1 is ON -> reboot -> relay numer 1 is OFF, relay numer 2 is ON

Smandurlo, 2018-06-15:
update: the buttons don't work properly. The R2 button turn on the relay 2, but it doesn't turn it off.

Jason2866, 2018-06-15:
Erase the flash completly and use a precompiled version.
It could be that your selfcompiled version is buggy.

Smandurlo, 2018-06-15:
Actually I didn't use any selfcompiled version. I used the 5.14 from the source code.

How to erase everything? reset 2 into console? Then Arduino IDE and upload everything back?

Jason2866, 2018-06-15:
Yes, reset 2.  After that flash the device again. Maybe flash was buggy.
After that configure your device type and post Log of status 0

Smandurlo, 2018-06-15:
reset 2 done. Flash done. I am not able to see it again. I think it keeps continue to reboot endless. The buttons don't work, the blue led blinks for a while then turn off and again flashing for a while then turn off in loop.

Scanning the subnet doesn't show it. Any idea?

Jason2866, 2018-06-15:
I had a sonoff basic that was reacting like your device. It came back to life with erasing it via esptool.
I cant explain why but it worked.
I did: esptool --port COM5 --baud 115200 erase_flash 
You have too change the COM Port to your settings

Smandurlo, 2018-06-15:
ok, I am fighting with installing esptool on windows :-) I will let you know. Thanks!

Smandurlo, 2018-06-15:
yeeeeeeeeeeeeeeeeeeeeeeeeeee!!!!!!!

After a lot of unintuitive procedures, I was able to use esptool to completly erase the flash. Then upload the firmware from Arduino IDE. No relays on after reboot :-) THANK YOU!!!

00:00:00 WIF: Connecting to AP1 XXXX in mode 11N as sonoff-5077...
00:00:07 WIF: Connected
00:00:07 DNS: Initialized
00:00:07 HTP: Web server active on sonoff-5077.local with IP address 192.168.xxx.xxx
16:27:09 MQT: Attempting connection...
16:27:09 MQT: Connected
16:27:09 MQT: tele/sonoff/LWT = Online (retained)
16:27:09 MQT: cmnd/sonoff/POWER = 
16:27:09 MQT: tele/sonoff/INFO1 = {"Module":"Sonoff 4CH Pro","Version":"5.14.0","FallbackTopic":"DVES_9F53D5","GroupTopic":"sonoffs"}
16:27:09 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-5077","IPAddress":"192.168.xxx.xxx"}
16:27:09 MQT: tele/sonoff/INFO3 = {"RestartReason":"Software/System restart"}
16:27:10 MQT: stat/sonoff/RESULT = {"POWER1":"OFF"}
16:27:10 MQT: stat/sonoff/POWER1 = OFF
16:27:10 MQT: stat/sonoff/RESULT = {"POWER2":"OFF"}
16:27:10 MQT: stat/sonoff/POWER2 = OFF
16:27:10 MQT: stat/sonoff/RESULT = {"POWER3":"OFF"}
16:27:10 MQT: stat/sonoff/POWER3 = OFF
16:27:10 MQT: stat/sonoff/RESULT = {"POWER4":"OFF"}
16:27:10 MQT: stat/sonoff/POWER4 = OFF
16:27:17 MQT: tele/sonoff/STATE = {"Time":"2018-06-15T16:27:17","Uptime":"0T00:00:16","Vcc":3.410,"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","Wifi":{"AP":1,"SSId":"xxxxxxxx","RSSI":94,"APMac":"xxxxxxxxxxxx"}}

closed 2018-06-15 by Smandurlo