arendst/Tasmota #3004: MQTT and switches

vampyre666, 2018-06-15:
Hi all,

i have a nodemcu board running Tasmota 6.0.0a and seem to have an issue
i have defined 2 switches connected to reed-switches

[image]

set switchtopic2 1
set switchtopic2 2
switchtopic  doorsensors

i do not seem to be getting the POWER1 / POWER2 / POWER3 to differentiate the different switches they all seem to be sending to the same topic see below.

20:48:10 MQT: tele/nodemcu_tas/SENSOR = {"Time":"2018-06-15T20:48:10","Switch1":"ON","Switch2":"ON","Switch3":"ON"}
20:48:13 MQT: cmnd/doorsensors/POWER = OFF   <--- output from triggering switch2
20:48:17 MQT: cmnd/doorsensors/POWER = ON   <--- output from triggering switch2
20:48:17 MQT: cmnd/doorsensors/POWER = OFF <--- output from triggering switch3
20:48:17 MQT: cmnd/doorsensors/POWER = ON <--- output from triggering switch2

any help will be appreciated

Thanks in advance

Jason2866, 2018-06-16:
Take a look at #1567  Here is the PIR the switch

Jason2866, 2018-06-16:
You can try to define virtual relays to not needed gpios. With this you dont have to change anything to switchtopic. You will get the relay status power on off

vampyre666, 2018-06-16:
Jason, thank that seemed to have worked, can you explain why the virtual relays are required ?

15:53:28 MQT: cmnd/doorsensors/POWER2 = ON
15:53:31 MQT: cmnd/doorsensors/POWER2 = OFF
15:53:32 MQT: cmnd/doorsensors/POWER2 = ON
15:53:33 MQT: cmnd/doorsensors/POWER3 = OFF
15:53:34 MQT: cmnd/doorsensors/POWER3 = ON

Jason2866, 2018-06-16:
Read here: https://github.com/arendst/Sonoff-Tasmota/wiki/Understanding-SwitchMode-and-SwitchTopic
May things get clear if you look at "USE CASES"
You do not change the result message with switchtopic! 
Only the topic as command suggests

boris099, 2018-06-19:
@Jason2866 
I did change my GPIO14 to Relay2, and I can see now in the sonoff basic GUI the second "button.
But I am connecting a PIR to GND and GPIO14 and when it closes nothing happens.
When I am using Switch1 it immediately switch the physical relay.
I would like to use either dummy Switch2 or dummy Relay2 (don´t know what works) to get a message triggered to OpenHab, that shows GPIO14 against Ground was closed. Relay2 seems not to work for this.

Using Switch2 for this seems to be complex therefore I thought I use Relay2 as you stated above and everything works automatically reflecting state POWER2 for OpenHab.

Can you help me here?

boris099, 2018-06-19:
I just tried and assigned SWITCH2 to GPIO14 and when I am closing against Ground it switches the physical relay as SWITCH1 would the only differnence I notice is the LED does not react.

Jason2866, 2018-06-19:
Hello boris099
I need a pull down resistor to get the PIR working.
My PIR goes in Status high, when it detects motion.
Without the pulldown resistor i get no valid signal change....

boris099, 2018-06-19:
Hi Jason2866, that is all working I do have a relay in between to switch without currency (sorry I might miss the right words). When GPIO014 is set to SWITCH1 all works but the physical relay gets switched.
I just would like to use the physical switch for something else.
Therefore I thought I can set GPIO014 to RELAY2 and read the mqqt command POWER2 in OpenHab.
Partially this seems to work the bad thing is that when I connect GPIO014 to Ground with the PIR nothing happens, why not?

Jason2866, 2018-06-19:
Setup:  Connect your PIR contact via relay contact to GPIO 14 AND define a virtual relay2 to a other GPIO that is free. With this you get a power2 on off message.

boris099, 2018-06-19:
Wait a second. I do have four GPIOs in sonoff basic. 
GPIO1 serial out, GPIO3 serial in, GPIO4 and GPIO14.
Currently only GPIO14 is set to SWiTCH2. Rest is on None.
The relay of the PIR is connected to GPIO14 and Ground.
What exactly should I use now - leave GPIO14 on SWITCH2 and additionally RELAY2 for GPIO4?

boris099, 2018-06-19:
OK THANKS, this works exactly as it should!!!

boris099, 2018-06-19:
This means when no RELAY2 is assigned SWITCH2 will switch also RELAY1 as SWITCH1 does

Jason2866, 2018-06-19:
Yes, this is the wanted behaviour. You can connect more switches to switch the one relay.
Switch2 reacts only on relay2 if there is one

ascillato, 2018-06-21:
Hi,

If your issue is solved, please close it. Thanks :+1:

closed 2018-06-27 by ascillato2

brucysNZ, 2019-05-22:
@boris099 
I am interested in your solution. I am wanting to do the same as you have
Sonoff with tasmota. I do not want to switch the relay onboard the local sonoff. I have the PIR connected to GPI03. on activation of the PIR I want to send a MQTT command to a service call as Automation to turn on a light in daughters room.
I want to leave the relay for some other plan, so just want to use GPI03 to turn on a light 
Is this something you did or similar?
cheers
Bruce