arendst/Tasmota #3006: When WiFi router is restarted sonoff and wemos hangs no AP mode switching

roblad, 2018-06-16:
Hi,

When I lost WiFi from router side i.e. restart usualy takes 30 sek. even more wpa2, the SONOFF CH4PRO, hangs with relay 1 status on, no switching to AP mode. Similar problem I have in WEMOS PRO when I restart wifi module hangs no wifi AP mode. For both the hard reset is needed. 

platform = espressif8266@1.7.0  ; v2.4.1

- [SONOF4CHPRO ] Hardware used : 
- [ ] Provide the output of command ``status 0`` :
```
10:03:27 MQT: stat/podlewaczka/STATUS = {"Status":{"Module":33,"FriendlyName":["Podlewaczka","Podlewaczka","Podlewaczka","Podlewaczka"],"Topic":"podlewaczka","ButtonTopic":"podlewaczka","Power":0,"PowerOnState":0,"LedState":6,"SaveData":1,"SaveState":0,"ButtonRetain":1,"PowerRetain":0}}
10:03:27 MQT: stat/podlewaczka/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"podlewanie,Topic_podlewaczka","OtaUrl":"http://192.168.100.113/SONOFF_4CHPRO_RL-ds18x20-PL.bin","RestartReason":"Software/System restart","Uptime":"0T00:01:32","StartupUTC":"2018-06-16T08:01:55","Sleep":0,"BootCount":240,"SaveCount":412,"SaveAddress":"F6000"}}
10:03:27 MQT: stat/podlewaczka/STATUS2 = {"StatusFWR":{"Version":"6.0.0a","BuildDateTime":"2018-06-16T09:55:40","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
10:03:27 MQT: stat/podlewaczka/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["robert_private","robert_private2"],"TelePeriod":300,"SetOption":["0080C248","95818000"]}}
10:03:27 MQT: stat/podlewaczka/STATUS4 = {"StatusMEM":{"ProgramSize":483,"Free":520,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000415","0D102594","0C000001","000000C0","00000000"]}}
10:03:27 MQT: stat/podlewaczka/STATUS5 = {"StatusNET":{"Hostname":"podlewaczka","IPAddress":"192.168.100.192","Gateway":"192.168.100.100","Subnetmask":"255.255.255.0","DNSServer":"192.168.100.100","Mac":"60:01:94:85:71:30","Webserver":2,"WifiConfig":2}}
10:03:27 MQT: stat/podlewaczka/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.100.113","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_857130","MqttUser":"client","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
10:03:27 MQT: stat/podlewaczka/STATUS7 = {"StatusTIM":{"UTC":"Sat Jun 16 08:03:27 2018","Local":"Sat Jun 16 10:03:27 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":2}}
10:03:27 MQT: stat/podlewaczka/STATUS10 = {"StatusSNS":{"Time":"2018-06-16T10:03:27","DS18B20-1":{"Id":"0000054C3854","Temperature":32.13},"DS18B20-2":{"Id":"0000054CCEC4","Temperature":57.19},"DS18B20-3":{"Id":"0000054D6492","Temperature":28.00},"AM2301":{"Temperature":26.60,"Humidity":41.3},"Gas":{"Type":"CLEANAIR","ppm":0},"TempUnit":"C"}}
10:03:27 MQT: stat/podlewaczka/STATUS11 = {"StatusSTS":{"Time":"2018-06-16T10:03:27","Uptime":"0T00:01:32","POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF", "Wifi":{"AP":1, "SSId":"robert_private", "RSSI":100, "APMac":"84:16:F9:C8:85:E6"}, "DeepSleep":-1, "Heap":17152}}
```
- [WEMOS PRO ] Hardware used : 
```
10:06:10 MQT: stat/woda/STATUS = {"Status":{"Module":19,"FriendlyName":["Woda"],"Topic":"woda","ButtonTopic":"0","Power":0,"PowerOnState":0,"LedState":6,"SaveData":10,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
10:06:10 MQT: stat/woda/STATUS1 = {"StatusPRM":{"Baudrate":9600,"GroupTopic":"sonoffs","OtaUrl":"http://192.168.100.113/wemeos-d1-watter-PL","RestartReason":"Deep-Sleep Wake","Uptime":"0T00:00:29","StartupUTC":"2018-06-16T08:05:41","Sleep":0,"BootCount":257,"SaveCount":623,"SaveAddress":"F4000"}}
10:06:10 MQT: stat/woda/STATUS2 = {"StatusFWR":{"Version":"6.0.0a","BuildDateTime":"2018-06-16T00:07:10","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
10:06:10 MQT: stat/woda/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["robert_private","robert_private2"],"TelePeriod":60,"SetOption":["05408209","95818000"]}}
10:06:10 MQT: stat/woda/STATUS4 = {"StatusMEM":{"ProgramSize":473,"Free":528,"Heap":18,"ProgramFlashSize":1024,"FlashSize":16384,"FlashMode":3,"Features":["00000415","08102594","0C000001","00000490","00000000"]}}
10:06:10 MQT: stat/woda/STATUS5 = {"StatusNET":{"Hostname":"woda","IPAddress":"192.168.100.193","Gateway":"192.168.100.100","Subnetmask":"255.255.255.0","DNSServer":"192.168.100.100","Mac":"5C:CF:7F:6F:AF:A8","Webserver":2,"WifiConfig":2}}
10:06:10 MQT: stat/woda/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.100.113","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_6FAFA8","MqttUser":"client","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
10:06:10 MQT: stat/woda/STATUS7 = {"StatusTIM":{"UTC":"Sat Jun 16 08:06:10 2018","Local":"Sat Jun 16 10:06:10 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":2}}
10:06:10 MQT: stat/woda/STATUS10 = {"StatusSNS":{"Time":"2018-06-16T10:06:10","Switch1":"OFF","COUNTER":{"C1":672259},"WATER":{"CW":672.260},"Vcc":{"A0":3.934}}}
10:06:10 MQT: stat/woda/STATUS11 = {"StatusSTS":{"Time":"2018-06-16T10:06:10","Uptime":"0T00:00:29", "Wifi":{"AP":1, "SSId":"robert_private", "RSSI":100, "APMac":"84:16:F9:C8:85:E6"}, "DeepSleep":-1, "Heap":17456}}
```

reloxx13, 2018-06-17:
cannot reproduce.

wifi config 5 works just fine:
restart router and after a few sec teh devices join again.

note: wifi config 2 is WPS mode and their will never be an AP with wifi mode 2.

guess you messed up your config while testing.

write down your steps you get that problem.

> I set WifiConfig 2 Wemos, Similar problem I have in WEMOS PRO when I restart wifi module hangs no wifi AP mode
it does not hang, it started wps mode.

roblad, 2018-06-18:
Hi,

I happening not always after reboot of router, once per 3-5 restarts , there is sonoff 4ch pro. When it happened the sonoff hangs, hard reset is needed. It cause me problem because there is a watter garden system. The relay still on after hang.  In my opinion there is long time of router boot, I want to extend retraing time in sonoff maybe it will help.

pnuding, 2018-06-18:
I also use Wificonfig 5 and I have also seen situations where Sonoffs get into a funny state where they simply will not reconnect to the WiFi network until they're reset. I've had it happen on Sonoff 4ch, Basic and Pow, on recent and some months old versions of Tasmota. 
The devices aren't actually hanging - they still work just fine locally, responding to button presses and attached switches. 
They just stay disconnected from WiFi.
For now as a workaround I've changed some of my Sonoffs (the hard to reach ones) to Wificonfig 0, but it would be nice if Wificonfig 5 could actually be relied on

roblad, 2018-06-19:
Hi,

I tryied wificonfig 0, but sonof 4ch pro hangs when I restart router, it is worest than 5 in my sonoff.

closed 2018-08-19 by stale[bot]