arendst/Tasmota #3009: MQT connected but Tasmota not sending all sensor data (MQT/RSL)

highground88, 2018-06-17:
I do apologise in advance if this is a user error but I've spent a good number of days trying to fix this without much luck.

I have a WemosD1 mini as a Generic module with a BMP180 (D3 GPIO0 - I2C SDA & D4 GPIO2 I2C SCL) )and a PIR sensor (D7 GPIO13 - Switch1) attached. MQTT broker is Mosquito running on Home Assistant - this is connecting fine apparently.

My BMP180 sensor is being recognised and data is displayed on the UI, and MQT transmits are occurring in the console. As soon as I enable the Switch1 module and the device reboots, my temperature telemetry gets tagged with "RSL" (which I'm led to believe is normally if the MQTT broker is offline?) instead of MQT. The broker is still connected though, as the Switch1 activations (PIR sensor) transmit with no issues.

This can be replicated by removing the Switch1 modeule, which results in the MQT tag in front of the temperature sensor data returning.

- With 'Switch1' enabled
12:55:56 MQT: stat/roofspace/RESULT = {"POWER":"ON"}
12:55:56 MQT: stat/roofspace/POWER = ON
12:56:12 RSL: tele/roofspace/STATE = {"Time":"2018-06-17T12:56:12","Uptime":"0T14:01:04","Vcc":2.703,"Wifi":{"AP":1,"SSId":"AP","RSSI":50,"APMac":"APMAC"}}
12:56:12 RSL: tele/roofspace/SENSOR = {"Time":"2018-06-17T12:56:12","Switch1":"ON","BMP180":{"Temperature":18.4,"Pressure":1012.2},"TempUnit":"C"}

-Without 'Switch1' enabled 
13:06:08 RSL: tele/roofspace/STATE = {"Time":"2018-06-17T13:06:08","Uptime":"0T00:00:16","Vcc":2.702,"Wifi":{"AP":1,"SSId":"XXXX","RSSI":50,"APMac":"10:XXXF"}}
13:06:08 MQT: tele/roofspace/SENSOR = {"Time":"2018-06-17T13:06:08","BMP180":{"Temperature":18.6,"Pressure":1012.0},"TempUnit":"C"}

 I've found similar bug reports, and have tried using Tasmota 5.12 as well as the 2.30 ESP libraries (Initially I was using 6.0.0.a and 2.41 lib). As of now I'm out of ideas...?

Many thanks.

```
12:53:07 RSL: stat/roofspace/STATUS = {"Status":{"Module":18,"FriendlyName":["roofspace"],"Topic":"roofspace","ButtonTopic":"pirstairstop","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
12:53:07 RSL: stat/roofspace/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T13:57:59","StartupUTC":"2018-06-16T12:55:08","Sleep":0,"BootCount":23,"SaveCount":2232,"SaveAddress":"3F4000"}}
12:53:07 RSL: stat/roofspace/STATUS2 = {"StatusFWR":{"Version":"6.0.0a","BuildDateTime":"2018-06-16T22:53:15","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
12:53:07 RSL: stat/roofspace/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Skynet",""],"TelePeriod":60,"SetOption":["000A8009","55818000"]}}
12:53:07 RSL: stat/roofspace/STATUS4 = {"StatusMEM":{"ProgramSize":538,"Free":2532,"Heap":13,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":2,"Features":["00000809","0FDAE794","00000000","17B6179A","00000000"]}}
12:53:07 RSL: stat/roofspace/STATUS5 = {"StatusNET":{"Hostname":"roofspace-6940","IPAddress":"192.168.0.127","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"2C:3A:E8:XXXXX","Webserver":2,"WifiConfig":2}}
12:53:07 RSL: stat/roofspace/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.101","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_35FB1C","MqttUser":"USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
12:53:07 RSL: stat/roofspace/STATUS7 = {"StatusTIM":{"UTC":"Sun Jun 17 02:53:07 2018","Local":"Sun Jun 17 12:53:07 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 02:00:00 2018","Timezone":10,"Sunrise":"06:36","Sunset":"17:00"}}
12:53:07 RSL: stat/roofspace/STATUS10 = {"StatusSNS":{"Time":"2018-06-17T12:53:07","Switch1":"ON","BMP180":{"Temperature":18.2,"Pressure":1011.6},"TempUnit":"C"}}
12:53:07 RSL: stat/roofspace/STATUS11 = {"StatusSTS":{"Time":"2018-06-17T12:53:07","Uptime":"0T13:57:59","Vcc":2.701,"Wifi":{"AP":1,"SSId":"MyAP","RSSI":46,"APMac":"MyAPMAC"}}}
```

Jason2866, 2018-06-17:
I would check first why mqtt is not working reliable. Nothing from client side should be able to.
For testing replace the PIR with a normal switch for testing. Result?

highground88, 2018-06-17:
Hi @Jason2866 , thanks for you comment.
Changing the PIR for a switch makes no difference, as the module still detects the input. I have a push button attached to the same GPIO for testing, and used a generic arduino style PIR, and they function the same.
```
18:36:47 MQT: stat/roofspace/RESULT = {"POWER":"OFF"}
18:36:47 MQT: stat/roofspace/POWER = OFF
18:36:49 RSL: stat/roofspace/STATUS = {"Status":{"Module":18,"FriendlyName":["roofspace"],"Topic":"roofspace","ButtonTopic":"pirstairstop","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
18:36:49 RSL: stat/roofspace/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T01:13:13","StartupUTC":"2018-06-17T07:23:36","Sleep":0,"BootCount":35,"SaveCount":2351,"SaveAddress":"3F5000"}}
18:36:49 RSL: stat/roofspace/STATUS2 = {"StatusFWR":{"Version":"6.0.0a","BuildDateTime":"2018-06-16T22:53:15","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
18:36:49 RSL: stat/roofspace/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Skynet",""],"TelePeriod":60,"SetOption":["000A8009","55818000"]}}
18:36:49 RSL: stat/roofspace/STATUS4 = {"StatusMEM":{"ProgramSize":538,"Free":2532,"Heap":13,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":2,"Features":["00000809","0FDAE794","00000000","17B6179A","00000000"]}}
18:36:49 RSL: stat/roofspace/STATUS5 = {"StatusNET":{"Hostname":"roofspace-6940","IPAddress":"192.168.0.127","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"XXXXX","Webserver":2,"WifiConfig":2}}
18:36:49 RSL: stat/roofspace/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.101","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_35FB1C","MqttUser":"Lee","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
18:36:49 RSL: stat/roofspace/STATUS7 = {"StatusTIM":{"UTC":"Sun Jun 17 08:36:49 2018","Local":"Sun Jun 17 18:36:49 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 02:00:00 2018","Timezone":10,"Sunrise":"06:36","Sunset":"17:00"}}
18:36:49 RSL: stat/roofspace/STATUS10 = {"StatusSNS":{"Time":"2018-06-17T18:36:49","Switch2":"OFF","BMP180":{"Temperature":20.5,"Pressure":1014.6},"TempUnit":"C"}}
18:36:49 RSL: stat/roofspace/STATUS11 = {"StatusSTS":{"Time":"2018-06-17T18:36:49","Uptime":"0T01:13:13","Vcc":2.701,"Wifi":{"AP":1,"SSId":"Skynet","RSSI":34,"APMac":"10:13:31:D0:19:4F"}}}
18:36:54 RSL: stat/roofspace/STATUS = {"Status":{"Module":18,"FriendlyName":["roofspace"],"Topic":"roofspace","ButtonTopic":"pirstairstop","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
18:36:54 RSL: stat/roofspace/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T01:13:18","StartupUTC":"2018-06-17T07:23:36","Sleep":0,"BootCount":35,"SaveCount":2351,"SaveAddress":"3F5000"}}
18:36:54 RSL: stat/roofspace/STATUS2 = {"StatusFWR":{"Version":"6.0.0a","BuildDateTime":"2018-06-16T22:53:15","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
18:36:54 RSL: stat/roofspace/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Skynet",""],"TelePeriod":60,"SetOption":["000A8009","55818000"]}}
18:36:54 RSL: stat/roofspace/STATUS4 = {"StatusMEM":{"ProgramSize":538,"Free":2532,"Heap":13,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":2,"Features":["00000809","0FDAE794","00000000","17B6179A","00000000"]}}
18:36:54 RSL: stat/roofspace/STATUS5 = {"StatusNET":{"Hostname":"roofspace-6940","IPAddress":"192.168.0.127","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"XXXX","Webserver":2,"WifiConfig":2}}
18:36:54 RSL: stat/roofspace/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.101","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_35FB1C","MqttUser":"XXXX","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
18:36:54 RSL: stat/roofspace/STATUS7 = {"StatusTIM":{"UTC":"Sun Jun 17 08:36:54 2018","Local":"Sun Jun 17 18:36:54 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 02:00:00 2018","Timezone":10,"Sunrise":"06:36","Sunset":"17:00"}}
18:36:54 RSL: stat/roofspace/STATUS10 = {"StatusSNS":{"Time":"2018-06-17T18:36:54","Switch2":"OFF","BMP180":{"Temperature":20.5,"Pressure":1014.5},"TempUnit":"C"}}
18:36:54 RSL: stat/roofspace/STATUS11 = {"StatusSTS":{"Time":"2018-06-17T18:36:54","Uptime":"0T01:13:18","Vcc":2.703,"Wifi":{"AP":1,"SSId":"XXXX","RSSI":44,"APMac":"XXXXX"}}}
18:36:59 RSL: stat/roofspace/STATUS2 = {"StatusFWR":{"Version":"6.0.0a","BuildDateTime":"2018-06-16T22:53:15","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
18:36:59 RSL: stat/roofspace/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Skynet",""],"TelePeriod":60,"SetOption":["000A8009","55818000"]}}
18:36:59 RSL: stat/roofspace/STATUS4 = {"StatusMEM":{"ProgramSize":538,"Free":2532,"Heap":12,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":2,"Features":["00000809","0FDAE794","00000000","17B6179A","00000000"]}}
18:36:59 RSL: stat/roofspace/STATUS5 = {"StatusNET":{"Hostname":"roofspace-6940","IPAddress":"192.168.0.127","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"XXXX","Webserver":2,"WifiConfi [... cut ...]
18:37:04 RSL: stat/roofspace/STATUS10 = {"StatusSNS":{"Time":"2018-06-17T18:37:04","Switch2":"OFF","BMP180":{"Temperature":20.5,"Pressure":1014.5},"TempUnit":"C"}}
18:37:04 RSL: stat/roofspace/STATUS11 = {"StatusSTS":{"Time":"2018-06-17T18:37:04","Uptime":"0T01:13:28","Vcc":2.702,"Wifi":{"AP":1,"SSId":"AP","RSSI":48,"APMac":"10:13:31:XXX"}}}
```

I've tested two different ESP's, both do the same thing also - as soon as a Switch is introduced to the module, the temperature data gets tagged with RSL. The switch then sends the MQT message, which is recieved by the broker. (I have also put a MQTT monitor - MQTT.fx - in line with this, and verified).

Jason2866, 2018-06-17:
Free Memory is low. Compile a version with features disabled and try again.
You use not the standard memory model (4 Meg, Tasmota is 1Meg). 
This is not supported and can generate strange errors.
Build a standard version with core 2.3.0 (like you already did)

highground88, 2018-06-19:
Thanks @Jason2866 , I've fully removed the Arduino IDE and reinstall a portable nightly version with core 2.3.0, and this has indeed fixed the issue. I think I initially made the mistake of using the Wemos D1 mini board rather than the generic ESP8266 in the upload config. Such a minor mistake, such annoying consequences! The 5 second status updates from the log above was due to SonWeb interface, which I've fixed also.

Many thanks, I'll mark as closed

closed 2018-06-19 by highground88