arendst/Tasmota #3010: no wifi and sonoff use anyway

fragolinux, 2018-06-17:
hi, yesterday i had wifi issues and i had a lot of troubles letting my desk lamp to stay on... my sonoff basic with tasmota 5.14 started shutting down the lamp and led started blinking and i saw its wifi access point appear on my phone (the one used to configure tasmotized sonoffs while brand new)... is there a way to maintain the sonoff standard switch functionalities even if wifi is absent? What am i missing? I don't care about state when wifi is back on, i just want it to remain in the state it was when wifi was lost and to use the physical switch to put lamp on and off even with no wifi...

Jason2866, 2018-06-17:
Status 0 ?

fragolinux, 2018-06-17:
10:14:39 CMD: status 0
10:14:40 MQT: stat/LampadaStudio/STATUS = {"Status":{"Module":1,"FriendlyName":["LampadaStudio"],"Topic":"LampadaStudio","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":1}}
10:14:40 MQT: stat/LampadaStudio/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T17:04:11","StartupUTC":"2018-06-16T16:10:29","Sleep":0,"BootCount":250,"SaveCount":1375,"SaveAddress":"F9000"}}
10:14:40 MQT: stat/LampadaStudio/STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018-05-15T15:36:49","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
10:14:40 MQT: stat/LampadaStudio/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId":["Telecom-33102991","indebuurt2"],"TelePeriod":300,"SetOption":["55800029","55C18000"]}}
10:14:40 MQT: stat/LampadaStudio/STATUS4 = {"StatusMEM":{"ProgramSize":527,"Free":476,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":255}}
10:14:40 MQT: stat/LampadaStudio/STATUS5 = {"StatusNET":{"Hostname":"LampadaStudio-3316","IPAddress":"192.168.1.107","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"5C:CF:7F:B4:AC:F4","Webserver":2,"WifiConfig":2}}
10:14:40 MQT: stat/LampadaStudio/STATUS6 = {"StatusMQT":{"MqttHost":"secret.ddns.net","MqttPort":1883,"MqttClientMask":"GIRI_%06X","MqttClient":"GIRI_B4ACF4","MqttUser":"admin","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
10:14:40 MQT: stat/LampadaStudio/STATUS7 = {"StatusTIM":{"UTC":"Sun Jun 17 09:14:40 2018","Local":"Sun Jun 17 10:14:40 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:46","Sunset":"20:56"}}
10:14:40 MQT: stat/LampadaStudio/STATUS10 = {"StatusSNS":{"Time":"2018-06-17T10:14:40"}}
10:14:40 MQT: stat/LampadaStudio/STATUS11 = {"StatusSTS":{"Time":"2018-06-17T10:14:40","Uptime":"0T17:04:11","Vcc":3.143,"POWER1":"OFF","Wifi":{"AP":1,"SSId":"Telecom-33102991","RSSI":72,"APMac":"C4:EA:1D:E4:DE:8F"}}}

Jason2866, 2018-06-17:
So far so good :-)
You have to use a non standard mqtt library to get a responsive sonoff if wifi or mqtt is down.
This change is done in user_config.h
Comment
#define MQTT_LIBRARY_TYPE      MQTT_PUBSUBCLIENT
and uncomment ONE of the alternatives mqtt library. 

`/*-------------------------------------------------------------------------------------------*\
 * Select ONE of possible three MQTT library types below
\*-------------------------------------------------------------------------------------------*/
  // Default MQTT driver for both non-TLS and TLS connections. Blocks network if MQTT server is unavailable.
#define MQTT_LIBRARY_TYPE      MQTT_PUBSUBCLIENT   // Use PubSubClient library
  // Alternative MQTT driver does not block network when MQTT server is unavailable. No TLS support
//#define MQTT_LIBRARY_TYPE      MQTT_TASMOTAMQTT    // Use TasmotaMqtt library (+4k4 code, +4k mem) - non-TLS only
  // Alternative MQTT driver does not block network when MQTT server is unavailable. No TLS support
//#define MQTT_LIBRARY_TYPE      MQTT_ESPMQTTARDUINO // Use (patched) esp-mqtt-arduino library (+4k8 code, +4k mem) - non-TLS only

`

fragolinux, 2018-06-17:
i use a standard bin firmware file from the releases page, so the default lib is not ok?
my mqtt is remote on a public vps server, but i'll soon revert to a local one... so the problem is given by the unreachable mqtt server?
thanks for the help :)

Jason2866, 2018-06-17:
The standard lib is ok. But if mqtt enabled and broker is not reachable it is laggy.
Yes the problem is the unreachable mqtt server.

fragolinux, 2018-06-17:
understood, thanks, i'll recompile my own firmware, then :)

closed 2018-06-17 by fragolinux