arendst/Tasmota #3011: Sudden MQTT issues?

chris-008, 2018-06-17:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ ] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x] Hardware used : Sonoff BASIC + TH
- [x ] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
14:46:51 CMD: status 0
14:46:51 RSL: Group 0, Index 1, Command STATUS, Data 0
14:46:51 RSL: stat/sonoff1/STATUS = {"Status":{"Module":1, "FriendlyName":"living links", "Topic":"sonoff1", "ButtonTopic":"0", "Power":1, "PowerOnState":3, "LedState":1, "SaveData":1, "SaveState":1, "ButtonRetain":0, "PowerRetain":0}}
14:46:51 RSL: stat/sonoff1/STATUS1 = {"StatusPRM":{"Baudrate":115200, "GroupTopic":"sonoffs", "OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin", "Uptime":1, "Sleep":50, "BootCount":332, "SaveCount":1297, "SaveAddress":"F6000"}}
14:46:51 RSL: stat/sonoff1/STATUS2 = {"StatusFWR":{"Version":"5.8.0", "BuildDateTime":"2017-10-05T21:43:16", "Boot":6, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
14:46:51 RSL: stat/sonoff1/STATUS3 = {"StatusLOG":{"SerialLog":4, "WebLog":3, "SysLog":4, "LogHost":"domus1", "LogPort":514, "SSId1":"WiFi-008-2.4GHz", "SSId2":"WiFi-008-5.0GHz", "TelePeriod":0, "SetOption":"55800009"}}
14:46:51 RSL: stat/sonoff1/STATUS4 = {"StatusMEM":{"ProgramSize":469, "Free":532, "Heap":25, "ProgramFlashSize":1024, "FlashSize":1024, "FlashMode":3}}
14:46:51 RSL: stat/sonoff1/STATUS5 = {"StatusNET":{"Hostname":"sonoff1-3228", "IPAddress":"192.168.1.6", "Gateway":"192.168.1.1", "Subnetmask":"255.255.255.0", "DNSServer":"192.168.1.1", "Mac":"5C:CF:7F:BD:2C:9C", "Webserver":2, "WifiConfig":1}}
14:46:51 RSL: stat/sonoff1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.62", "MqttPort":1883, "MqttClientMask":"DVES__%06X", "MqttClient":"DVES__BD2C9C", "MqttUser":"pi", "MAX_PACKET_SIZE":512, "KEEPALIVE":15}}
14:46:51 RSL: stat/sonoff1/STATUS7 = {"StatusTIM":{"UTC":"Sun Jun 17 13:46:51 2018", "Local":"Sun Jun 17 14:46:51 2018", "StartDST":"Sun Mar 25 02:00:00 2018", "EndDST":"Sun Oct 28 03:00:00 2018", "Timezone":1}}
14:46:51 RSL: stat/sonoff1/STATUS10 = {"StatusSNS":{"Time":"2018-06-17T14:46:51"}}
14:46:51 RSL: stat/sonoff1/STATUS11 = {"StatusSTS":{"Time":"2018-06-17T14:46:51", "Uptime":1, "Vcc":3.187, "POWER":"ON", "Wifi":{"AP":1, "SSId":"WiFi-008-2.4GHz", "RSSI":98, "APMac":"B8:D9:4D:DB:FC:FE"}}}
14:46:59 MQT: Attempting connection...
14:47:00 DNS: Query done. MQTT services found 0
14:47:00 MQT: Connect failed to 192.168.1.62:1883, rc -2. Retry in 10 sec

Hi guys,

After a recent update of my additional Sonoff accessories, I am having problems connecting to the MQTT host.

Over the last couple of weeks I have done next actions:

Reflash Sonoff firmware to 5.10.0 and 5.13.0
Completely  upgraded the Pi (as MQTT broker) firmware
New Mosquitto installation (+ homebridge since I am using the iOS Home app)
Reset to factory settings of my router (due to the rc -2 error indicating network issues) 
Lowered firewall level/ temp.switch off

Unfortunately nothing helped to get it up and running again.

The Sonoff module just wont connect to the MQTT host server (192.168.1.62) anymore…

Manual controls through the web UI console work fine though.
I have checked nearly all related rc-2 issues from the issues section but nothing worked so far.

Here are a few additionalstatus reports:

14:08:45 DNS: Query done. MQTT services found 0
14:08:45 MQT: Connect failed to 192.168.1.62:1883, rc -2. Retry in 10 sec
14:08:50 CMD: status 1
14:08:50 RSL: Group 0, Index 1, Command STATUS, Data 1
14:08:50 RSL: stat/sonoff1/STATUS1 = {"StatusPRM":{"Baudrate":115200, "GroupTopic":"sonoffs", "OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin", "Uptime":1, "Sleep":50, "BootCount":332, "SaveCount":1295, "SaveAddress":"F8000"}}
14:08:56 MQT: Attempting connection...
14:08:57 DNS: Query done. MQTT services found 0
14:08:57 MQT: Connect failed to 192.168.1.62:1883, rc -2. Retry in 10 sec
14:09:08 MQT: Attempting connection...
14:09:09 DNS: Query done. MQTT services found 0
14:09:09 MQT: Connect failed to 192.168.1.62:1883, rc -2. Retry in 10 sec
14:09:15 CMD: status 5
14:09:15 RSL: Group 0, Index 1, Command STATUS, Data 5
14:09:15 RSL: stat/sonoff1/STATUS5 = {"StatusNET":{"Hostname":"sonoff1-3228", "IPAddress":"192.168.1.6", "Gateway":"192.168.1.1", "Subnetmask":"255.255.255.0", "DNSServer":"192.168.1.1", "Mac":"5C:CF:7F:BD:2C:9C", "Webserver":2, "WifiConfig":1}}
14:09:20 CMD: status 6
14:09:20 RSL: Group 0, Index 1, Command STATUS, Data 6
14:09:20 RSL: stat/sonoff1/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.62", "MqttPort":1883, "MqttClientMask":"DVES__%06X", "MqttClient":"DVES__BD2C9C", "MqttUser":"pi", "MAX_PACKET_SIZE":512, "KEEPALIVE":15}}
14:09:21 MQT: Attempting connection...
14:09:22 DNS: Query done. MQTT services found 0
14:09:22 MQT: Connect failed to 192.168.1.62:1883, rc -2. Retry in 10 sec

Here is a screenshot of the Sonoff (web UI) MQTT settings:
<img width="364" alt="schermafbeelding 2018-06-17 om 15 51 35" src="https://user-images.githubusercontent.com/32146240/41508526-8f51b4f4-7246-11e8-8d44-c687aaec3d17.png">

Frogmore42, 2018-06-17:
Are you sure that is still the correct IP address for the MQTT broker?
What do the logs of the broker say?
Do you have unique MQTT client names for your devices?

chris-008, 2018-06-17:
yes, IP is  verified through the router.

pi@orangepizero:~$ systemctl status mosquitto.service
mosquitto.service - LSB: mosquitto MQTT v3.1 message broker
   Loaded: loaded (/etc/init.d/mosquitto; generated; vendor preset: enabled)
   Active: active (exited) since Sun 2018-06-17 16:29:44 CEST; 2h 11min ago
     Docs: man:systemd-sysv-generator(8)
    Tasks: 0 (limit: 4915)
   CGroup: /system.slice/mosquitto.service

Jun 17 16:29:43 orangepizero systemd[1]: Starting LSB: mosquitto MQTT v3.1 message broker..
Jun 17 16:29:44 orangepizero mosquitto[474]: Starting network daemon:: mosquitto.
Jun 17 16:29:44 orangepizero systemd[1]: Started LSB: mosquitto MQTT v3.1 message broker.

All have unique names like sonoff 1, sonoff2, ....

Jason2866, 2018-06-17:
I think broker isnt working or isnt reachable.

chris-008, 2018-06-17:
my guessing too to be honest...

chris-008, 2018-06-17:
Verification that the MQTT server is working (by sending messages as subscriber, publisher) was succesfull though.

chris-008, 2018-06-17:
I think it went wrong when I renamed all my modules MQTT topic names. All were renamed using the web UI as well as in the config.json file. Could it be that somewhere in the flash memory the old name is still stored and therefore creates this?

Frogmore42, 2018-06-17:
What does the mqtt log on the broker say?
Turn off all of the modules and then turn them on, one at a time.  Look at the broker log file and see if you can see any connection attempt.

stefanbode, 2018-06-19:
I can confirm, that MQTT is still working. Please set the user and password again. rc -4 is also an indicator that the password is wrong. You should see more info in your broker log.

chris-008, 2018-06-19:
I have reset th esername and password but it didnt help either.
Not sure where you got that rc-4 though?

Broker seems to be doing its job:

pi@orangepizero:~$ sudo mosquitto -v
1529440334: mosquitto version 1.5 starting
1529440334: Using default config.
1529440334: Opening ipv4 listen socket on port 1883.
1529440334: Opening ipv6 listen socket on port 1883.
1529440338: New connection from 192.168.1.9 on port 1883.
1529440338: New client connected from 192.168.1.9 as DVES_BD2CCA (c1, k15, u'pi').
1529440338: Will message specified (7 bytes) (r1, q1).
1529440338: 	tele/sonoff7/LWT
1529440338: Sending CONNACK to DVES_BD2CCA (0, 0)
1529440338: Received PUBLISH from DVES_BD2CCA (d0, q0, r1, m0, 'tele/sonoff7/LWT', ... (6 bytes))
1529440338: Received PUBLISH from DVES_BD2CCA (d0, q0, r0, m0, 'cmnd/sonoff7/POWER', ... (0 bytes))
1529440338: Received SUBSCRIBE from DVES_BD2CCA
1529440338: 	cmnd/sonoff7/# (QoS 0)
1529440338: DVES_BD2CCA 0 cmnd/sonoff7/#
1529440338: Sending SUBACK to DVES_BD2CCA
1529440338: Received SUBSCRIBE from DVES_BD2CCA
1529440338: 	cmnd/sonoffs/# (QoS 0)
1529440338: DVES_BD2CCA 0 cmnd/sonoffs/#
1529440338: Sending SUBACK to DVES_BD2CCA
1529440339: Received SUBSCRIBE from DVES_BD2CCA
1529440339: 	cmnd/DVES_BD2CCA/# (QoS 0)
1529440339: DVES_BD2CCA 0 cmnd/DVES_BD2CCA/#
1529440339: Sending SUBACK to DVES_BD2CCA

Frogmore42, 2018-06-19:
Now that you know where the logs are, you will need to cross-correlate the logs from Tasmota and MQTT to see how they compare.  There are online unix timestamp to datetime converters that will make it possible to compare the times and see what is going on.

If the broker log shows a complete error free client connection and subscription to all topics and the Tasmota log shows an error, that would likely indicate some sort of packet loss.

You are using an, at least somewhat, old version of the code, so you should probably update to the latest.  You also did not indicate if you have more than one and if they are all not working or only one.

chris-008, 2018-06-21:
Hi Frogmore, thanks again for your help.
I will look into that cross reference logging.

I have a total of 12 devices, majority still on 5.8 or 5.10, one on 5.23(still struggling with the OTA function but that’s a different subject) None of the Sonoff devices can be controllled through the home App on IOS, except for the nest thermostat and apple tv(acting as a hub for external connection)

Frogmore42, 2018-06-21:
You have a very basic failure to communicate problem here.  The Tasmota log you shared indicated it was unable to connect to your broker.  There are only a few sets of reasons for that.

You also shared a log from the broker that indicated there was a connection.  That seems to indicate at least that device is somewhat configured.

So, if I were troubleshooting this, I would.  Unplug every mqtt device.  Restart the MQTT broker.  Verify the broker is up by looking at its log.  Turn on one Tasmota device.  Look at its log.  Look at the MQTT broker log.  If there is an error in the Tasmota log, but there is a valid connection in the MQTT log, that would be wierd.  So, make sure you are running current Tasmota version, with core 2.3.0, which seems more stable still.

If both logs indicate success, that means one device can work correctly.  Turn on the next device and see what happens.

chris-008, 2018-06-21:
So all modules are disconnected except for one , which I updated to 5.14.
Broker is running fine, tasmota log also without errors.
I Forced mosquitto to enable/restart so MQT is now connected. 

[image]

[image]

However, still no action when pressing the toggle buttons in iOS Home app.
I have re-added Homebridge and even deleted the persistent folder but still no succes.

chris-008, 2018-06-21:
Weird....seems that activating the broker log (sudo mosquitto -v) triggers the MQT to connect, not the enable/start command.
While connected with homebridge running in systemd mode, now all is working :-)
Still have to find out though what causes this cause most likely it will go back to a non functioning state most likely after a reboot of the pi / broker.
Anyways, it seems more of an installation/ systemd (autoboot) problem.

Frogmore42, 2018-06-22:
Well, it certainly seems like the broker is not starting fully/properly when you think it is.  I have found Tasmota to be correct when it says it can't connect.

chris-008, 2018-06-23:
thanks for the advices @Frogmore42  for showing me the logging and trial/error procedure!

Frogmore42, 2018-06-23:
@chris-008 that is not called trial and error, that is called fault isolation 😄 
https://en.wikipedia.org/wiki/Fault_detection_and_isolation

closed 2018-06-24 by chris-008