arendst/Tasmota #3012: Loghost in network 1.0.0.0/8 not possible

MatzeJoerling, 2018-06-17:
"Loghost" command with Ip inside 1.0.0.0/8 results in empty Loghost.

Also, make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x] Searched the problem in issues and in the wiki
- [x] Hardware used : WemosD1 mini
- [x] Provide the output of command ``status 0`` :
```
15:47:44 MQT: stat/teufel/STATUS = {"Status":{"Module":3,"FriendlyName":["teufel","Device2"],"Topic":"teufel","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
15:47:44 MQT: stat/teufel/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"group","OtaUrl":"http://192.168.10.1/tasmota/git/sonoff-minimal.bin","RestartReason":"Software/System restart","Uptime":"0T00:25:03","StartupUTC":"2018-06-17T14:22:41","Sleep":0,"BootCount":46,"SaveCount":156,"SaveAddress":"F8000"}}
15:47:44 MQT: stat/teufel/STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018-05-15T15:29:54","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
15:47:44 MQT: stat/teufel/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["X",""],"TelePeriod":300,"SetOption":["00000009","55818000"]}}
15:47:44 MQT: stat/teufel/STATUS4 = {"StatusMEM":{"ProgramSize":526,"Free":476,"Heap":18,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
15:47:44 MQT: stat/teufel/STATUS5 = {"StatusNET":{"Hostname":"teufel-2076","IPAddress":"192.168.2.124","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.3.19","Mac":"5C:CF:7F:A0:08:1C","Webserver":2,"WifiConfig":2}}
15:47:44 MQT: stat/teufel/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.3.34","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_A0081C","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
15:47:44 MQT: stat/teufel/STATUS7 = {"StatusTIM":{"UTC":"Sun Jun 17 14:47:44 2018","Local":"Sun Jun 17 15:47:44 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:46","Sunset":"20:56"}}
15:47:44 MQT: stat/teufel/STATUS10 = {"StatusSNS":{"Time":"2018-06-17T15:47:44"}}
15:47:44 MQT: stat/teufel/STATUS11 = {"StatusSTS":{"Time":"2018-06-17T15:47:44","Uptime":"0T00:25:03","Vcc":3.190,"POWER1":"OFF","POWER2":"OFF","Dimmer":20,"Color":"333333","HSBColor":"0,0,20","Channel":[20,20,20],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"X","RSSI":54,"APMac":"GG:C8:82:AB:F2:A0"}}}

```

**To Reproduce**
enter: loghost 1.1.1.1

**Expected behavior**

15:52:57 CMD: loghost 1.1.1.1
15:52:57 MQT: stat/teufel/RESULT = {"LogHost":""}

this is what i expect, also from loghost 1.1.1.1
15:53:04 CMD: loghost 2.2.2.2
15:53:04 MQT: stat/teufel/RESULT = {"LogHost":"2.2.2.2"}

**Additional context**
I think no one is running the loghost in Cloudflare, China or Korea Telekom networks, so this bug is likely cosmetic. i hit it while writing a script to configure sonoff via bash / mosquitto_sub ...

fragolinux, 2018-06-17:
why do you use that network for local lan? the allowed networks for private lans are
192.168.0.0/16
172.16.0.0/12
10.0.0.0/8

MatzeJoerling, 2018-06-17:
Honestly, where did i say that i use that network? But someone could.
if you ask for the owner of the network it is Cloudflare, China Telekom or Korea Telekom - who tells you that they dont run tasmota with official IPs and have official syslog ip???

As i said, "only cosmetic"!

curzon01, 2018-06-21:
1.0.0.0/8 was reserved and has been assigned by IANA to APNIC for use for as public unicast space for further address allocations and assignments.

e. g. 1.1.1.1 is now live (it's a public DNS resolver) and there are some reasons not using it (e. g. see [The Network Engineer’s Favorite Dummy Address](https://hainc.com/2018/03/29/blog/the-network-engineer-s-favorite-dummy-address-1-1-1-1-is-now-live-on-the-internet)).

Anyway, it looks like a bug, because you can set 1.1.1.1 using Web interface under "Configuration->Configure Logging" but neither with any of the command possibilities (MQTT, http, Web interface "Console")

ascillato2, 2018-07-10:
Hi,

Seems that your question has been addressed. Closing this issue. Please, reopen if needed. Thanks.

MatzeJoerling, 2018-07-10:
Nope, sorry, this has not been addressed. Also in the latest 6.1.0 i cannot set loghost to 1.1.1.1!
But what i can is set Loghost to 260.300.400.1! there is something wrong with the "check_that_ip_address_code"! 

```
stat/sonoff/STATUS {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":3,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
stat/sonoff/STATUS1 {"StatusPRM":{"Baudrate":115200,"GroupTopic":"hauz","OtaUrl":"http://ota.domain.local/tasmota/git/6.1.0/sonoff-minimal.bin","RestartReason":"Software/System restart","Uptime":"0T00:04:19","StartupUTC":"2018-07-10T12:35:15","Sleep":0,"BootCount":54,"SaveCount":200,"SaveAddress":"F9000"}}
stat/sonoff/STATUS2 {"StatusFWR":{"Version":"6.1.0","BuildDateTime":"2018-07-06T21:01:47","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
stat/sonoff/STATUS3 {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":2,"LogHost":"","LogPort":514,"SSId":["xxxxxx",""],"TelePeriod":300,"SetOption":["00000009","55818000"]}}
stat/sonoff/STATUS4 {"StatusMEM":{"ProgramSize":535,"Free":468,"Heap":17,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FDAE794","0C000000","23B6179E","00000000"]}}
stat/sonoff/STATUS5 {"StatusNET":{"Hostname":"sonoff-2985","IPAddress":"192.168.4.101","Gateway":"192.168.4.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.4.1","Mac":"68:C6:3A:9F:AB:A9","Webserver":2,"WifiConfig":5}}
stat/sonoff/STATUS6 {"StatusMQT":{"MqttHost":"192.168.3.34","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9FABA9","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
stat/sonoff/STATUS7 {"StatusTIM":{"UTC":"Tue Jul 10 12:39:34 2018","Local":"Tue Jul 10 13:39:34 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:58","Sunset":"20:53"}}
stat/sonoff/STATUS10 {"StatusSNS":{"Time":"2018-07-10T13:39:34","Switch1":"OFF","Switch2":"OFF","BMP280":{"Temperature":27.4,"Pressure":1003.5},"TempUnit":"C"}}
stat/sonoff/STATUS11 {"StatusSTS":{"Time":"2018-07-10T13:39:34","Uptime":"0T00:04:19","Vcc":2.744,"POWER":"ON","Dimmer":25,"Color":"3F3F00","HSBColor":"60,100,25","Channel":[25,25,0],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"OFF","Wifi":{"AP":1,"SSId":"xxxxx","RSSI":42,"APMac":"0E:EC:DA:1D:F2:A9"}}}
```

here some tests with different IPs.
```
cmnd/sonoff/LogHost (null)
stat/sonoff/RESULT {"LogHost":"192.168.2.5"}
cmnd/sonoff/LogHost 1.1.1.1
stat/sonoff/RESULT {"LogHost":""}
cmnd/sonoff/LogHost 1.0.0.1
cmnd/sonoff/LogHost 999.300.400.1
stat/sonoff/RESULT {"LogHost":"999.300.400.1"}
cmnd/sonoff/LogHost 260.300.400.1
stat/sonoff/RESULT {"LogHost":"260.300.400.1"}
cmnd/sonoff/LogHost 1.1.1.1
stat/sonoff/RESULT {"LogHost":""}
```

this is not a problem i have! i know how to enter my syslog host! 
But is it a bug in the code checking the IP address!

regards Matze

curzon01, 2018-07-11:
A workaround is using Web GUI - Configuration->Configure Logging. Didn't you try that?

ascillato2, 2018-07-11:
Reopening due to the problem remains.

MatzeJoerling, 2018-07-11:
@ascillato2 thanks for unterstanding. 
many exploids are from bad checked vadiables, that leads to buffer overflows which leads to hacked devices. https://en.wikipedia.org/wiki/Buffer_overflow

looks like even this is possible:
```
18:35:03 CMD: loghost AAAA.BBB.CCC.DDD
18:35:03 MQT: stat/sonoff/RESULT = {"LogHost":"AAAA.BBB.CCC.DDD"}
18:36:35 CMD: loghost 11111111111111.22.22.222
18:36:35 MQT: stat/sonoff/RESULT = {"LogHost":"11111111111111.22.22.222"}
```
what else, does it overwrite some stuff stored behind loghost in ram/flash?

dont want a trillion tasmotas run a ddos against electric companys because they get hacked and switch on all the billion bulbs at night... 

@curzon01 you are a programmer also, you should also know that. Und ich hab immer gesagt das das nicht mein Problem in meinem Netz ist (da nutze ich RFC1918 IPs) sondern das das ein Programierproblem aka BUG ist. 

tnx M.

arendst, 2018-07-11:
@MatzeJoerling the only problem you have is that the loghost won't work with invalid ip addresses. buffer overflow is not an issue as you have up to 32 characters to invent a nice ip address or hostname.

Non issue.

closed 2018-07-13 by ascillato2