arendst/Tasmota #3014: RF Bridge Limitation to 24bit Codes

haegar33, 2018-06-18:
I understand that my Sonoff RF Bridge with Tasmota 5.14.0 is not able to decode RF Codes other than 24bit length? I got a very old set of RF power plugs (Company REV) which works fine. I got just now a Brennenstuhl Comfort Line power plug set where I cannot learn any rfkey. Actually in the console I see different codes every time I press the same key on the remote control.
Questions:
- is there a list of compatible RF power plug sets which works with Tasmota?
- can I solve this issue with an update to V6.0 and the additional new efm8bb1 firmware?

Jason2866, 2018-06-19:
With Tasmota and original efm8bb1 software the supported rf power plugs are the same as with ITEAD firmware.
Just try the efm8bb1 portisch firmware. For me it works better for the code 2262.
The actual build supports not really more codes than the original but has options for doing that.
Not an ready solution out of the box

haegar33, 2018-06-21:
well now I have a chicken and egg problem as my bridge has now Tasmota and no longer the original ITEAD firmware, so no testing which power plugs are compatible. Actually it would be nice if the Wiki would contain such a list based on the experience of users. Brennenstuhl plugs do not work.

wongnam, 2018-06-22:
In my experience, i will backup the the factory original FW before i flash it with 3rd party FW.

Jason2866, 2018-06-22:
@haegar33 
TASMOTA doesnt change anything in compability which devices are supported!
Brennenstuhl may work if you use the EFM Firmware from Portisch (now included in development Tasmota). You have to flash this firmware in a extra step!

ascillato2, 2018-07-10:
Hi,

Seems that your question has been addressed. Closing this issue. Please, reopen if needed. Thanks.

closed 2018-07-10 by ascillato2

sroettgermann, 2018-09-23:
I'm currently trying to run the brennestuhl device in combination with the RF bridge and the Portisch firmware. But it gave me no results. I'm not able to receive any code. :(

wongnam, 2018-09-23:
Please note that it only support RF 433Mhz not 315Mhz.

Jason2866, 2018-09-23:
@sroettgermann  look at https://github.com/Portisch/RF-Bridge-EFM8BB1 and try it with
Bucket sniffing using command 0xB1