arendst/Tasmota #3016: Alexa not finding RF Bridge with Hue Emulation

law1964, 2018-06-18:
The RF Bridge with Hue Emulation activated is not discovered by Alexa. This appears to be an issue similar to the problem that was fixed with #854. Strangely enough, if I change the module configuration to Sonoff Basic, Alexa is able to find one single Sonoff device. If I switch the module configuration back to Sonoff Bridge, the Sonoff device stays in Alexa, but it can not find more devices.

I have two other Sonoff devices (4CH and 4CH PRO) that use the Hue Emulation and they both work properly. It is only with the RF Bridge that I am having this problem.

- [x] Searched the problem in issues and in the wiki
- [x] Hardware used : Sonoff RFBridge (I believe it is the second version)
- [x] Provide the output of command ``status 0`` :
```
20:52:22 CMD: status 0
20:52:22 RSL: stat/sonoff/STATUS = {"Status":{"Module":25,"FriendlyName":["Sonoff"],"Topic":"sonoff","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
20:52:22 RSL: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":19200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:19:09","StartupUTC":"2018-06-18T19:33:13","Sleep":0,"BootCount":7,"SaveCount":31,"SaveAddress":"F5000"}}
20:52:22 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018-05-15T15:29:54","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
20:52:22 RSL: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["neuronet",""],"TelePeriod":300,"SetOption":["00008009","55C18000"]}}
20:52:22 RSL: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":526,"Free":476,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
20:52:22 RSL: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"sonoff-5707","IPAddress":"192.168.20.5","Gateway":"192.168.20.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.20.1","Mac":"DC:4F:22:82:56:4B","Webserver":2,"WifiConfig":3}}
20:52:22 RSL: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_82564B","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
20:52:22 RSL: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Mon Jun 18 19:52:22 2018","Local":"Mon Jun 18 20:52:22 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:46","Sunset":"20:56"}}
20:52:22 RSL: stat/sonoff/STATUS10 = {"StatusSNS":{"Time":"2018-06-18T20:52:22"}}
20:52:22 RSL: stat/sonoff/STATUS11 = {"StatusSTS":{"Time":"2018-06-18T20:52:22","Uptime":"0T00:19:09","Vcc":3.208,"Wifi":{"AP":1,"SSId":"neuronet","RSSI":94,"APMac":"14:CC:20:B8:F2:85"}}}
```

**To Reproduce**
Flash the RF Bridge with tasmota 5.14.  Configure the module as Sonoff Bridge. Turn on Hue Emulation. Try to "Add Device" in the Smart Home section of the Alexa app on your phone.

**Expected behavior**
Multiple devices should be discovered by the Alexa app.

**Screenshots**
NA

reloxx13, 2018-06-18:
their is no switch to control with alexa.
{"StatusSNS":{"Time":"2018-06-18T20:52:22"}}

law1964, 2018-06-19:
My sonoff basic has the same StatusSNS, but I am able to control it with Alexa:

stat/sonoff-tree/STATUS10 = {"StatusSNS":{"Time":"2018-06-19T12:04:08"}}

Jason2866, 2018-06-19:
Just an hypothetic question. If Alexa would find the RF Bridge what should be done?

law1964, 2018-06-19:
My blinds can be controlled by RF at 433MHz.  I would use Alexa to control (over wifi) one of the switches on the RF bridge. When I ask Alexa to "turn on blinds", Alexa would activate the switch and the RF bridge would send an RF signal to the blinds to control them.

I was already able to do this with Ewelink, but I would rather do the control without using an external server. That's the reason why I flashed the RF Bridge with tasmota.

reloxx13, 2018-06-19:
i have to correct, u r right, statusSNS does not show the relais, only the switches.

but still, their needs to be a relais to get an entry for alexa.

sonof basic template got one, sonoff 4CH template got 4 by default set to the GPIOs, thats why they show up.

Jason2866, 2018-06-19:
Just define a virtual relay on a free GPiO pin from the RF Bridge.
Now you can control this relay via Alexa.
With a rule it should be possible to send something over RF...

Jason2866, 2018-06-19:
If you need 2 different commands, define 2 relays on the rf bridge

law1964, 2018-06-19:
@reloxx13 and @Jason2866 Thank you for the help. You are both right. I was able to get Alexa to find two relays that I linked to GPIOs in the module configuration. However, there are already 16 pushbutton switches defined (see below):
[image]

I believe, based on the information that you have given me, that these 16 pushbuttons need to be linked to virtual relays to make them work with Alexa. The reason why I would want to do this is because they are already pre-configured as momentary switches. For example, I am able to click on pushbutton 1 which pulses a pre-recorded RF Key to my blinds. So the question is how to do I link the pushbuttons with virtual relays?

BTW,, I only need one RF Key. The same RF Key command activates the blind, stops it, and makes it reverse direction (much like the remote control for a garage door).

Jason2866, 2018-06-20:
Read how rules work. This is the way it will work.
Trigger is power on / off from the virtual relay and action is sending command rfkeyx
https://github.com/arendst/Sonoff-Tasmota/wiki/Rules

law1964, 2018-06-20:
Thank you. This will resolve the problem nicely.

Is the following entry correct for the user_config_override.h file?:

#undef  USE_RULES
#define USE_RULES         1

I am not certain what value to use to enable the USE_RULES. I am assuming 1= enable.
I have already enabled the build flag in platform.ini to use the override.h file.

ascillato, 2018-06-20:
> Although I can define a rule now, I don't have access to some of the commands. For example, when I enter mem1 or var1 on the Console

`MEM `and `VAR `commands were added to Tasmota after v5.14.0

You need to update to a newer version to use that.

law1964, 2018-06-20:
I thought the latest release is v5.14.0. Where would I be able to get a newer version?
Also, I am only able to use one rule. If I try to define a rule2, it overwrites rule1.

23:41:53 CMD: rule1
23:41:53 RSL: RESULT = {"Rule":"ON","Once":"OFF","Rules":"on power1#state=1 do rfkey1 endon"}
23:41:56 CMD: rule2
23:41:56 RSL: RESULT = {"Rule":"ON","Once":"OFF","Rules":"on power1#state=1 do rfkey1 endon"}

Should it not let me define up to 3 rules?

ascillato, 2018-06-20:
Yes, exactly. All those commands are on v6.0.0. Need more testing for Theo to make a new release.

If you don't know/want to compile the new Tasmota Source Code, you can use for testings the ones precompiled v6.0.0 Tasmota firmwares at:

- https://github.com/reloxx13/Sonoff-Tasmota-Modified/releases
- https://github.com/ascillato/Sonoff-Tasmota_KNX/releases

law1964, 2018-06-21:
Perfect. I was able to compile v6.0.0. In fact, I had to do so because I needed to enable USER_RULES to be able to define more than one rule and for access to rule-related commands like mem and var.

Now, in addition to being able to control my blinds with Alexa, I can implement a physical security system that uses reed switches to send out an RF signal when triggered.

Is there a list of new features for v6.0.0 that can be shared?  I found the following in the _releasenotes.ino file, but it doesn't seem like enough content to be a major release:

 * 6.0.0a
 * Add CRC to Settings making future upgrades more fail-safe
 * Remove version 3, 4 and pre 5.2 settings auto-upgrade. See https://github.com/arendst/Sonoff-Tasmota/wiki/Upgrade#migration-path
 * Change default CFG_HOLDER from 0x20161209 to 4617 (=0x1209) - no impact on default upgrades

ascillato, 2018-06-21:
Hi

Everything is in the release notes, but you have to see from 5.14.0 and up. You should read also 5.14.0a, 5.14.0b, etc

All changes are:

```
* 6.0.0a
 * KNX: Support for DS18S20 Temperature sensor
 * KNX: Fixed Bug when doing reply of sensors values
 * Add CRC to Settings making future upgrades more fail-safe
 * Add support for uploading Sonoff Bridge firmware found in tools/fw_efm8bb1 folder build by Portisch using Web Gui File Upload (#2886)
 * Add support for I2C temperature sensor LM75AD (#2909)
 * Add command RfRaw to control Portisch firmware features
 * Remove version 3, 4 and pre 5.2 settings auto-upgrade. See https://github.com/arendst/Sonoff-Tasmota/wiki/Upgrade#migration-path
 * Change default CFG_HOLDER from 0x20161209 to 4617 (=0x1209) - no impact on default upgrades
 * Fix Pzem004T checksum error
 *
 * 5.14.0b
 * Add Console Commands to send KNX Commands
   usage: KnxTx_Cmnd[slot] command
   where [slot] is any of the 5 slots on the KNX Menu and command is 0 or 1
   example: KnxTx_Cmnd1 0
 * Add Console Commands to send KNX Values
   usage: KnxTx_Val[slot] value
   where [slot] is any of the 5 slots on the KNX Menu and value is a number
   example: KnxTx_Val1 35
 * Add Slots on the KNX Web Menu to select Group Addess to send data from console commands
 * Add Events to trigger rules when a command is received from KNX
   usage on rules as: event#KnxRx_Cmnd[slot]
   where [slot] is any of the 5 slots on the KNX Menu
   example: rule on event#KnxRx_Cmnd1 do VAR1 %value% endon
   (where %value% can be 0 or 1)
 * Add Events to trigger rules when received read requests from KNX
   usage on rules as: event#KnxRx_Req[slot]
   where [slot] is any of the 5 slots on the KNX Menu
   example: rule on event#KnxRx_Req1 do KnxTx_Val1 35 endon
 * Add Slots on the KNX Web Menu to select Group Addess to receive data to trigger rules
 * Add two rule sets of 511 characters using commands rule1, rule2 and rule3
 * Add Ukranian language file
 * Add rule support for IrReceive and RfReceive (#2758)
 * Add command WebSend [<host>(:<port>,<user>:<password>)] <command> (#2821)
 * Add source information to command execution to be shown with logging option 3 (#2843)
 * Fix some Pow R2 and S31 checksum errors (#1907)
 * Fix Hardware Watchdog restart when using event command (#2853)
 *
 * 5.14.0a
 * Add feature information to Status 4
 * Add tools folder with python script decode-status.py for decoding some status fields like SetOption and Features
 * Add Eastron SDM630 energy meter by Gennaro Tortone (#2735)
 * Add KNX communication enhancement by Adrian Scillato (#2742)
 * Add KNX energy data by Adrian Scillato (#2750)
 * Add python script fw-server.py in tools folder to create a simple OTA server by Gennaro Tortone (#2759)
 * Add rules %mem1% to %mem5% variable names storing data in flash (#2780)
 * Add rules test on %varx% or %memx% (#2780)
 * Add optional token %id% substituting the unique MAC address to fulltopic by Michael Graf (#2794)
 * Fix display selection of un-available GPIO options in Module Configuration webpage (#2718)
 * Fix timer re-trigger within one minute after restart (#2744)
 * Fix IRSend not accepting data value of 0 by David Conran (#2751)
 * Fix vars on rules by Adrian Scillato (#2769)
 * Fix bug in KNX menu by Adrian Scillato (#2770)
 * Fix anomalies in rules (#2778)
 * Fix HUE bridge V1 software version by Heiko Krupp (#2788)
```

If your issue is solved, please close it. Thanks! :+1:

closed 2018-06-22 by law1964