arendst/Tasmota #3017: Analogread very slow for low voltage

kpcharles, 2018-06-18:
I am using analogread to read external voltage using D1 mini. It works fine when the input voltage is above 375 mV (the reading >100). When the voltage is below that, my program of reading 300 readings time out because it takes a long time. Is this normal?.

Frogmore42, 2018-06-18:
You're going to have to provide at least a little more information.  What code are you using? How are you using it?  Where is your program running?

kpcharles, 2018-06-18:
The code is below:
#define stp D2
#define dir D1
#define MS1 D6
#define MS2 D5
#define MS3 D0
#define EN  D7
#define AA D3
#define AB D4

int Di = 0;
int Count = 0;
int S1 = 0;
int S2 = 0;
int S3 = 0;
int val[1000] = {};

       //declare a global reset all pins function
void resetBEDPins()
{
  digitalWrite(stp, LOW);
  digitalWrite(dir, LOW);    //clockwise
  digitalWrite(MS1, LOW);    //000 (full) 100(hal) 010(1/4) 110(1/8) 111(1/16, defualt)
  digitalWrite(MS2, LOW);
  digitalWrite(MS3, LOW);
  digitalWrite(EN, HIGH);
  digitalWrite(AA, LOW);
  digitalWrite(AB, LOW);    //disable all outputs to motor
}

void setup()
{
  Serial.begin(921600);
  pinMode(A0, INPUT);
  pinMode(stp, OUTPUT);
  pinMode(dir, OUTPUT);
  pinMode(MS1, OUTPUT);
  pinMode(MS2, OUTPUT);
  pinMode(MS3, OUTPUT);
  pinMode(EN, OUTPUT);
  pinMode(AA, OUTPUT);
  pinMode(AB, OUTPUT);
  resetBEDPins();           //All pins reset. Direction=clockwise, full step and enabled
}

void loop()
{
 while (Serial.available() > 0)
 { 
   int Di = Serial.parseInt();       //direction?
   int Count = Serial.parseInt();    //number of steps
   int S1 = Serial.parseInt();       //step size
   int S2 = Serial.parseInt(); 
   int S3 = Serial.parseInt();
   int S4 = Serial.parseInt();
   int S5 = Serial.parseInt();
    if (Di == 0){digitalWrite(dir, LOW);}   //set direction
    else { digitalWrite(dir, HIGH);}
    if (S1 == 0){digitalWrite(MS1, LOW);}   //set step size
    else { digitalWrite(MS1, HIGH);}
    if (S2 == 0){digitalWrite(MS2, LOW);}
    else { digitalWrite(MS2, HIGH);}
    if (S3 == 0){digitalWrite(MS3, LOW);}
    else { digitalWrite(MS3, HIGH);}
    if (S4 == 0){digitalWrite(AA, LOW);}    //choose motor
    else { digitalWrite(AA, HIGH);}
    if (S5 == 0){digitalWrite(AB, LOW);}
    else { digitalWrite(AB, HIGH);}
      
    digitalWrite(EN, LOW);           //enable motor  
    for(int i=0; i<Count; i++) {
      digitalWrite(stp,HIGH);        //pull stp pin up (Trigger one step)
      delay(1);
      digitalWrite(stp,LOW);         //Pull stp pin low (so it can be triggered again)
      delay(1);
      val[i] = analogRead(A0);       //read the analog input (3.3V = 1024)
    }
    digitalWrite(EN, HIGH);           //disable motor  
    for(int i=0; i<Count; i++) {
    Serial.print(val[i]);          //send the data
    Serial.print(",");              //termination character
    }
}
}
The D1 Mini is providing digital signals to switch a number of channels. Once the channel is chosen, I do a series of analog read and then send the results to a PC through the usb port. At voltage above 375 mV, I can do 2000 ADC per second. At below 375 mV, the program just time out after roughly 10 seconds. All the indication is that the D1 Mini has not yet completed the 300 ADC called for.

kpcharles, 2018-06-18:
A little more detail:
The program trigger the stepper motor to advance one step and then read the voltage..

Frogmore42, 2018-06-18:
This does not appear to have anything to do with Tasmota.

ascillato2, 2018-06-19:
Hi

Seems that your issue is for Arduino. Here is for Sonoff-Tasmota related issues.

Please, post it there. Thanks 

Off-topic. Closing.

closed 2018-06-19 by ascillato2