arendst/Tasmota #3019: Sonoff Sensor- DW1 with Tasmota?

prankousky, 2018-06-19:
Hi everybody,

a month ago or so I found this Sonoff DW1 sensor while browsing a Chinese shopping plattform. Without even reading the specs (which I should have!), I ordered a few, figuring if it is Sonoff, it will run with Tasmota ;)

Now that I received them, I realize they are not wifi but RF433. There is no info in the tiny paper manual that came with them, but I also have an **RF bridge** by Sonoff. Is there a way to **first** have an RF bridge running Tasmota capture whatever RF signals these sensors are sending, **and then** sending those signals to my mosquitto server with the appropriate information?

These are open/closed sensors to attach to doors and/or windows. So what I'd actually like to do is place them at a test location, open/close the sensor (I assume it's just a magnet sensor) and then have the current status transmitted from the sensor to my rf bridge, from my rf bridge to my mqtt server, and from there to node-red, which I use to convert and forward these mqtt packets in order to display "door x is currently closed" on my smartphone, for example.

Does anybody have one of these sensors and has an idea on how to do this?

Here is a picture of what I received in my order.

[https://i.imgur.com/0xHkRHt.jpg](https://i.imgur.com/0xHkRHt.jpg)

Thanks in advance for your help and best regards

ascillato, 2018-06-19:
Hi,

The DW1 sensor is not open/close. It just senses when it opens and it sends a unique rf code to the sonoff bridge. It will not send anything when closing.

The information for mqtt from rf bridge is on the wiki https://github.com/arendst/Sonoff-Tasmota/wiki

Also, please search in the previous issues because this was addressed before.

Thanks.

Jason2866, 2018-06-19:
Hi, does anybody know a Window Rf 433 Mhz sensor which sends codes for open AND close?
Best case different ones for open and close.

ascillato, 2018-06-20:
Hi, I'm looking for the same thing. Up to now, Sonoff don't have that.

ascillato2, 2018-06-27:
Hi,

If your issue is not solved, please feel free to reopen this issue. Thanks.

closed 2018-06-27 by ascillato2

pedrosimao, 2019-02-28:
I would also love a sonoff sensore with close signal (and open).

fanlitcn, 2019-08-16:
Is there have any new device can solve this problem?

ascillato, 2019-08-16:
Solution:

See issues #3649 and #3195

pedrosimao, 2019-08-16:
@fanlitcn Sonoff bridge works with any RF device, so any open and close RF door sensor you buy will work with Sonoff Bridge. Unhappily I had to ditch my DW1s...

fanlitcn, 2019-08-17:
I hv a another plan, using XIAOMI gateway, it has door windows sensor, and can connect to Homeassistant, then send MQTT message

pedrosimao, 2019-08-17:
I currently use Sonoff RF Bridge connected to HomeAssistant via MQTT, so, it is basically the same.

wvsbsp, 2019-09-18:
I'm using different RF Sensors and transmitters to control an monitor different things in my home. NodeRed gets the MQTT message from the Sonoff RF Gateway, the it is converted to JSON and the received RF-Code is extracted from the JSON-Array. The next function-nodes compare the incoming Code with one known code (or multiple codes from my window-sensor with tamper-switch which sends 3 different codes).
Here is an example:
`[{"id":"8544ade2.0f45c8","type":"mqtt in","z":"3b58a437.2d96c4","name":"Sonoff RF","topic":"sonoff/bridge/RESULT","qos":"1","datatype":"auto","broker":"55032d8f.5c52c4","x":171,"y":336,"wires":[["11893b1f.0fa88d"]]},{"id":"11893b1f.0fa88d","type":"json","z":"3b58a437.2d96c4","name":"","property":"payload","action":"","pretty":false,"x":310,"y":335,"wires":[["6e8f504f.436e5"]]},{"id":"6e8f504f.436e5","type":"function","z":"3b58a437.2d96c4","name":"RF-Sensor Code","func":"msg.payload=msg.payload.RfReceived.Data;\n\nreturn msg;","outputs":1,"noerr":0,"x":478,"y":335,"wires":[["acd092c4.5dc36","7488b28e.1032dc","2b4d2cc1.320dd4"]]},{"id":"acd092c4.5dc36","type":"function","z":"3b58a437.2d96c4","name":"RF Remote Button A","func":"if (msg.payload==\"112233\") {\n    msg.payload = \"ON\";\n    return [msg, null];}\nelse {\n    msg.payload = \"OFF\"\n    return [null, msg];}","outputs":2,"noerr":0,"x":735,"y":280,"wires":[["4f1958c1.20e37"],[]]},{"id":"7488b28e.1032dc","type":"function","z":"3b58a437.2d96c4","name":"RF Remote Button B","func":"if (msg.payload==\"334455\") {\n    msg.payload = \"ON\";\n    return [msg, null];}\nelse {\n    msg.payload = \"OFF\"\n    return [null, msg];}","outputs":2,"noerr":0,"x":736,"y":332,"wires":[["83c34ac9.4776d"],[]]},{"id":"2b4d2cc1.320dd4","type":"function","z":"3b58a437.2d96c4","name":"RF Window 1","func":"if (msg.payload==\"11001E\") {\n    msg.payload = \"ON\";\n    return [msg, null];}\nelse if (msg.payload==\"11001A\") {\n    msg.payload = \"OFF\";\n    return [msg, null];}\nelse if (msg.payload==\"110017\"){\n    msg.payload = \"ON\"\n    return [null, msg];}","outputs":2,"noerr":0,"x":718,"y":411,"wires":[["11834a1c.34d6a6"],["91618938.045e4"]]},{"id":"4f1958c1.20e37","type":"trigger","z":"3b58a437.2d96c4","op1":"ON","op2":"OFF","op1type":"str","op2type":"str","duration":"10","extend":true,"units":"s","reset":"","bytopic":"all","name":"","x":949,"y":278,"wires":[["d5a965e3.0433"]]},{"id":"83c34ac9.4776d","type":"trigger","z":"3b58a437.2d96c4","op1":"ON","op2":"OFF","op1type":"str","op2type":"str","duration":"10","extend":true,"units":"s","reset":"","bytopic":"all","name":"","x":948,"y":329,"wires":[["f4690159.437ad"]]},{"id":"d5a965e3.0433","type":"ui_switch","z":"3b58a437.2d96c4","name":"","label":"RF Button A","tooltip":"","group":"19f01bdc.bd8d04","order":7,"width":0,"height":0,"passthru":false,"decouple":"true","topic":"","style":"","onvalue":"ON","onvalueType":"str","onicon":"","oncolor":"","offvalue":"OFF","offvalueType":"str","officon":"","offcolor":"","x":1141,"y":278,"wires":[[]]},{"id":"f4690159.437ad","type":"ui_switch","z":"3b58a437.2d96c4","name":"","label":"RF Button B","tooltip":"","group":"19f01bdc.bd8d04","order":7,"width":0,"height":0,"passthru":false,"decouple":"true","topic":"","style":"","onvalue":"ON","onvalueType":"str","onicon":"","oncolor":"","offvalue":"OFF","offvalueType":"str","officon":"","offcolor":"","x":1141,"y":328,"wires":[[]]},{"id":"11834a1c.34d6a6","type":"ui_switch","z":"3b58a437.2d96c4","name":"","label":"Window 1","tooltip":"","group":"19f01bdc.bd8d04","order":7,"width":0,"height":0,"passthru":false,"decouple":"true","topic":"","style":"","onvalue":"ON","onvalueType":"str","onicon":"","oncolor":"","offvalue":"OFF","offvalueType":"str","officon":"","offcolor":"","x":937,"y":435,"wires":[[]]},{"id":"91618938.045e4","type":"trigger","z":"3b58a437.2d96c4","op1":"ON","op2":"OFF","op1type":"str","op2type":"str","duration":"10","extend":true,"units":"s","reset":"","bytopic":"all","name":"","x":946,"y":488,"wires":[["944eaf91.2725d"]]},{"id":"944eaf91.2725d","type":"ui_switch","z":"3b58a437.2d96c4","name":"","label":"Window 1 Alarm","tooltip":"","group":"19f01bdc.bd8d04","order":7,"width":"0","height":"0","passthru":false,"decouple":"true","topic":"","style":"","onvalue":"ON","onvalueType":"str","onicon":"","oncolor":"","offvalue":"OFF","offvalueType":"str","officon":"","offcolor":"","x":1140,"y":479,"wires":[[]]},{"id":"55032d8f.5c52c4","type":"mqtt-broker","z":"","broker":"localhost","port":"1883","clientid":"","usetls":false,"compatmode":true,"keepalive":"60","cleansession":true,"birthTopic":"","birthQos":"0","birthPayload":"","willTopic":"","willQos":"0","willPayload":""},{"id":"19f01bdc.bd8d04","type":"ui_group","z":"","name":"Wohnzimmer","tab":"53b2d5a5.656c0c","order":1,"disp":true,"width":"6","collapse":false},{"id":"53b2d5a5.656c0c","type":"ui_tab","z":"","name":"Wohnung","icon":"dashboard","order":2,"disabled":false,"hidden":false,"info":"# Test2"}]`