arendst/Tasmota #3020: Belkin Wemo emulation

tarasfrompir, 2018-06-19:
Please create a temperature sensor for Belkin Wemo emulation, and also create additional devices present in SONOOFF devices

Frogmore42, 2018-06-19:
Why?
What do you expect to do with it?
In the quick search that I did, Wemo doesn't have any temperature sensors, so it is currently impossible to do what you want in wemo mode.

But, if your goal is to automate something based on temperature with Sonoff devices, that is easy with Tasmota.  You will need to setup a mqtt broker and some type of home automation.  There are lots of choices for that.

tarasfrompir, 2018-06-20:
Sorry, i dont like mqtt. I like UPNP protocol.

Frogmore42, 2018-06-20:
Not sure what you have against MQTT, nor why you are a fan of UPNP.  But, I could find no evidence that Wemo has any temperature sensor support.  So you'll probably be waiting a long time for Tasmota to support something that doesn't exist, especially when there is that type of functionality already supported with MQTT.

ascillato, 2018-06-20:
Hi

The actual Wemo Support is only for Lights. Sorry.

closed 2018-06-21 by ascillato2