arendst/Tasmota #3021: Failed to connect to Sonoff basic module using MQTT

sagilo, 2018-06-19:
I've flashed the latest version successfully using platform.io on Sonoff basic relay.  
I've connected the module using serial connection, opened the serial monitor and executed:
 `pio device monitor --port /dev/cu.usbserial-00000000 --baud 1
15200`
 output:
`
--- Miniterm on /dev/cu.usbserial-00000000  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
00:00:01 WIF: Checking connection...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Checking connection...
....
00:00:07 WIF: Checking connection...
00:00:07 WIF: Attempting connection...
00:00:07 WIF: Connect failed as AP cannot be reached
00:00:07 WIF: Patch issue 2186
00:00:07 WIF: Connecting to AP1 lhome_2_4 in mode 11N as sonoff-001-2446...
00:00:08 CFG: Saved to flash at F4, Count 92, Bytes 3584
00:00:08 WIF: Checking connection...
00:00:08 WIF: Attempting connection...
00:00:09 WIF: Checking connection...
00:00:09 WIF: Attempting connection...
...
00:00:13 WIF: Checking connection...
00:00:13 WIF: Attempting connection...
00:00:14 WIF: Checking connection...
00:00:14 WIF: Connected
00:00:14 DNS: Initialized
00:00:14 HTP: Web server active on sonoff-001-2446.local with IP address 192.168.1.6
00:00:15 MQT: Attempting connection...
00:00:19 APP: (UTC) Tue Jun 19 19:40:57 2018, (DST) Sun Mar 25 02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
20:41:00 MQT: Connect failed to domus1:1883, rc -2. Retry in 10 sec
20:41:11 MQT: Attempting connection...
20:41:17 MQT: Connect failed to domus1:1883, rc -2. Retry in 10 sec
20:41:25 WIF: Checking connection...
20:41:25 WIF: Connected
20:41:28 MQT: Attempting connection...
20:41:34 MQT: Connect failed to domus1:1883, rc -2. Retry in 10 sec
...
`
As you can see the WiFi connection is OK but the MQTT one fails.
If I disconnect the serial connection and use the module with live electricity it doesn't even connect to the WiFi network and restarts itself every few minutes.
I've tried re-flashing, no change.

I'm using Atom on Macbook pro with macOS sierra.

chris-008, 2018-06-19:
This link helped me to get MQTT working:
http://www.apdaga.com/2018/01/raspberry-pi-install-and-test-mosquitto.html

Frogmore42, 2018-06-20:
also, please provide status 0 as the template requires

sagilo, 2018-06-20:
The mqtt brocker was down for some reason (the service was up but it failed to work).
Restarting the service did the trick.
Thank you all for your assistance!

closed 2018-06-20 by sagilo