arendst/Tasmota #3023: How to define Item that reads sensor switch on Sonoff GPIO14

sparkgap, 2018-06-19:
Using Sonoff Basic,  this turns on my carport lights...
`Switch Carport_Lights      {mqtt=">[HABbroker:cmnd/sonoff-11/POWER:command:*:default]"}`
Now I want to sense motion using a #HC SR501
to be continued....

Frogmore42, 2018-06-20:
See this for how to set up a motion sensor: https://github.com/arendst/Sonoff-Tasmota/wiki/Wemos-D1-Mini-and-HC-SR501-PIR-Motion-Sensor

Since you are using the relay and don't want the motion sensor to control it directly, set up the motion sensor on Switch2 and also add Relay2.  You need both to get it to send the MQTT messages.  You can then look at the console log and see the exact messages it sends out when it detects motion and you can then figure out what you need to put into HAB to respond to it.  I don't use HAB so I don't know exactly what you want to do with it.

sparkgap, 2018-06-20:
Is this openHAB syntax correct?
`Contact Carport_Motion {mgtt="<[HABbroker:stat/sonoff-11/SENSOR:state:default]"}`

boris099, 2018-06-20:
Hi I am also quite a beginner but was yesterday successfully getting the GPIO14/Ground working as I require. My setup is a bit different but comparable with that you want to achieve.
So you need to set GPIO14 as SWITCH2 and assign additionally RELAY2 to GPIO4.
When done you will see suddenly a second switch on the TASMOTA GUI.
And you need to change your current setup (item/rules) a bit cause you now get POWER1 and POWER2 instead POWER as used before. That´s more or less the trick. 
I hope this helps even I am absolutely lost about the syntax for MQTT.
I assume you need to change like:
Switch Carport_Lights {mqtt=">[HABbroker:cmnd/sonoff-11/POWER1:command:*:default]"}   and
Switch Carport_Motion {mqtt=">[HABbroker:cmnd/sonoff-11/POWER2:command:*:default]"}

so please let me know if this works?

Jason2866, 2018-06-20:
@boris099 
Nearly perfect👍
Switch Carport_Lights {mqtt=">[HABbroker:cmnd/sonoff-11/POWER1:command:*:default]"}
Switch Carport_Motion {mqtt=">[HABbroker:cmnd/sonoff-11/POWER2:command:*:default]"}

sparkgap, 2018-06-20:
I'm slowly beginning to understand.   which is........ that Switch1 always controls Relay1 however I can create as nonexistent relay on GPIO4 as `18 Relay2` and associate it as GPIO14 `10 Switch2` .  I have done this and now see Togge 1 and Toggle 2 buttons (blue rectangles) on top of the Sonoff main menu.  Every 5 minutes I get this Console report `00:03:50 MQT: tele/sonoff-4/SENSOR = {"Time":"2018-06-21T00:03:50","Switch2":"OFF"}`  with GPIO14 open (hi) , then if holding GPIO14  grounded (low) when time for the 5 minute report Switch2 shows off like this` 00:13:59 MQT: tele/sonoff-4/SENSOR = {"Time":"2018-06-21T00:13:59","Switch2":"OFF"}`   To me this....

Frogmore42, 2018-06-21:
You'll need to look at the log to see what is actually happening.  You have added command and control with OpenHAB that may be complicating your understanding of what is happening in Tasmota.

Frogmore42, 2018-06-21:
Read the wiki here
https://github.com/arendst/Sonoff-Tasmota/wiki/Understanding-SwitchMode-and-SwitchTopic

You want switch mode 3 or 4 and switch topic 1.
This should then publish when there is motion.

Status 0 will help indicate how you have things configured.

sparkgap, 2018-06-21:
I am having difficulty grasping some of the steps you all have presented herein.   Possibly it is my inability to describe what I am trying to do.   I know how to create a Switch Item and a Rule to control my Sonoff Basic relay.    I do not want to loose that function.... it's working great.   Now can someone please tell me if I can somehow read GPIO14 into an openHAB Item (possibly a Contact Item type) that I can use in a Rule.   This is really the reverse of what I have already working, one is an output the other is an input.   Possibly the terminology is the confusion factor here; the word switch could be interpreted as an openHAB Switch Item type in software, or a switch could be interpreted as a physical electrical switch connected to GPIO-14.   This is as if I wanted to connect a window or door ajar sensing switch to GPIO14 which would be called an openHAB Contact item.  IN NO WAY IS MY USE OF THE SONOFF POWER RELAY ASSOCIATED WITH MY INTENDED  USE OF GPIO14.  GPIO14 just seemed like an available (free) GPIO that could be used for anything.  Again ..... can someone tell me if I can use them seperately and if so kindly suggest how I might reag GPIO14 into an openHAB Item.

sparkgap, 2018-06-21:
Just for clarification.......  I do not intend to connect GPIO14 to a contact switch to sense a door or window ajar.  I actually want to connect the on/off signal from a motion detector to GPIO14.  And again..... my rule for the relay (which is working fine now) will in no way be associated with sensing motion on GPIO14.   AGAIN...... I just want to use what I think is a free available GPIO for something totally unrelated to the use of the Power Relay .

Frogmore42, 2018-06-21:
Yes, what you want to do should be simple.  Based on what I read, I assumed that you want to have two switches in the device running Tasmota.  To get two switches working the way you want, you also need to have two relays.  One can be a virtual relay, ie you don't have to actually have one there or hook anything up to it, but you have to configure Tasmota for it.  You also probably want to have the 2nd switch send out MQTT commands when it is activated.

If you really want to learn how this all works, disconnect the device from OpenHAB, so what it is doing doesn't confuse the behavior.  Then add the virtual switch and relay, using any unused GPIOs.  Then use the web UI to activate the switches and look at the serial/web console to see what happens.  This will give the MQTT messages that get sent when the switch is triggered.  You can then add it to OpenHAB.

sparkgap, 2018-06-24:
Thanks Frog
I think I'm good now.   With GPIO4 - RELAY2 and GPIO14 - SWITCH2,  I see POWER2 going ON and OFF in the web Console, as well as from the Mosquitto commamd line on my Rasberry PI with `Mosguitto_sub -d -t stat/sonoff-4/POWER2`  

Next I need to get an openHAB Item working......   something like 
`Contact Carport_Motion     {mgtt="<[HABbroker:stat/sonoff-4/POWER2:state:default]"}`
and a rule trigger something like this.....
`when Carport_Motion.state == (ON)`
But not working.... getting some log errors.

I'll close this now.   But if you are familiar with openHAB  any comments on my Item and Rule would be appreciated.

Jason2866, 2018-06-24:
OpenHab Items:
`Switch Carport_Motion { mqtt="<[HABbroker:stat/sonoff/RESULT:state:JSONPATH($.POWER):.*POWER.*]" } 
`
The sensor is defined as switch because you get on off message from mqtt.  In a rule you can query the status of the item. Go to https://community.openhab.org  for searching examples or asking for help

sparkgap, 2018-06-25:
Jason thanks, I'll try this.  But I still don't understand.

The first evening I looked at openHAB basics, over a year ago, it was ingrained clearly in my mind that a Switch Item was an (on/off) output and a Contact Item was an (on/off) input.  https://www.openhab.org/docs/concepts/items.html#items .   It would seem that a door contact sensing switch on GPIO14 would be an openHAB Contact Item.  Similarly an (on/off) signal from a motion detector would seem to be sensed input item like a door contact, an openHAB Contact Item.
The term switch and contact are often in contradictory.   A physical light switch is a contact and a physical door switch is a contact.  Reading the openHAB Item descriptions literally there is only one way to do an on/off output and that is with a Switch Item and the only way to sense a on/off input condition is with a Contact Item.

You seem to be saying that an openHAB Switch OItem can be an sensed status input or an output command???  But then what is an openHAB Contact Item????

sparkgap, 2018-06-25:
The following suggestion did not work.....
`Switch Carport_Motion { mqtt="<[HABbroker:stat/sonoff-4/RESULT:state:JSONPATH($.POWER):.*POWER.*]" }`
No error when saving Item file, but get error pointing to this part of the Rule file.....
`when Carport_Motion.state == (ON) `  (error points to first letter of Carport_Motion) , which seems to indicate it did not like the format of the Item.

Here is what is working: `Mosguitto_sub -d -t stat/sonoff-4/POWER2`
Gives me an ON and OFF when I ground and un-ground Sonoff Basic GPOI14 , simulating my motion detector.  From my MQTT test client I see that I need to use `stat/sonoff-4/POWER2` but just don't understand how to build the rest of the syntax.

Based on my test client....... Would someone PLEASE give me exact code for a Carport_Motion Item that will respond to the following rule `when Carport_Motion.state == (ON)`

Please don't just send me to the openHAB examples, unless there is something very specific.   I have exhausted all the openHAB examples;  found nothing showing a simple input to an openHAB Item, only very complex application code there.

Jason2866, 2018-06-25:
Plz post the output `Mosguitto_sub -d -t stat/sonoff-4/POWER2`
It isnt easy for me too, to write the correct definition of the item without the exact mqtt message...

Jason2866, 2018-06-25:
In OpenHAB a switch can be used as a contact too when the state is the interesting thing...
In this case it is easier because there is no mapping needed for the received ON / OFF messages
As long as you dont put a switch item in your sitemap this is no problem ;-)
For a contact item a open / close mqtt messages is expected (in this case mapping would be needed)....

Jason2866, 2018-06-25:
Ok, i did it a bit complicated (because i need a regEx "Filter" for my connected sensors.
Here is the more clear and hopefully working one. I did overlook Power2!!
`Switch Carport_Motion { mqtt="<[HABbroker:stat/sonoff-4/POWER2:state:default]" }   `

Jason2866, 2018-06-25:
and a Rule Example:
`rule CarPort
	when 
		Item Carport_Motion received update ON
	then
		logInfo("RULES", "Lights ON for 5 Minutes")
		sendCommand(Sonoff_S20_1, ON)  
		createTimer(now.plusMinutes(5)) [|
		sendCommand(Sonoff_S20_1, OFF)
		]
	end	`

sparkgap, 2018-06-26:
Thanks Jason
I have finally closed the loop, I can sense a condition and take action.   I did not get all of your logging and timer script working properly yet but have the basics and can now build upon it.   Because Carport_lights sonoff is installed already,  I set up another test sonoff on the workbench called Test_Light and Test_Motion.

Here are my Items:
```
Switch Test_Light          {mqtt=">[HABbroker:cmnd/sonoff-7/POWER:command:*:default]"}
Switch Test_Motion      {mqtt="<[HABbroker:stat/sonoff-7/POWER2:state:default]" }
```
Here is my rule:
```
rule "test2"
when Item Test_Motion received update ON
then  Test_Light.sendCommand(ON)
end
```
Here is the log when I ground GPIO14 and release it:
```
[vent.ItemStateChangedEvent] - Test_Motion changed from OFF to ON
[ome.event.ItemCommandEvent] - Item 'Test_Light' received command ON
[vent.ItemStateChangedEvent] - Test_Motion changed from ON to OFF
```

I'll work on the logging and timer thing tomorrow night.

sparkgap, 2018-06-29:
My previous post demonstrates successfully sensing of a state on GPIO-14 and taking simple action.  This all seems so simple now, but getting there was pain full.  Thanks to all who helped me unravel the mystery of using a non existing (virtual) relay (Relay2) to be able to activate GPIO-14 as Switch2.   Hopefully someone will clear this up in the wiki. 

Logging and timer function suggested in a previous example is not working yet, but not part of this issue, therefor I am closing this issue.

closed 2018-06-29 by sparkgap