arendst/Tasmota #3024: Sonoff restarts it self and doesn't connect to WiFi

sagilo, 2018-06-20:
When used with serial connection it works perfectly.
As i disconnect the serial and connect it to the power cables, the led blinks, sometimes gets solid and then restarts itself.
I can't access it's webserver and the MQTT doesn't response as well.

Using Sonoff basic, latest Tasmota version.
Flashed using Atom/platformIO on macOS Sierra.

Frogmore42, 2018-06-20:
It is standard to fill out the template.  Not doing so means someone has to take the time to ask you to provide all the information in the template.  There are many reasons why things can not work.  Having all the information details about what you are using and what you have tried means we don't have to waste time.

There can be hardware (your device is defective), software compilation, and configuration issues that could cause the problem.

sagilo, 2018-06-20:
Hi @Frogmore42 
I've followed the template.
I've searched all the relevant places, mentioned the used HW, but I can't provide the `status 0` command output since the device won't start/connect.
I **highly appreciate** the effort every person puts in assisting the other, and I do my best to provide all the necessary information up front.

That said, Ill be happy if you could please point me to what is missing in the post so it will be more clear.

Thank you

Frogmore42, 2018-06-20:
You erased all of the template, so no one else knows what you did.

Provide 0 when it is connected to serial where you said it worked.  That will indicate all the details of excatly what you are running.  There are a few ways you could have done it and some of them cause issues for some devices some times.

You may have a broken device.  If you only have one, it is hard to know.

You may have misconfigured something, if you only have one device and have never configured something before it is easy to make a mistake.

Also provide the log showing it working on serial.  This usually indicates most things are configured well.

There have been a few cases of people having devices that didn't work when plugged in that were hardware related.

reloxx13, 2018-06-20:
> Make sure these boxes are checked [x] before submitting your issue - Thank you!
> 
> - [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
> - [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
> - [ ] Searched the problem in the forum (https://forum.digital-pig.com/)
> - [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
> - [ ] Hardware used : 
> - [ ] Provide the output of command ``status 0`` :
> ```
> STATUS 0 OUTPUT HERE
> ```
> 
> **(Please, remember to close the issue when the problem has been addressed)**
>

sagilo, 2018-06-20:
@Frogmore42 @reloxx13 Thank you for clarifying.
I've updated the thread with (hopefully) all the necessary information.

Frogmore42, 2018-06-20:
The log has this:
"RestartReason":"Hardware Watchdog"

That is making it look more likely that you have an at least somewhat defective device.  It could be the 5V supply is not stable or it could be something else.  I would suggest getting a different one.  I had one dual that I could never get to work correctly.  If it had been my first Sonoff device, I would have been much more confused than I was.

The log indicated it was connected to MQTT, so assuming you tried it plugged in at the same location, it seems the 5V and 3V3 power supplies are the likely culprits.  Unless you have experience repairing electronic circuits, it is probably time for a new one.

sagilo, 2018-06-20:
Bummer .. 
I'll try another one.
@Frogmore42  Thanks a lot!

closed 2018-06-20 by sagilo