arendst/Tasmota #3025: Sonoff RF 433 On R2 V1.0 Boards, hardware hack needed?

sarsw, 2018-06-20:
Hi,

I'm using the Sonoff RF Bridge, R2 V1 and Tasmota Version 5.14.0 20180515 (I tried the sonoff.bin & built the code).  There is no indication in the console that it is receiving a signal.  If I try to learn a code it stays in "Learning active" (or similar) and there is no beep.  With the default Sonoff firmware it can receive my PIR and remote RF signals.  I restored the default to make sure I'd not imagined it!

I've also tried Espurna - same issue.

Must I do the direct hack to have any success https://github.com/xoseperez/espurna/wiki/Hardware-Itead-Sonoff-RF-Bridge---Direct-Hack ?

This link indicates the GPIO swap https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-RF-Bridge-433 but it's not clear (to me) what this means to the Tasmota firmware.

I just want to make sure before breaking tracks and soldering resistors!

[image]
[image]

Thanks,
Steve

Jason2866, 2018-06-20:
Tasmota doesnt change anything in decoding rf signals!
This is done by the efm8bb1. The originial firmware of this chip has problems to decode 2262 code.
With portisch firmware 2262 code works...
You can flash via Tasmota 😎 the portisch open source firmware
https://github.com/Portisch/RF-Bridge-EFM8BB1
I have both bridges both work with TASMOTA with orig. Efm8bb1 Firmware and portisch firmware.
So if your bridge doesnt work with Tasmota only 2 possibilitys flash failed or hardware defect

Jason2866, 2018-06-20:
You dont have to cut wires to upgrade the firmware of R2!!!! Do not power it over USB
You have to power the bridge over the pin header with 3.3V from the usb serial adapter

sarsw, 2018-06-21:
The stock Itead firmware does capture my RF signals so the hardware works in some capacity and suggests the EFM8BB1 is ok with my RF protocols.  The flashing of Espurna and Tasmota went ok with Esptool.py & Arduino IDE & I could configure the board via the web UI - just no indication of receiving any RF signals.  I'll try again. Thanks for confirming the hardware changes are unnecessary.

Jason2866, 2018-06-22:
No difference to recognize whether ITEAD or Tasmota firmware because this is done from the EFM!
So the orig. Firmware cant do more! Technical not possible.

ascillato, 2018-06-22:
Hi,

if your questions is addressed, please close this issue. thanks :+1:

sarsw, 2018-06-23:
Ok, I flashed the same Tasmota firmware again from the Ardiuno IDE.  No errors as before. Configured it as "25 Sonoff Bridge" as before.  Now I get a beep and a flashing red LED with this at the console:

22:00:39 CMD: RfKey1 2
22:00:39 MQT: RF/stat/RESULT = {"RfKey1":"Start learning"}   ----> beep and LED flashes at bridge
22:01:39 MQT: RF/stat/RESULT = {"RfKey1":"Learn failed"}

I have two remotes that I've used with the default demos of RcSwitch code with Wemos D1 Mini devices to sniff the same Maplin remotes, much like this https://www.makeuseof.com/tag/control-cheap-rf-power-sockets-openhab/. The Tasmota firmware offers far more features & allows me to use the same firmware in all my devices. Any advice?  Should I flash the EFM ic? Can I back it up first to restore functionality - the docs do not mention this.

I've now read someone else here https://github.com/arendst/Sonoff-Tasmota/issues/1181 had the same issue but no indication of a solution.  It's like I've missed something! Maybe there is an issue with the Tasmota version I'm using?  What Tasmota firmware have you used in your unaltered R2 V1.0 brdige Jason?

Here's the back of my bridge with flashing connector.

[image]

Jason2866, 2018-06-24:
You can go back to the original firmware.
I use latest Tasmota V.6.0a. The behaviour you describe is normal for Codes that the Bridge cant handle.
You find the orig. Firmware here.
https://github.com/Portisch/RF-Bridge-EFM8BB1

zeliant, 2018-06-25:
@sarsw seems that we have the same issues. Same hardware as well.

I tried Tasmota v5, can't remember the actual version. I tried ESPurna too. The unit not picking up any RF signal from my remotes.

Would be great if you could post here if you find any success with Tasmota or ESPurna. 

Eagerly waiting for any possible hints/direction to get our RFBridge working with Tasmota or ESpurna

sarsw, 2018-06-25:
@Jason2866 Ah, I see the original firmware.  I'll try flashing the new one in the next few days to see how things change. Thanks.

Jason2866, 2018-06-25:
@sarsw 
I think your remotes will work with the Portisch firmware. I read the article you provided a link.
This are 433 Power Outlets with Code 2262 ("Intertechno"). This are the same which didnt work for me
with orig. firmware. With portisch firmware the work!

Jason2866, 2018-06-26:
@zeliant 
Items definition to switch RF Poweroutlets:

`Number RfBridge_RSSI "RfBridge: RSSI [%d %%]" (gRSSI)
    { mqtt="<[broker:tele/sonoff-8E8329/STATE:state:JSONPATH($.Wifi.RSSI)]" }
String RfBridge_Up "RfBridge: Uptime [%s]" (gUptime)
    { mqtt="<[broker:tele/sonoff-8E8329/STATE:state:JSONPATH($.Uptime)]" }
String RfBridge_RfReceive "RfReceive1 [%s]"
    { mqtt="<[broker:tele/sonoff-8E8329/RESULT:state:JSONPATH($.RfReceived.Data)]" }  

Switch WohnFluter       "Wohnz. Fluter"         <light> (LR,gLight)     [ "Lighting" ] 
		{ mqtt=">[broker:cmnd/sonoff-8E8329/Backlog:command:ON:RfSync 11320; RfLow 360; RfHigh 1090; RfCode #150551],
                >[broker:cmnd/sonoff-8E8329/Backlog:command:OFF:RfSync 11320; RfLow 360; RfHigh 1090; RfCode #150554]" }

Switch SchrankLicht     "Schrankwand Licht"     <light> (LR,gLight)     [ "Lighting" ] 
		{ mqtt=">[broker:cmnd/sonoff-8E8329/Backlog:command:ON:RfSync 11320; RfLow 360; RfHigh 1090; RfCode #151151],
                >[broker:cmnd/sonoff-8E8329/Backlog:command:OFF:RfSync 11320; RfLow 360; RfHigh 1090; RfCode #151154]" }

Switch SchrankTechni    "Schrankwand Technik"  (LR,gLight)              [ "Lighting" ] 
		{ mqtt=">[broker:cmnd/sonoff-8E8329/Backlog:command:ON:RfSync 11320; RfLow 360; RfHigh 1090; RfCode #151451],
                >[broker:cmnd/sonoff-8E8329/Backlog:command:OFF:RfSync 11320; RfLow 360; RfHigh 1090; RfCode #151454]" }
 
and so on......

`

and a rule to update Status if Power Outlet is switched with old school remote:

`
rule "RF433 switch sonoff and Update RF Switch (Bridge 1)"
when
    Item RfBridge_RfReceive changed
then
if (RfBridge_RfReceive.state == NULL)
    {
    logInfo("RfBridge1.rules", "Item is null, cancelling...")
    return;
    } 
    var rfData = RfBridge_RfReceive.state.toString
    logInfo("rfbridge1.rules", "Received IT Codes: " + rfData)
        switch (rfData) {         
            case "A00A1F":
               {
                publish("broker", "cmnd/sonoff-13D92F/POWER", "ON")
                }
            case "A00A17":
                {
                publish("broker", "cmnd/sonoff-13D92F/POWER", "OFF")
                }
            case "A00A1B":
               {
                publish("broker", "cmnd/sonoff-065CA6/POWER", "ON")
                }
            case "A00A13":
                {
                publish("broker", "cmnd/sonoff-065CA6/POWER", "OFF")
                }
            case "A00A1D":
               {
                publish("broker", "cmnd/sonoff-101AEE/POWER", "ON")
                }
            case "A00A15":
                {
                publish("broker", "cmnd/sonoff-101AEE/POWER", "OFF")
                }                
            case "A00A1E":
               {
                publish("broker", "cmnd/sonoff-5DE6D5/POWER", "ON")
                }
            case "A00A16":
                {
                publish("broker", "cmnd/sonoff-5DE6D5/POWER", "OFF")
                }
        	case "85D5CC":
        	    {
			    Pir433.sendCommand(ON)
			    }                   
            case "150551":
               {
				postUpdate(WohnFluter, ON)
                }
            case "150554":
                {
				postUpdate(WohnFluter, OFF)
                }   
            case "151151":
               {
				postUpdate(SchrankLicht, ON)
                }
            case "151154":
                {
				postUpdate(SchrankLicht, OFF)
                }   
            case "151451":
               {
				postUpdate(SchrankTechni, ON)
                }
            case "151454":
                {
				postUpdate(SchrankTechni, OFF)
                }   
            case "151511":
               {
				postUpdate(Glaskugel, ON)
                }
            case "151514":
                {
				postUpdate(Glaskugel, OFF)
                }   
            case "151541":
               {
				postUpdate(Leselampe, ON)
                }
            case "151544":
                {
				postUpdate(Leselampe, OFF)
                }   
            case "150151":
               {
				postUpdate(SideRechts, ON)
                }
            case "150154":
                {
				postUpdate(SideRechts, OFF)
                }   
            case "151051":
               {
				postUpdate(Salzlampe, ON)
                }
            case "151054":
                {
				postUpdate(Salzlampe, OFF)
                }   
            case "3575C0":
               {
				postUpdate(Schlafzimmer, ON)
                }
            case "35750C":
                {
				postUpdate(Schlafzimmer, OFF)
                }   
    }
 rfData=""
 RfBridge_RfReceive.postUpdate(rfData)
end

`

zeliant, 2018-06-28:
@Jason2866 thank you very much for showing me your OH items.

I tried to record my remote buttons using the original firmware and failed too. I guess my remote is using unsupported protocol.

Do you have a guide on how to flash the portisch firmware?

Jason2866, 2018-06-28:
@zeliant  Take a look #3061 and #2886

zeliant, 2018-06-29:
@Jason2866 thank you so much for showing me those threads.

I am going to try to flash my rfbridge to v6.0a first then proceed to flash portisch firmware.

then i will try again with learn my remote again. Will be very busy for JUL 2018. I will report back when I have time in AUG.

Thanks @Jason2866

mc1271, 2018-07-05:
Hi all,
just received my RF Bridge today.
Managed to flash to 5.14.
However can't pair my button.
I use a Sonoff T433 86 Button and I see events happening in the console once I press the button.
But trying to pair the Button fails:

`23:46:29 CMD: RfKey1 2
23:46:29 MQT: stat/sonoffbridge/RESULT = {"RfKey1":"Start learning"}
23:46:43 CMD: RfKey1 4
23:46:43 MQT: stat/sonoffbridge/RESULT = {"RfKey1":"Learning active"}
23:47:29 MQT: stat/sonoffbridge/RESULT = {"RfKey1":"Learn failed"}`  

Upgraded to 6.0.0c with the same effect.

Any suggestion greatly appreciated

ascillato2, 2018-07-10:
@sarsw Have you managed to solve your issue?

@mc1271 Please, open a new issue with the troubleshooting template completed. Thanks

SASpirit, 2018-07-11:
Any updates on this issue? I am trying to learn a remote and it simply doesn't work...

08:16:30 RSL: stat/Sonoff-RF/RESULT = {"RfKey1":"Start learning"}
08:16:30 SER: Received A0 55
08:17:04 CMD: rfkey1 2
08:17:04 RSL: Group 0, Index 1, Command RFKEY, Data 2
08:17:04 RSL: stat/Sonoff-RF/RESULT = {"RfKey1":"Learning active"}
08:17:31 SER: Received A2 55
08:17:31 RSL: stat/Sonoff-RF/RESULT = {"RfKey1":"Learn failed"}

ascillato2, 2018-07-16:
@SASpirit  Please, open a new issue with the troubleshooting template completed, so we can have the information to help you. Thanks

@sarsw Seems that your question has been addressed. Closing this issue. Please, reopen if needed. Thanks

closed 2018-07-16 by ascillato2

sarsw, 2018-07-17:
Well, it's been a while... my Rock64 psu failed so I lost my MQTT sever for a few days. Running on a slower,yet reliable backup Pi for a while.

I upgraded to Tasmota to 6.1.1a-2_4_0.. Built it and flashed via Ardiono IDE.
My RF could not be learned so I uploaded https://github.com/Portisch/RF-Bridge-EFM8BB1/blob/master/Keil%208051%20v9.53%20-%20Release/RF_Bridge.hex, the sonoff beeped and displayed "Success" in the web interface. I repowered it & it still will not learn my codes. Disappointing. Even an ESP Wemo D1 Mini and a £0.40 RF antenna will pickup the codes....I wanted the sonoff device to work for greater range & Tasmota is a much better solution.  If I return the sonoff to the manufacturer's firmware it does pick up the RF signals so it must be the firmware? I did not do any hardware mods to my sonoff.

Jason2866, 2018-07-18:
Power the bridge over USB for normal use and simply do nothing (no command entering)
Please post Web Console Output when you press Remote and after that let PIR send.
There is no need to learn the codes to the bridge when correct received and you want to do actions via mqtt.

sarsw, 2018-07-18:
@Jason2866 Here's the output after boot:
00:00:00 APP: Set Baudrate to 19200
00:00:00 Project sonoff Sonoff (Topic rf, Fallback DVES_A396BB, GroupTopic sonoffs) Version 6.1.1a-2_4_0
00:00:00 WIF: Connecting to AP1 BTHub6-8MGQ in mode 11N as rf-5819...
00:00:06 WIF: Connected
00:00:06 DNS: Initialized
00:00:06 HTP: Web server active on rf-5819.local with IP address 192.168.8.15
00:00:07 MQT: Attempting connection...
00:00:08 MQT: Connected
00:00:08 MQT: rf/tele/LWT = Online (retained)
00:00:08 MQT: rf/cmnd/POWER = 
00:00:08 MQT: rf/tele/INFO1 = {"Module":"Sonoff Bridge","Version":"6.1.1a","FallbackTopic":"DVES_A396BB","GroupTopic":"sonoffs"}
00:00:08 MQT: rf/tele/INFO2 = {"WebServerMode":"Admin","Hostname":"rf-5819","IPAddress":"192.168.8.15"}
00:00:08 MQT: rf/tele/INFO3 = {"RestartReason":"Power on"}

and then when I pressed the RF key I got..........drum roll.....

19:17:29 MQT: rf/tele/RESULT = {"RfReceived":{"Sync":13240,"Low":420,"High":1260,"Data":"544555","RfKey":"None"}}
19:17:30 MQT: rf/tele/RESULT = {"RfReceived":{"Sync":13300,"Low":390,"High":1240,"Data":"545557","RfKey":"None"}}

PIR movement caused:

19:23:34 MQT: rf/tele/RESULT = {"RfReceived":{"Sync":11620,"Low":380,"High":1140,"Data":"8947E8","RfKey":"None"}}
19:23:38 MQT: rf/tele/RESULT = {"RfReceived":{"Sync":11630,"Low":380,"High":1130,"Data":"8947E8","RfKey":"None"}}

Success!  I had tried the device powered only by the USB last night, but then I left the two cables to flash the EFM8BB1 connected. I'd tried without those too but maybe I'd not re-powered it or left it stuck in "learning" mode. Anyway. Thanks to you  guys I got there.  Cheers all.

Jason2866, 2018-07-18:
Have Phun!

patvdleer, 2019-06-13:
What am I missing? 

I followed a bunch of guides but I seem to be unable to flash the Sonoff bridge, I have 2, both of them give the same error. No matter what command I run I get the same output, Timed out waiting for packet header.

Since I only had a 5V USB FTDI laying around I hooked it up via a Arduino. I've tried this on Windows 10 as well as on Ubuntu 18.04.

The switch is set to off, rx and tx are correctly connected, held the binding button while powering. 

I do get the chip type detected but nothing after that...

I've tested it with esptool 2.5.0 and 2.5.2 (provided via the Arduino IDE)

```
$ esptool.py read_mac
esptool.py v2.5.0
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting........
Detecting chip type... ESP8266

A fatal error occurred: Timed out waiting for packet header
```

Even erase doesn't work...

```
patrick@patrick-X470-AORUS-ULTRA-GAMING:~$ esptool.py -t erase_flash
esptool.py v2.5.0
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting...TRACE +0.000 command op=0x08 data len=36 wait_response=1 timeout=0.100 data=
    0707122055555555 5555555555555555 | ... UUUUUUUUUUUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    55555555                          | UUUU
TRACE +0.000 Write 46 bytes: 
    c000082400000000 0007071220555555 | ...$........ UUU
    5555555555555555 5555555555555555 | UUUUUUUUUUUUUUUU
    5555555555555555 5555555555c0     | UUUUUUUUUUUUU.
TRACE +0.026 Read 1 bytes: c0
TRACE +0.000 Read 95 bytes: 
    0108020007071220 0000c0c001080200 | ....... ........
    070712200000c0c0 0108020007071220 | ... ........... 
    0000c0c001080200 070712200000c0c0 | ........... ....
    0108020007071220 0000c0c001080200 | ....... ........
    070712200000c0c0 0108020007071220 | ... ........... 
    0000c0c001080200 070712200000c0   | ........... ...
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000
TRACE +0.000 Received full packet: 01080200070712200000

Detecting chip type...TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=78000060
TRACE +0.000 Write 14 bytes: c0000a04000000000078000060c0
TRACE +0.015 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010a0200002006000000c0
TRACE +0.000 Received full packet: 010a0200002006000000
 ESP8266
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=5c00f03f
TRACE +0.000 Write 14 bytes: c0000a0400000000005c00f03fc0
TRACE +0.016 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010a0200940160000000c0
TRACE +0.000 Received full packet: 010a0200940160000000
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=5800f03f
TRACE +0.000 Write 14 bytes: c0000a0400000000005800f03fc0
TRACE +0.016 Read 1 bytes: c0
TRACE +0.000 Read 11 bytes: 010a020000b000170000c0
TRACE +0.000 Received full packet: 010a020000b000170000
TRACE +0.000 command op=0x0a data len=4 wait_response=1 timeout=3.000 data=5400f03f
TRACE +0.000 Write 14 bytes: c0000a0400000000005400f03fc0
TRACE +3.001 Timed out waiting for packet header

A fatal error occurred: Timed out waiting for packet header
```

Jason2866, 2019-06-13:
Using a 5V FTDI has probably killed the device. See [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki/Prerequisites)

patvdleer, 2019-06-13:
I didnt

dacrypt, 2020-02-11:
> I didnt, I noticed it before hand (for a change)

I also used 5V to flash it (a r2 v1.0) before finding that out. I'm able to flash it again and I don't get any errors but the WIFI led just doesn't light up. Have I killed mine too?