arendst/Tasmota #3026: Compiling Error

MrG1981, 2018-06-20:
Hello, I'm having some problems compiling firmware for my Sonoff basic. via Arduino IDE.
I'm following this guide and when I try to compile/verify the .ino file I have this error:

Arduino: 1.8.5 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, ck, 26 MHz, 40MHz, DOUT, 1M (no SPIFFS), 2, v1.4 Higher Bandwidth, Disabled, None, All Flash Contents, 115200"

Build-Optionen wurden verändert, alles wird neu kompiliert
C:\Users\MrG\Desktop\Sonoff-Tasmota-development\sonoff\xdrv_05_irremote.ino: In function 'boolean IrSendCommand()':

xdrv_05_irremote:291: error: 'StaticJsonBuffer' was not declared in this scope

       StaticJsonBuffer<128> jsonBuf;

       ^

C:\Users\MrG\Desktop\Sonoff-Tasmota-development\sonoff\xdrv_05_irremote.ino:291:7: note: suggested alternative:

In file included from C:\Users\MrG\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/StaticJsonDocument.hpp:8:0,

                 from C:\Users\MrG\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:15,

                 from C:\Users\MrG\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from C:\Users\MrG\Desktop\Sonoff-Tasmota-development\sonoff\sonoff.ino:54:

C:\Users\MrG\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Memory/StaticJsonBuffer.hpp:110:7: note:   'ArduinoJson::Internals::StaticJsonBuffer'

 class StaticJsonBuffer : public Internals::StaticJsonBufferBase {

       ^

xdrv_05_irremote:291: error: 'jsonBuf' was not declared in this scope

       StaticJsonBuffer<128> jsonBuf;

                             ^

C:\Users\MrG\Desktop\Sonoff-Tasmota-development\sonoff\xdrv_07_domoticz.ino: In function 'boolean DomoticzMqttData()':

xdrv_07_domoticz:168: error: 'StaticJsonBuffer' was not declared in this scope

     StaticJsonBuffer<400> jsonBuf;

     ^

C:\Users\MrG\Desktop\Sonoff-Tasmota-development\sonoff\xdrv_07_domoticz.ino:168:5: note: suggested alternative:

In file included from C:\Users\MrG\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/StaticJsonDocument.hpp:8:0,

                 from C:\Users\MrG\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:15,

                 from C:\Users\MrG\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from C:\Users\MrG\Desktop\Sonoff-Tasmota-development\sonoff\sonoff.ino:54:

C:\Users\MrG\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Memory/StaticJsonBuffer.hpp:110:7: note:   'ArduinoJson::Internals::StaticJsonBuffer'

 class StaticJsonBuffer : public Internals::StaticJsonBufferBase {

       ^

xdrv_07_domoticz:168: error: 'jsonBuf' was not declared in this scope

     StaticJsonBuffer<400> jsonBuf;

                           ^

C:\Users\MrG\Desktop\Sonoff-Tasmota-development\sonoff\xdrv_09_timers.ino: In function 'boolean TimerCommand()':

xdrv_09_timers:364: error: 'StaticJsonBuffer' was not declared in this scope

           StaticJsonBuffer<256> jsonBuffer;

           ^

C:\Users\MrG\Desktop\Sonoff-Tasmota-development\sonoff\xdrv_09_timers.ino:364:11: note: suggested alternative:

In file included from C:\Users\MrG\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/StaticJsonDocument.hpp:8:0,

                 from C:\Users\MrG\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:15,

                 from C:\Users\MrG\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from C:\Users\MrG\Desktop\Sonoff-Tasmota-development\sonoff\sonoff.ino:54:

C:\Users\MrG\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Memory/StaticJsonBuffer.hpp:110:7: note:   'ArduinoJson::Internals::StaticJsonBuffer'

 class StaticJsonBuffer : public Internals::StaticJsonBufferBase {

       ^

xdrv_09_timers:364: error: 'jsonBuffer' was not declared in this scope

           StaticJsonBuffer<256> jsonBuffer;

                                 ^

C:\Users\MrG\Desktop\Sonoff-Tasmota-development\sonoff\xdrv_10_rules.ino: In function 'bool RulesRuleMatch(byte, String&, String&)':

xdrv_10_rules:214: error: 'StaticJsonBuffer' was not declared in this scope

   StaticJsonBuffer<1024> jsonBuf;

   ^

C:\Users\MrG\Desktop\Sonoff-Tasmota-development\sonoff\xdrv_10_rules.ino:214:3: note: suggested alternative:

In file included from C:\Users\MrG\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/StaticJsonDocument.hpp:8:0,

                 from C:\Users\MrG\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:15,

                 from C:\Users\MrG\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from C:\Users\MrG\Desktop\Sonoff-Tasmota-development\sonoff\sonoff.ino:54:

C:\Users\MrG\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Memory/StaticJsonBuffer.hpp:110:7: note:   'ArduinoJson::Internals::StaticJsonBuffer'

 class StaticJsonBuffer : public Internals::StaticJsonBufferBase {

       ^

xdrv_10_rules:214: error: 'jsonBuf' was not declared in this scope

   StaticJsonBuffer<1024> jsonBuf;

                          ^

C:\Users\MrG\Desktop\Sonoff-Tasmota-development\sonoff\xplg_wemohue.ino: In function 'void HueLights(String*)':

xplg_wemohue:652: error: 'StaticJsonBuffer' was not declared in this scope

       StaticJsonBuffer<400> jsonBuffer;

       ^

C:\Users\MrG\Desktop\Sonoff-Tasmota-development\sonoff\xplg_wemohue.ino:652:7: note: suggested alternative:

In file included from C:\Users\MrG\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/StaticJsonDocument.hpp:8:0,

                 from C:\Users\MrG\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:15,

                 from C:\Users\MrG\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from C:\Users\MrG\Desktop\Sonoff-Tasmota-development\sonoff\sonoff.ino:54:

C:\Users\MrG\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Memory/StaticJsonBuffer.hpp:110:7: note:   'ArduinoJson::Internals::StaticJsonBuffer'

 class StaticJsonBuffer : public Internals::StaticJsonBufferBase {

       ^

xplg_wemohue:652: error: 'jsonBuffer' was not declared in this scope

       StaticJsonBuffer<400> jsonBuffer;

                             ^

exit status 1
'StaticJsonBuffer' was not declared in this scope

ascillato, 2018-06-20:
Hi,

You are missing some libraries.

Please read the wiki, there are all the steps for installation. Also search in the issues. Thanks

MrG1981, 2018-06-21:
Thank you installed new version of the IDE now its Compiling fine

closed 2018-06-21 by MrG1981

cybersquatter, 2018-06-22:
I'm running into the same issue. Can anybody provide a solution a bit more descriptive? I'm already running the latest IDE.

mistrovly, 2018-06-30:
Im running into the same issue. I've installed all the necessaries and i even installed the latest IDE version (1.8.5). I'm still stuck there.

warst, 2018-07-01:
I found clearing out my libraries folder and then re-copying from the tasmota 'lib' folder fixed the issue for me. I'd been using Arduino IDE to program some other modules and I think that's where it became confused, using the wrong versions and not being compatible. Emptied, re-copied and then opened IDE, clicked 'Verify' and it all passed fine.

RSABear, 2018-07-12:
Use the ArduinoJson Version 5.13.2 Library.

mike2nl, 2018-07-14:
I had the same issue but i had updated all libraries which where named in the manager.
Jep and ArduionJson was in a beta version, didn't seen it. grrr.

But thanks to RSABear and a strong hit from him ;-), all is back to normal and copiling works again.

geyseran, 2020-10-09:
I am trying to compile the latest Tasmota v8.5.1 from source, but compilation fails with the following error:
E:\Arduino\arduino-builder -dump-prefs -logger=machine -hardware E:\Arduino\hardware -hardware C:\Users\Home\AppData\Local\Arduino15\packages -tools E:\Arduino\tools-builder -tools E:\Arduino\hardware\tools\avr -tools C:\Users\Home\AppData\Local\Arduino15\packages -built-in-libraries E:\Arduino\libraries -libraries C:\Users\Home\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:xtal=80,vt=flash,exception=disabled,ssl=all,eesz=4M,led=2,ip=lm2f,dbg=Disabled,lvl=None____,wipe=sdk,baud=115200 -vid-pid=1A86_7523 -ide-version=10813 -build-path C:\Users\Home\AppData\Local\Temp\arduino_build_615351 -warnings=none -build-cache C:\Users\Home\AppData\Local\Temp\arduino_cache_667426 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.python3.path=C:\Users\Home\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\Home\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Home\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=C:\Users\Home\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.mklittlefs.path=C:\Users\Home\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.mklittlefs-2.5.0-4-fe5bb56.path=C:\Users\Home\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.mkspiffs.path=C:\Users\Home\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=C:\Users\Home\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -verbose C:\Users\Home\Documents\Arduino\Tasmota-8.5.1\Tasmota-8.5.1\tasmota\tasmota.ino
E:\Arduino\arduino-builder -compile -logger=machine -hardware E:\Arduino\hardware -hardware C:\Users\Home\AppData\Local\Arduino15\packages -tools E:\Arduino\tools-builder -tools E:\Arduino\hardware\tools\avr -tools C:\Users\Home\AppData\Local\Arduino15\packages -built-in-libraries E:\Arduino\libraries -libraries C:\Users\Home\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:xtal=80,vt=flash,exception=disabled,ssl=all,eesz=4M,led=2,ip=lm2f,dbg=Disabled,lvl=None____,wipe=sdk,baud=115200 -vid-pid=1A86_7523 -ide-version=10813 -build-path C:\Users\Home\AppData\Local\Temp\arduino_build_615351 -warnings=none -build-cache C:\Users\Home\AppData\Local\Temp\arduino_cache_667426 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.python3.path=C:\Users\Home\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\Home\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Home\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-4-b40a506.path=C:\Users\Home\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506 -prefs=runtime.tools.mklittlefs.path=C:\Users\Home\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.mklittlefs-2.5.0-4-fe5bb56.path=C:\Users\Home\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\2.5.0-4-fe5bb56 -prefs=runtime.tools.mkspiffs.path=C:\Users\Home\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -prefs=runtime.tools.mkspiffs-2.5.0-4-b40a506.path=C:\Users\Home\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-4-b40a506 -verbose C:\Users\Home\Documents\Arduino\Tasmota-8.5.1\Tasmota-8.5.1\tasmota\tasmota.ino
Using board 'nodemcuv2' from platform in folder: C:\Users\Home\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4
Using core 'esp8266' from platform in folder: C:\Users\Home\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4
Detecting libraries used...
"C:\\Users\\Home\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-4-b40a506/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\Home\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/include" "-IC:\\Users\\Home\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/lwip2/include" "-IC:\\Users\\Home\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\Home\\AppData\\Local\\Temp\\arduino_build_615351/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=gnu++11 -ffunction-sections -fdata-sections -fno-exceptions -DNEW_OOM_ABORT -w -x c++ -E -CC -DNONOSDK22x_190703=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10813 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_NODEMCU\"" -DLED_BUILTIN=2 -DFLASHMODE_DIO -DESP8266 "-IC:\\Users\\Home\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4\\cores\\esp8266" "-IC:\\Users\\Home\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.7.4\\variants\\nodemcu" "C:\\Users\\Home\\AppData\\Local\\Temp\\arduino_build_615351\\sketch\\tasmota.ino.cpp" -o  [... cut ...]
Using library ESP8266HTTPClient at version 1.2 in folder: C:\Users\Home\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266HTTPClient 
Using library ESP8266httpUpdate at version 1.3 in folder: C:\Users\Home\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266httpUpdate 
Using library ArduinoJsonParser-master in folder: C:\Users\Home\Documents\Arduino\libraries\ArduinoJsonParser-master (legacy)
exit status 1
JsonGenerator.h: No such file or directory
Specified folder/zip file does not contain a valid library

geyseran, 2020-10-10:
I am trying to compile the latest Tasmota v8.5.1 from source, but compilation fails with the following error:
C:\Users\Home\Documents\Arduino\Tasmota-development\Tasmota-development\tasmota\support.ino: In function 'bool JsonTemplate(char*)':
support:1513:14: error: missing template arguments before 'parser'
   JsonParser parser((char*) dataBuf);
              ^
support:1513:14: error: expected ';' before 'parser'
support:1514:3: error: 'JsonParserObject' was not declared in this scope
   JsonParserObject root = parser.getRootObject();
   ^
support:1514:20: error: expected ';' before 'root'
   JsonParserObject root = parser.getRootObject();
                    ^
support:1515:8: error: 'root' was not declared in this scope
   if (!root) { return false; }
        ^
support:1518:3: error: 'JsonParserToken' was not declared in this scope
   JsonParserToken val = root[PSTR(D_JSON_NAME)];
   ^
support:1518:19: error: expected ';' before 'val'
   JsonParserToken val = root[PSTR(D_JSON_NAME)];
''''''
exit status 1
missing template arguments before 'parser'

I have installed the following libraries:
ArduinoJson v5.13.2
ArduinoJsonParser-master
EspAsyncWiFiManager-master
EspAsyncWebserver-master
UniShox_Arduino_Progmem_lib_master
UniShox_Progmem_Decompressor

Any information would be greatly appreciated.

s-hadinger, 2020-10-10:
You shouldn't install any library, they are all included in the Tasmota project. Just use Platform.io and follow instructions here: https://tasmota.github.io/docs/Compile-your-build/