arendst/Tasmota #3027: Upgrading from 5.2.1

tam481, 2018-06-20:
Hello all,
It's been a while since I looked at this. I downloaded the most recent version v6.0.0a. Compiled it after setting my own config override and uploaded the resulting bin to my Sonoff Basic and it restarted but I don't notice anything different. The version still shows 5.2.1 but the number of flashes is incremented as expected. What am I doing wrong?

Tried reset, reset 1 and reset 2

01:09:37 CMND: reset 2
01:09:37 RSLT: stat/TableLamp/RESULT = {"Reset":"Erase, Reset and Restarting"}
01:18:18 CMND: reset 1
01:18:18 RSLT: stat/TableLamp/RESULT = {"Reset":"Reset and Restarting"}

Thanks in advance

ascillato, 2018-06-20:
Hi, can you post all the information required in the troubleshooting template?

> Make sure these boxes are checked [x] before submitting your issue - Thank you!
> 
> - [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
> - [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
> - [ ] Searched the problem in the forum (https://forum.digital-pig.com/)
> - [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
> - [ ] Hardware used : 
> - [ ] Provide the output of command ``status 0`` :
> ```
> STATUS 0 OUTPUT HERE
> ```
> 
> **(Please, remember to close the issue when the problem has been addressed)**

curzon01, 2018-06-21:
Unfortunately you did not fill the information from issue template but I assume your binary is bigger than 512 kB so update failed.

Use minimal version first, the upgrade to your full - see [Troubleshooting - Update over WebUI not working](https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#update-over-webui-not-working)

tam481, 2018-06-21:
01:08:14 Upload: File TableLamp.ino.generic.bin ...
01:08:20 Upload: Successful 504656 bytes. Restarting
01:09:37 CMND: reset 2
01:09:37 RSLT: stat/TableLamp/RESULT = {"Reset":"Erase, Reset and Restarting"}
01:18:18 CMND: reset 1
01:18:18 RSLT: stat/TableLamp/RESULT = {"Reset":"Reset and Restarting"}
16:55:38 CMND: status 0
16:55:38 RSLT: stat/TableLamp/STATUS = {"Status":{"Module":1, "FriendlyName":"Table Lamp", "Topic":"TableLamp", "ButtonTopic":"0", "Power":0, "PowerOnState":3, "LedState":1, "SaveData":1, "SaveState":1, "ButtonRetain":1, "PowerRetain":1}}
16:55:38 RSLT: stat/TableLamp/STATUS1 = {"StatusPRM":{"Baudrate":115200, "GroupTopic":"Lights", "OtaUrl":"http://192.168.2.18/TableLamp.ino.generic.bin", "Uptime":1193, "Sleep":250, "BootCount":7, "SaveCount":217}}
16:55:38 RSLT: stat/TableLamp/STATUS2 = {"StatusFWR":{"Program":"5.2.1", "BuildDateTime":"2017-07-26T17:10:09", "Boot":31, "Core":"2_3_0", "SDK":"1.5.3(aec24ac9)"}}
16:55:38 RSLT: stat/TableLamp/STATUS3 = {"StatusLOG":{"Seriallog":0, "Weblog":2, "Syslog":0, "LogHost":"domus1", "SSId1":"TheWifi", "SSId2":"TheWifi", "TelePeriod":60}}
16:55:38 RSLT: stat/TableLamp/STATUS4 = {"StatusMEM":{"ProgramSize":435, "Free":504, "Heap":27, "ProgramFlashSize":1024, "FlashSize":1024, "FlashMode":0}}
16:55:38 RSLT: stat/TableLamp/STATUS5 = {"StatusNET":{"Host":"TableLamp-5997", "IP":"192.168.2.10", "Gateway":"192.168.2.1", "Subnetmask":"255.255.255.0", "DNSServer":"192.168.2.1", "Mac":"18:FE:33:CB:F8:6D", "Webserver":2, "WifiConfig":4}}
16:55:38 RSLT: stat/TableLamp/STATUS6 = {"StatusMQT":{"Host":"192.168.2.20", "Port":1883, "ClientMask":"DVES_%06X", "Client":"DVES_CBF76D", "User":"mqttUser", "MAX_PACKET_SIZE":512, "KEEPALIVE":120}}
16:55:38 RSLT: stat/TableLamp/STATUS7 = {"StatusTIM":{"UTC":"Thu Jun 21 15:55:38 2018", "Local":"Thu Jun 21 16:55:38 2018", "StartDST":"Sun Mar 25 02:00:00 2018", "EndDST":"Sun Oct 28 03:00:00 2018", "Timezone":99}}
16:55:38 RSLT: stat/TableLamp/STATUS10 = {"StatusSNS":{"Time":"2018-06-21T16:55:38", "DHT22":{"Temperature":24.80, "Humidity":25.90}, "TempUnit":"C"}}
16:55:38 RSLT: stat/TableLamp/STATUS11 = {"StatusSTS":{"Time":"2018-06-21T16:55:38", "Uptime":1193, "Vcc":3.175, "POWER":"OFF", "Wifi":{"AP":1, "SSID":"TheWifi", "RSSI":98, "APMac":"34:CE:00:19:15:2A"}}}

ascillato2, 2018-06-21:
Hi,

The v6.0.0 is bigger than your v5.2.1

Please, try as @curzon01 has explained. First the minimal and then the full version.

Also take into account that, as it is an old version, you may loose your actual config. Several memory changes were made since your version. If you experiment some issues after upgrading, try erase all flash by wire and then flash the last version.

Thanks

tam481, 2018-06-22:
Hi guys,
I've tried that. Tried he minimal and it said it said it was successful. I think updated using the latest one and in all cases nothing happened.

23:58:45 Upload: File sonoff-minimal.bin ...
23:58:51 Upload: Successful 436944 bytes. Restarting
23:59:22 CMND: reset 2
23:59:22 RSLT: stat/TableLamp/RESULT = {"Reset":"Erase, Reset and Restarting"}
18:14:59 Upload: File TableLamp.ino.generic.bin ...
18:15:05 Upload: Successful 504656 bytes. Restarting

ascillato, 2018-06-22:
So, OTA is failing. It is not giving you errors but it is not being done. May be a solution is to flash the new version by wire. Sorry.

tam481, 2018-06-25:
Flashed by wire and all is well now. Memory usage decreased from 29k to 17k. Not sure why. Possibly due to the timer functionality.

Thank you all

ascillato, 2018-06-25:
Great! Happy to know that it is working for you. Please close this issue. Thanks

closed 2018-06-25 by tam481