arendst/Tasmota #3028: MHZ-19B Sensor Stops Reporting After 6-8 hours.

newzealandpaul, 2018-06-21:
I am struggling to get my MHZ-19B Sensor to work with Sonoff-Tasmoto on a Wemos D1 Mini board (possibly a clone board, not sure).

After a number of hours it stops working and reports 0 ppm and 0 temperature. Reseting the Wemos (or rebooting via. the web interface) does not fix it. Only powering it down by unplugging the USB cable will reset it. 

I originally had it on the RX/TX pins, but move it to other pins after [reading this issue](https://github.com/arendst/Sonoff-Tasmota/issues/2659) I switched them to other pins (see my configuration below). Before I switched pins it was going to zero far faster. I thought I had solved the problem but unfortunately consistently stops working after a number of hours. 

I tried turning off Automatic Calibration (sent command Sensor15 2) but that has not solved it. 

Any ideas anyone? Thanks!

[image]
[image]
[image]

- [ x] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ x] Hardware used : Wemos d1 mini (possibly clone)
- [x ] Provide the output of command ``status 0`` :
```
03:18:27 MQT: stat/sonoff-co2/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff"],"Topic":"co2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
03:18:27 MQT: stat/sonoff-co2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"External System","Uptime":"0T16:14:33","StartupUTC":"2018-06-20T10:03:54","Sleep":0,"BootCount":24,"SaveCount":38,"SaveAddress":"FB000"}}
03:18:27 MQT: stat/sonoff-co2/STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018-06-19T04:40:42","Boot":31,"Core":"2_4_1","SDK":"2.2.1(cfd48f3)"}}
03:18:27 MQT: stat/sonoff-co2/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"SysLog":0,"LogHost":"192.168.xx.xx","LogPort":xxx,"SSId":["xxxx","xxxx"],"TelePeriod":60,"SetOption":["00008009","55818000"]}}
03:18:27 MQT: stat/sonoff-co2/STATUS4 = {"StatusMEM":{"ProgramSize":533,"Free":468,"Heap":17,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3}}
03:18:27 MQT: stat/sonoff-co2/STATUS5 = {"StatusNET":{"Hostname":"co2-7317","IPAddress":"192.168.xx.xx","Gateway":"192.168.xx.xx","Subnetmask":"255.255.255.0","DNSServer":"192.168.xx.xxx","Mac":"xxxxxxxxxxxx","Webserver":2,"WifiConfig":3}}
03:18:27 MQT: stat/sonoff-co2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.xx.xx","MqttPort":xxx,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_359C95","MqttUser":"xxx","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
03:18:27 MQT: stat/sonoff-co2/STATUS7 = {"StatusTIM":{"UTC":"Thu Jun 21 02:18:27 2018","Local":"Thu Jun 21 03:18:27 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:46","Sunset":"20:57"}}
03:18:27 MQT: stat/sonoff-co2/STATUS10 = {"StatusSNS":{"Time":"2018-06-21T03:18:27","MHZ19B":{"CarbonDioxide":0,"Temperature":0.0},"TempUnit":"C"}}
03:18:27 MQT: stat/sonoff-co2/STATUS11 = {"StatusSTS":{"Time":"2018-06-21T03:18:27","Uptime":"0T16:14:33","Vcc":2.940,"Wifi":{"AP":1,"SSId":"xxxx","RSSI":56,"APMac":"24:A4:3C:29:42:60"}}}
```

Frogmore42, 2018-06-21:
You need to have it on the real serial pins according to the other issue.
GPIO 1 to 57MHZ Rx
GPIO 3 to 56MHZ Tx

And you need to disable serial logging.

Frogmore42, 2018-06-21:
But you have a Wemos so that might not work, since the serial lines are connected to the USB chip.  This will make it more complicated.

newzealandpaul, 2018-06-21:
So if I disconnect the pins going from USB is should work? Any downsides apart from not being able to flash any alternative firmware onto it in the future?

Frogmore42, 2018-06-21:
There are the alternative serial lines, which are discussed in the other issue.  I have not used them, so don't know exactly what you need to do.

Frogmore42, 2018-06-22:
Glad it is working for you and good job on desoldering.  That is not a task I would expect many people to be able to do well.

newzealandpaul, 2018-06-24:
Been working fine now for over 2 days. Thanks to @Frogmore42 for the great advice. 

The only slightly strange thing that happened is once it kept going into flash mode, and I had to disconnect the MHZ when powering it up. But **hasn't** happened since.

closed 2018-06-24 by newzealandpaul

quintendewilde, 2020-10-08:
@newzealandpaul 

Hi I want to achieve the same thing you did with the wemos. Is it possible you can provide step by step how to of some sort to get this to work. Because I'm confused by this thread on what to do and what not to do... :D

newzealandpaul, 2020-10-08:
Just wire it straight to the wemos. Install tasmota and set it up. It's
honestly very simple.

I then 3D printed a case.

quintendewilde, 2020-10-08:
Any idea to do this without tasmota because if I'm not mistaken Tasmota doesn't have OSC communication and I need this for my project..

I'll also have to search for the same wemos to desolder the same parts