arendst/Tasmota #3029: Support for PIR AM312 Sensor

akpuggy, 2018-06-21:
I bought a couple of PIR AM312 motion sensors thinking that I would easily be able to add them to my Sonoff Basic but couldn't find this sensor in the list of available for GPIO14 and there is nothing in the wiki about it.

The goal was to attach them to the Sonoff's inside light switches so they can turn the lights on or off based on motion level.

Thanks!

Since these are rated to have a working voltage of 2.7-12V, shouldn't there be a way to use these with the Sonoff Basic with Tasmota?

[image]

[image]

lobradov, 2018-06-21:
There's nothing special about this sensor - it acts as a regular switch, setting output to HIGH when theres someone in the room or... well, you guess it.

Output can be a bit flappy especially on a longer line, so you might want to add pulldown to make it stable. 

Note that these mini PIRs have no settings (sensitivity, duration...) so might not work perfectly for all your use cases.

akpuggy, 2018-06-21:
Thanks for that feedback. I was hoping to get it to turn on the light if there was motion and then turn it off if no motion for 10 minutes. After testing it as a regular switch it just goes to off state in a few seconds after the last movement, so It may not be as useful as I thought. 

Do you know if the bigger HC-SR501 would have the sensitivity and duration options that you mentioned?

wunderbaum, 2018-06-21:
Use the pulsetime command of tasmota.

ascillato2, 2018-06-21:
Hi,

A PIR can be used as a switch.

See Switchtopic command at wiki for your desired behavior: https://github.com/arendst/Sonoff-Tasmota/wiki/Understanding-SwitchMode-and-SwitchTopic

There is also an explanation video for that.

For more information on how to connect a PIR see tutorials.

Thanks.

closed 2018-06-21 by ascillato2

rominator1983, 2019-11-23:
For anyone landing here from google and seeing a dead link: Go to https://github.com/arendst/Tasmota/wiki/Buttons-and-switches

sfromis, 2020-03-08:
For anyone landing here from Google, and wanting the current link, go to
https://tasmota.github.io/docs/#/Buttons-and-Switches

tsaost, 2021-06-30:
You can extend the time by adding a few components:

1. A MOSFET as a switch to turn the light on/off.  You need this MOSFET if you want to drive a LED that requires more current than what can be supplied by the AM312's output pin  (I use AO3400 A09T N-Channel MOSFET)
2. A Capacitor to hold the MOSFET's gate voltage (this will control how long the light will stay on, a large 10uF can hold the light on for minutes).
3. A diode with a low reverse current to prevent the capacitor from being discharged through the AM312 input pin (I used a 1N4148 with a reverse current of 5uA)

Alternatively, you can use the AM612, which you can buy for about $10 for 10 sensors (but you'll need to add voltage regulators, Fresnel len, etc)

prabbit237, 2023-03-29:
> Thanks for that feedback. I was hoping to get it to turn on the light if there was motion and then turn it off if no motion for 10 minutes. After testing it as a regular switch it just goes to off [...]

Do it in software;

rule 1;
=======================
On  button2=1
      backlog power2=1; timer2=600                  #10 minutes
endon

on time2=1
    power2=0
endon
======================

(I may have the syntax a bit off.)

When it triggers, t sets the timer, If it ever hits 1, it turns stuff off. as long as someone moves around enough, it keeps resetting the timer.

barbudor, 2023-03-29:
Syntax is totally off, as well as the logic:
A button never gets a 0 (off) or 1 (on) state. Only 2 (toggle) so your initial trigger will not work

Also tasmota includes a feature for that: PulseTime

sfromis, 2023-03-29:
That syntax is so bad that it looks like coming from ChagGPT :grimacing: (or maybe worse)

prabbit237, 2023-03-29:
> Syntax is totally off, as well as the logic: A button never gets a 0 (off) or 1 (on) state. Only 2 (toggle) so your initial trigger will not work
> 
> Also tasmota includes a feature for that: PulseTime

My example was totally off the cuff and when I was tired and yes, it wasn't quite correct (I left out the "DO", the "power2=1" should have been "power2 1" or "power2 on" as well as the other = symbols that should have been spaces, etc.) The rules should have been more like:

````  
ON button1#state 
DO Backlog Power1 on; RuleTimer1 600 
ENDON

ON Rules#Timer=1 
DO Power1 off 
ENDON
````

(and that's from the rules examples page with one tiny change.) 

And yes, setting PulseTime would do the same but only if you're wanting to turn on and off a relay, where-as using the timer method could be used for anywhere that you need a retrigger-able delay. Maybe you want the delay to do a **Power1 off** and then turn it back on at the end of the delay. Or you want to change an LED's color. Or you want to sent a MQTT message. Or you want to......whatever.

As to the "A button never gets a 0 (off) or 1 (on) state" part, 
````
Button2#State    when a button changes state:
       0 = OFF  
       1 = ON  
       2 = TOGGLE  
       3 = HOLD
````
(Note; only works for buttons >= 2. Button1 is special in this regard.) Or you define it as a switch and not a button. Then set SwitchMode 13, where pressing it turns it on and then it goes off at PulseTime, etc.

The main point was that there's any number of ways to use the motion sensor and to be able to stretch out the action time far beyond whatever the on-time is on the sensor itself.

barbudor, 2023-03-29:
If that's from the doc, 0 and 1 are wrong and needs fixing
But indeed 2 for toggle, 3 for hold and over for multipress
While a switch can have states 0 and 1, a button can't