arendst/Tasmota #3030: Set Password via HTTP requests Commands char # (hash) problem

araczkowski, 2018-06-21:
**Describe the bug**

it is not possible to set the correctly STA_PASS1 or STA_PASS2 via HTTP requests if the password has character #
in such case the first part of the password (string before character #) is saved as an password

- [x] Searched the problem in issues and in the wiki
- [x] Hardware used : 
- [x] Provide the output of command ``status 0`` :
```
07:38:19 MQT: stat/dom_s20_24B87B/STATUS = {"Status":{"Module":8,"FriendlyName":["Gniazdo najdlej od WiFi"],"Topic":"dom_s20_24B87B","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
07:38:19 MQT: stat/dom_s20_24B87B/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"dom","OtaUrl":"http://ai-speaker.com/ota/s20/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:06:40","StartupUTC":"2018-06-21T06:31:39","Sleep":0,"BootCount":63,"SaveCount":167,"SaveAddress":"FB000"}}
07:38:20 MQT: stat/dom_s20_24B87B/STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018-06-20T09:34:34","Boot":31,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
07:38:20 MQT: stat/dom_s20_24B87B/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Andrzej-test","dom"],"TelePeriod":300,"SetOption":["000A8009","55818000"]}}
07:38:20 MQT: stat/dom_s20_24B87B/STATUS4 = {"StatusMEM":{"ProgramSize":501,"Free":500,"Heap":20,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
07:38:20 MQT: stat/dom_s20_24B87B/STATUS5 = {"StatusNET":{"Hostname":"dom_s20_24B87B-6267","IPAddress":"192.168.1.125","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.1","Mac":"B4:E6:2D:24:B8:7B","Webserver":2,"WifiConfig":2}}
07:38:20 MQT: stat/dom_s20_24B87B/STATUS6 = {"StatusMQT":{"MqttHost":"ais-dom","MqttPort":1883,"MqttClientMask":"DOM_%06X","MqttClient":"DOM_24B87B","MqttUser":"DOM_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
07:38:20 MQT: stat/dom_s20_24B87B/STATUS7 = {"StatusTIM":{"UTC":"Thu Jun 21 06:38:20 2018","Local":"Thu Jun 21 07:38:20 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:46","Sunset":"20:57"}}
07:38:20 MQT: stat/dom_s20_24B87B/STATUS10 = {"StatusSNS":{"Time":"2018-06-21T07:38:20"}}
07:38:20 MQT: stat/dom_s20_24B87B/STATUS11 = {"StatusSTS":{"Time":"2018-06-21T07:38:20","Uptime":"0T00:06:41","Vcc":3.208,"POWER":"ON","Wifi":{"AP":1,"SSId":"Andrzej-test","RSSI":94,"APMac":"00:12:2A:F2:43:38"}}}
```

**To Reproduce**
open in the browser web page served by your sonoff-tasmota device 
http://<your_device_ip>
go to http://<your_device_ip>/cm?cm?cmnd=Password2%20test1#test2
only the first part of the test1#test2 password is stored as STA_PASS2 -> test1

**Expected behavior**
the provided password test1#test2 should be correctly stored as STA_PASS2

**Screenshots**
[image]

**Additional context**
I think this is somewhere here 
https://github.com/arendst/Sonoff-Tasmota/blob/d4119fe54d46debbf494726a28228bb5ec537473/sonoff/sonoff.ino#L941

I can try to fix this, test and deliver the correction back but please point me where exactly is the problem Thank you all for this great software! :)

curzon01, 2018-06-21:
Tis is a browser issue - you have to mask special chars like # same as e. g. space (# is 23 hex, so use %23 instead of #)

`http://<your_device_ip>/cm?cmnd=Password2%20test1%23test2`

araczkowski, 2018-06-21:
you are right :)
thanks a lot @curzon01 !

closed 2018-06-21 by araczkowski