arendst/Tasmota #3032: Counter2 in Domoticz

zuevav, 2018-06-21:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [ X] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [ X] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ X] Searched the problem in the forum (https://forum.digital-pig.com/)
- [ X] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [ X] Hardware used : Sonoff Basic
- [ X] Provide the output of command ``status 0`` :
```
STATUS 0 OUTPUT HERE
```

Hi, can you help me? how to see Counter2 in Domoticz, I see special line in firmware for Counter1, but for Counter2 - no. 

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2018-06-26:
Use rules. See wiki - rules about an example using temperatures send to domoticz

zuevav, 2018-06-28:
Thank you, I do it!
Console command:
rule on
rule on tele-COUNTER#C2 do %value% do publish domoticz/in {"idx":23,"svalue":"%value%"} endon

closed 2018-06-28 by zuevav