arendst/Tasmota #3034: Sonoff-Tasmota sonoff.bin DS18B20 not showing accurate temperature

peteeee, 2018-06-21:
Hardware SonOff basic WiFi module testing connecting to:
1 - GPIO 4
2 - GPIO 14

~ 12" leads to DS18B20

DQ-VCC 10K Resistor

Program Version 5.14.0
Build Date & Time 2018-05-15T15:29:54
Core/SDK Version 2_3_0/1.5.3(aec24ac9)

Console shows:
DS18B20 Temperature 38°F
6:35:40 MQT: GarageDoor1/SENSOR = {"Time":"2018-06-21T16:35:40","DS18B20":{"Temperature":38},"TempUnit":"F"}

Inside home testing and actual temperature is around 70 °F

If I upload sonoff-allsensors.bin to the Sonoff Basic Wifi temperature reads are correct.

DS18S20 Temperature 71°F
16:41:19 MQT: GarageDoor1/SENSOR = {"Time":"2018-06-21T16:41:19","DS18S20":{"Temperature":71},"TempUnit":"F"}

Frogmore42, 2018-06-22:
10K is kind of high for the pull up, but it still should work.

There are about 4 combinations of settings for the DS18B20.  I always use the one that supports many and have not had an issue.

stefanbode, 2018-06-22:
There is a big difference between the TWO logs. One is reporting the B Version. One the S version. I assume you have the S version because this gives correct results. DS18S20 <-> DS18B20. if you correct this one, it should work.

Frogmore42, 2018-06-22:
Yes, you have the "S" version of the sensor.  That sensor is not supported (correctly) by the code that supports a single "B" sensor.  Use the code that supports multiple sensors and things work correctly as you saw.

ascillato2, 2018-06-29:
Hi,

If your issue is solved, please close it. Thanks :+1:

closed 2018-07-10 by ascillato2

peteeee, 2018-11-04:
Thank you ascillator2 for closing this issue.  I did update the Tasmota firmware and it does work fine now.