arendst/Tasmota #3035: Can't set Arduino Flash Size to 1M(No-SPIFFS) with 2.3.0"

sparkgap, 2018-06-22:
After selecting 2.3.0 while installing Generic ESP8266 Board Manager, the Tool drop down menu does not allow 1M(no SPIFFS).  I had to select 1M(64K SPIFFS).   I was told that this will cause problems then doing OTA Updates, and that I should go to platformIO to get (no SPIFFS).   I have on two occasions tired to use platformIO with great frustration and failure due to the busy and complicated GUI).   I have several years of experience invested in Arduion IDE and do not wish to learn a new IDE. 

Does anyone know how to get 1M(no-SPIFFS) with Aduino IDE ???

Will  1M(64K-SPIFFS)  in all cases cause OTA to fail ???

Frogmore42, 2018-06-22:
Did you miss this section of the instructions for Arduino?

Copy files
ESP8266 Board version before 2.4.0
If not available copy from the Tasmota release Source code folder arduino\version 2.3.0\tools\sdk\ld file eagle.flash.1m0.ld to Arduino IDE folder portable\packages\esp8266\hardware\esp8266\2.3.0\tools\sdk\ld.
Replace in Arduino IDE folder portable\packages\esp8266\hardware\esp8266\2.3.0 file boards.txt with the Tasmota Source code file arduino\version 2.3.0\boards.txt.

Look in Theo's tips for the details on the structure of the memory over the different versions of Tasmota.  This explains where stuff is stored and how settings are remembered when flashing an update.

I don't know the details of how OTA is being used, but I would trust people when they say OTA doesn't work.  If it does work for you, consider yourself luck.  Your luck might continue for several updates.  But, it might change at any time.  You will be able to do a flash by serial, though your settings might not be preserved.

ascillato, 2018-06-22:
Arduino configuration: https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE

ascillato, 2018-06-22:
Hi,

If your question is addressed, please close this issue. Thanks :+1:

Frogmore42, 2018-06-23:
I went back and re-read the other issue you created when you first ran into this.  The instructions in the Wiki are not always the easiest to understand and usually require some thought on what is really meant.  The instructions include the paths that might be used for a "portable" installation of Arduino, but there is more than one way to install Arduino.  The instructions are also not perfect English, but then who ever writes that anymore :-).

Arduino with core 2.3.0 does NOT have the correct loader file, which is why you can't select 1M(No-SPIFFS).  You have to add the loader file, which is included with the Sonoff code on GitHub, to the correct location in your install of the Arduino code.  For someone that is reasonably well versed in software development this is simple.  But, for someone who has only ever used Arduino and is not experienced in software development, it could be hard to understand exactly what was required.

In your earlier issue, I believe you had core 2.4.0 installed and did not see anything with 2.3.0 in your installation, so you assumed that the instructions were wrong.  They were/are not wrong, but they did not match your installation.  Later, you installed core 2.3.0, and it "worked" for you with the wrong linker model. That is one of the things that can be frustrating about software development.  It is really detail oriented, but sometimes "works" when you get some little detail incorrect.

Here is what the wiki says now:
If not available copy from the Tasmota release Source code folder arduino\version 2.3.0\tools\sdk\ld file eagle.flash.1m0.ld to Arduino IDE folder portable\packages\esp8266\hardware\esp8266\2.3.0\tools\sdk\ld.
Replace in Arduino IDE folder portable\packages\esp8266\hardware\esp8266\2.3.0 file boards.txt with the Tasmota Source code file arduino\version 2.3.0\boards.txt.

Here is perhaps what it might say to be easier to understand:
If the 1M(No-SPIFFS) option is not available, copy from the Tasmota release Source code folder arduino\version 2.3.0\tools\sdk\ld file eagle.flash.1m0.ld to your Arduino IDE folder. 
If you have a portable installation it will be here:
portable\packages\esp8266\hardware\esp8266\2.3.0\tools\sdk\ld.
If you have a regular installation (non-portable version, i.e. the zip version) it will usually be here: C:\Program Files (x86)\Arduino\packages\esp8266\hardware\esp8266\2.3.0\tools\sdk\ld

In addition, you need to also update the boards.txt file
Replace in Arduino IDE folder:
<Arduino location>\packages\esp8266\hardware\esp8266\2.3.0\boards.txt
with the Tasmota Source code file:
 arduino\version 2.3.0\boards.txt.

Does that make it easier to understand what you must do to use core 2.3.0 with Arduino?

sparkgap, 2018-06-24:
It makes perfect sense now.

The following should be added to the second bullet point under **Install ESP bord software**

Open Tools - Boards ... - Boards Manager ... and scroll down and click on esp8266 by ESP8266 Community.  Then Click the Install button to download and install the latest ESP8266 board software. Select Close.

Open Tools - Boards ... - Boards Manager ... and scroll down ### to esp8266 by ESP8266 Community.  Then Click the Install button to download and install the latest ESP8266 board software. Select Close.

closed 2018-06-25 by sparkgap