arendst/Tasmota #3036: Sonoff basic turning light on after a restart ?

c3reeman, 2018-06-22:
Hi, I’ve got WiFiconfig set to 0 and also tried WiFiconfig 4 but what ever I seem to try the sonoff basic turned on if it restarts due to bad WiFi below consol information shows a forced restart by me, you can see where It’s turned it’s self on, PowerOnState is set to 3

00:00:00 WIF: Connecting to AP1 Mill House in mode 11N as Sonoff001_ChrisBedsideTable...
00:00:06 WIF: Connected
00:00:06 DNS: Initialized
00:00:06 HTP: Web server active on Sonoff001_ChrisBedsideTable.local with IP address 192.168.86.239
00:00:08 MQT: Attempting connection...
00:00:08 MQT: Connected
00:00:08 MQT: tele/sonoff001/LWT = Online (retained)
00:00:08 MQT: cmnd/sonoff001/POWER = 
00:00:08 MQT: stat/sonoff001/RESULT = {"POWER":"OFF"}
00:00:08 MQT: stat/sonoff001/POWER = OFF
00:00:08 MQT: tele/sonoff001/INFO1 = {"Module":"Sonoff Basic","Version":"5.14.0","FallbackTopic":"DVES_A7CC54","GroupTopic":"sonoffs"}
00:00:08 MQT: tele/sonoff001/INFO2 = {"WebServerMode":"Admin","Hostname":"Sonoff001_ChrisBedsideTable","IPAddress":"192.168.86.239"}
00:00:08 MQT: tele/sonoff001/INFO3 = {"RestartReason":"Software/System restart"}
00:00:08 MQT: stat/sonoff001/RESULT = {"POWER":"ON"}
00:00:08 MQT: stat/sonoff001/POWER = ON
00:00:09 MQT: stat/sonoff001/RESULT = {"POWER":"ON"}
00:00:09 MQT: stat/sonoff001/POWER = ON
00:00:16 MQT: tele/sonoff001/STATE = {"Time":"1970-01-01T00:00:16","Uptime":"0T00:00:16","Vcc":3.179,"POWER":"ON","Wifi":{"AP":1,"SSId":"Mill House","RSSI":40,"APMac":"70:3A:CB:32:67:6A"}}
00:00:16 MQT: tele/sonoff001/SENSOR = {"Time":"1970-01-01T00:00:16","Switch1":"ON"}
06:55:38 MQT: stat/sonoff001/RESULT = {"POWER":"OFF"}
06:55:38 MQT: stat/sonoff001/POWER = OFF

c3reeman, 2018-06-22:
Ok little more info, it’s very clear that it’s mqtt causing the problems, if I disable mqtt on the sonoff and restart I note that what ever Powerstate it had is retained, I’m using home assistant

andrethomas, 2018-06-22:
Check state of Setption0 by typing it into the console.

It should be set to 0 if you want it to ignore the previous state so if it does not use the command
SetOption0 0
from console.

stefanbode, 2018-06-22:
I think this could also be an "issue" with the Switch1. depending on the configuration the switch will drive the relay. Try to disable the switch and check, if you still have the unwanted behavior.

Jason2866, 2018-06-22:
I am pretty sure HA has retained messages.
Search the issues here. It is many times discussed.
My 2 cents: using retain function is crab....

Frogmore42, 2018-06-22:
https://github.com/arendst/Sonoff-Tasmota/issues/2929

closed 2018-07-10 by ascillato2