arendst/Tasmota #3037: Please update some typedefinitions to power_t

stefanbode, 2018-06-22:
Hi @arendst,

as you know I support the 32 relay version. power_t was already introduced to make it flexible to change. I found 2 places where the typedef needs to be changed to power_t. See below. Thanks.

`boolean XdrvCommand(uint8_t grpflg, char *type, power_t index, char *dataBuf, uint16_t data_len, int16_t payload, uint16_t payload16)`

```
struct XDRVMAILBOX {
  uint16_t      valid;
  power_t      index;
  uint16_t      data_len;
  uint16_t      payload16;
  int16_t       payload;
  uint8_t       grpflg;
  uint8_t       notused;
  char         *topic;
  char         *data;
} XdrvMailbox;
```

Frogmore42, 2018-06-22:
Pull request?

stefanbode, 2018-06-22:
my code is too different to easily create a pull request. Take it or leave it.

Frogmore42, 2018-06-22:
Git supports branches, which would make it easy to make the change you are requesting.  If you are reporting a bug in the standard code it is not clear what impact it has.

closed 2018-07-03 by stefanbode