arendst/Tasmota #3038: Sonoff Rebooting!

Lawrencezarb, 2018-06-22:
I have one sonoff basic that ramdomly keep rebooting.

I have set wificonfig = 4. 

The Syslog information at reboot show this:

22/06/2018 09:49 | sonoff-5212 ESP-WIF: Checking connection...
-- | --
22/06/2018 09:49 | sonoff-5212 ESP-WIF: Connected
22/06/2018 09:50 | sonoff-5212 ESP-WIF: Checking connection...
22/06/2018 09:50 | sonoff-5212 ESP-WIF: Connected
22/06/2018 09:50 | sonoff-5212 ESP-WIF: Checking connection...
22/06/2018 09:50 | sonoff-5212 ESP-WIF: Connected
22/06/2018 09:50 | sonoff-5212 ESP-RSL: INFO1 = {"Module":"Sonoff   Basic","Version":"6.0.0a","FallbackTopic":"DVES_41F45C","GroupTopic":"sonoffs"}
22/06/2018 09:50 | sonoff-5212 ESP-RSL: INFO2 =   {"WebServerMode":"Admin","Hostname":"sonoff-5212","IPAddress":"192.168.0.26"}
22/06/2018 09:50 | sonoff-5212 ESP-RSL: INFO3 = {"RestartReason":"Power   on"}
22/06/2018 09:50 | sonoff-5212 ESP-WIF: Checking connection...
22/06/2018 09:50 | sonoff-5212 ESP-WIF: Connected
22/06/2018 09:50 | sonoff-5212 ESP-DNS: Initialized
22/06/2018 09:50 | sonoff-5212 ESP-HTP: Web server active on sonoff-5212.local with IP   address 192.168.0.26
22/06/2018 09:50 | sonoff-5212 ESP-APP: (UTC) Fri Jun 22 08:50:50 2018, (DST) Sun Mar 25   02:00:00 2018, (STD) Sun Oct 28 03:00:00 2018
22/06/2018 09:50 | sonoff-5212 ESP-RSL: RESULT = {"POWER":"OFF"}
22/06/2018 09:50 | sonoff-5212 ESP-RSL: POWER = OFF
22/06/2018 09:50 | sonoff-5212 ESP-RSL: STATE =   {"Time":"2018-06-22T09:50:58","Uptime":"0T00:00:17","Vcc":3.447,"POWER":"OFF","Wifi":{"AP":2,"SSId":"Private   Network","RSSI":64,"APMac":"D8:EB:97:15:9E:B7"}}
22/06/2018 09:50 | sonoff-5212 ESP-RSL: SENSOR =   {"Time":"2018-06-22T09:50:58","Switch1":"ON"}

Its seems the reboot is initailsed by "Power On"

The unit is permamently wired and there are no power feed issues.

Anyone have any ideas?

curzon01, 2018-06-22:
Like asked in the issue template please provide `Status 0` info

Lawrencezarb, 2018-06-22:
13:16:37 RSL: STATUS = {"Status":{"Module":1,"FriendlyName":["Heating"],"Topic":"sonoff","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
13:16:37 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T01:40:41","StartupUTC":"2018-06-22T10:35:56","Sleep":50,"BootCount":204,"SaveCount":268,"SaveAddress":"F8000"}}
13:16:37 RSL: STATUS2 = {"StatusFWR":{"Version":"6.0.0a","BuildDateTime":"2018-06-07T22:46:45","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}
13:16:37 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":2,"LogHost":"","LogPort":514,"SSId":["Private Network-Plus","Private Network"],"TelePeriod":300,"SetOption":["00008001","55818000"]}}
13:16:37 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":556,"Free":444,"Heap":12,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3,"Features":["00000809","0FDAE794","00000000","17B6179A","00000000"]}}
13:16:37 RSL: STATUS5 = {"StatusNET":{"Hostname":"sonoff-5212","IPAddress":"192.168.0.26","Gateway":"192.168.0.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.0.1","Mac":"5C:CF:7F:41:F4:5C","Webserver":2,"WifiConfig":4}}
13:16:37 RSL: STATUS7 = {"StatusTIM":{"UTC":"Fri Jun 22 12:16:37 2018","Local":"Fri Jun 22 13:16:37 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:46","Sunset":"20:57"}}
13:16:37 RSL: STATUS10 = {"StatusSNS":{"Time":"2018-06-22T13:16:37","Switch1":"ON"}}
13:16:37 RSL: STATUS11 = {"StatusSTS":{"Time":"2018-06-22T13:16:37","Uptime":"0T01:40:41","Vcc":3.447,"POWER":"OFF","Wifi":{"AP":2,"SSId":"Private Network","RSSI":14,"APMac":"4C:60:DE:82:AA:B2"}}}
13:17:08 RSL: STATE = {"Time":"2018-06-22T13:17:08","Uptime":"0T01:41:12","Vcc":3.448,"POWER":"OFF","Wifi":{"AP":2,"SSId":"Private Network","RSSI":18,"APMac":"4C:60:DE:82:AA:B2"}}
13:17:08 RSL: SENSOR = {"Time":"2018-06-22T13:17:08","Switch1":"ON"}

Jason2866, 2018-06-22:
Erase flash completly with esptool. Build firmware with Arduino Core 2.3.0
Now do a clean flash with this firmware.

curzon01, 2018-06-22:
Using library 2.4.x with Sleep <> 0 does not work properly.

Set Sleep to 0 or use Lib 2.3.0

closed 2018-07-10 by ascillato2