arendst/Tasmota #3040: Meaning of LED blinks

rebel7580, 2018-06-22:
- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [ x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [x ] Hardware used : Sonoff Basic
- [ ] Provide the output of command ``status 0`` : Not possible at the moment

I have a Sonoff Basic that I hadn't used for a while. It was flashed OK with 5.13.1. When I powered it up, it seems to be in a loop of some kind. I get the following blinking pattern on the LED:
- blinks 2-4 times/second for about 2 minutes.
- pauses for a second or two.
- Blinks once ofr about one second, then blinks twice quickly.
- pauses for about 20 seconds and repeats.

Once it stopped this cycle and I was able to connect via the web. The restart button was red, so I clicked that. Since then it's been in this loop.

I searched for definitions of LED blink patterns but couldn't find a discussion on that topic.

Not sure what this means...

I assume I'll have to serial reflash the device, but wanted to know where to find the blink descriptions, if they exist somewhere in the Wiki, etc.

Thanks.

rebel7580, 2018-06-23:
After a re-power, the device settled down and was working. However, a subsequent re-power got back to the same loop.

hagenbuch, 2018-06-23:
AFAIK, the 2 minutes of blinking indicates it it is access point mode (no wifi credentials confugured or you pressed the button four times in a row to throw it into AP mode)

Then, you should see an unencrypted AP SSID like "sonoff-1234" and be able to connect to the device in order to configre it. (I think this is part of the genius of tasmota).

rebel7580, 2018-06-23:
I moved the device close to my router, got a stable connection and upgraded to 5.14.0. Seems OK now.

closed 2018-06-23 by rebel7580

yarix, 2022-06-18:
i have (had) the same problem, wasted hours to flash latest (12.01) on SonOff Basic... although flashing is successful, the device won't connect to wifi after after initial configuration. The sonoff will just blink and not connect to the router.
### workaround:
1. start with flashing version 9.3.0 (which seems to work fine) and configure the device.
2. run firmware upgrade from the web http://192.168.1.7/up? 
    this upgraded me to latest 12.0.1 and everything kept working!