arendst/Tasmota #3042: Documentation for developers?

renne, 2018-06-22:
Where can we find API-documentations of Tasmota to develop code (e.g. for I2C sensors)?

andrethomas, 2018-06-22:
I'm not sure there is documentation - what sensor do you want to add?

Have a look at xsns_26_lm75ad.ino which I added to support the LM75AD I2C sensor - Could use it as a template to add others I guess since it conforms to the minimum requirements.

The main thing to remember is that a new sensor would require a new XSNS number so at minimum it would be a new filename for that next number (which currently would be 27 as at the time of writing this) and then just the defines at the top

```
#ifdef USE_I2C
#ifdef USE_LM75AD

```
and basic change at the bottom to use correct number

```
#define XSNS_26

boolean Xsns26(byte function)
{
}

```

And then a definition to include the I2C sensor in user_config.h inside the

```
#define USE_I2C  
#endif USE_I2C

```
section

renne, 2018-06-22:
The sensor is an [Avago APDS9960.](https://docs.broadcom.com/docs/AV02-4191EN) What does  the function `boolean Xsns26(byte function)` do?

andrethomas, 2018-06-22:
What would your application be for this sensor? Looks like an interesting sensor... I should get one.

I should warn you though that there doesn't currently seem to be a way to use active sensors that raise interrupts via an interrupt pin.

The current form of driver implementation is more with passive sensors insofar that the sensor would be read at regular intervals and not necessarily when the sensor demands to be read.

andrethomas, 2018-06-22:
From my testing, it seems Xsns26 to be executed approximately every 50 milliseconds so it should, in theory, be possible to catch an interrupt there I guess.

renne, 2018-06-22:
I'm going to do a closed-loop control of brightness and color temperature for LED-controllers. I also want to use the gestures to control ON/OFF, dimming and color temperature. We have an [issue](https://github.com/arendst/Sonoff-Tasmota/issues/2217) and an old [pull request](https://github.com/arendst/Sonoff-Tasmota/pull/566). The H801 does not have enough GPIOs for an interrupt. So it will be polling.

renne, 2018-06-24:
@andrethomas 
@Staars committed #3051 for the APDS-9960 driver today. If your interested you can give a thumbs up. :)

ascillato2, 2018-07-17:
#3199

Good Job :+1:

closed 2018-07-17 by ascillato2