arendst/Tasmota #3046: Can't setup stock bin without WPS

prabbit237, 2018-06-23:
**Describe the bug**

If you download any of the prebuilt bins and start it up, it tries to go to a stock AP. When it fails, it fails over to WPS connection. Believe it or not, not everyone has a router with that capability. So those people can't use the stock bin but must roll their own.

**To Reproduce**

Install prebuilt bin on a clean ESP8266 device.

**Expected behavior**

Have a fail-over that ANY hardware can connect and start configuring the stock bin, The WIFI_MANAGER option can work on, and be setup with, anything that has a wireless adapter and can load a simple webpage.

**(Please, remember to close the issue when the problem has been addressed)**

curzon01, 2018-06-23:
[Use button](https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage) to get into proper mode after startup the stock bin

After successull connected to your network, you can setup using [wifi commands](https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#wifi) to set your device to your final needs.

It's not neccessary to compile your own version with your own user data. Using this two steps you can always get connected and setup your device to your own needs also with stock firmware.

prabbit237, 2018-06-24:
From that page:
**The main button on a Sonoff-Tasmota flashed module provides the following functionalities:**

If you're not putting this on a Sonoff, then that won't work and you're SOL. The software should provide a way to configure it "out of the box", not "out of the box on only certain devices and only after pressing specific buttons." I'm using it on nodemcu's, for example, and there the user-button is on GPIO16, not GPIO2 (I think that's the one on the Sonoff and I'm being too lazy to confirm.)

WIFI_MANAGER seems to be the only setup method that would work on **all** esp8266 modules/devices/boards that could be flashed (assuming that they have enough flash to even upload the software, thus ruling out older 512K devices.)

reloxx13, 2018-06-24:
We had those discussions already, some preferrer wps, some other wifi manager as default.

just build teh firmware yourself with your preffered default and the problem is solved.

Geitde, 2018-06-27:
I found the easiest way to deal with the issue is to simply grab a spare wifi router or AP and configure it to 

 SSID:  indebuurt1 
 Password: VnsqrtnrsddbrN

So every lost/reset/new tasmota device connects automatically to that accesspoint and you can directly configure it using any phone, tablet or computer.

Just ensure to unplug the AP after you are done for security reasons.

If you don´t have a spare AP, use an existing one and backup the settings, so you can restore them later on.

ascillato, 2018-06-27:
Hi,

@Geitde 

> SSID: indebuurt1
> Password: VnsqrtnrsddbrN

are not there anymore. Now it is blank SSID and PASSWORD.

I think that the best option is to change to WIFI_MANAGER before compilation as @prabbit237 has explained.

A fast solution is as @reloxx13 said. I do the same.

prabbit237, 2018-06-27:
OK, so we have the WPS team starting at the home-team end of the field. The Manager team just came out of the locker room and they're thumping their chests as they walk to the other end of the field in anticipation of the battle to come.

The starting whistle blows and the challenge begins. We see the WPS quarterback, router in hand,grab the Sonoff and he's OFF running and pushes buttons and avoiding the defence and he gets them paired! And the crowd goes absolutely WILD with a score so early in the game.

A missed kick and a fumble gave control back over to the WPS gang where they undertook what many said was impossible. The took a router with NO WSP button and a sonoff basic and ran the field. They lost valuable points/time trying remember how many button presses gets it to MANAGER.

Now the Managers  have possession of the router but....O-M-G, their router still has no WPS button.
 Whatever will they do? Well as the router has been kicked off, quarterback whips out his nodemcu and his cell phone and with a style and grace never seen on this field before, he's able to hook his phone to his nodemcu and then in another blink of the eye move so graceful and that looked almost it came from heaven, he got the nodemcu connected and pushing data like a champ. He then was able to connect a SECOND and then even a ESP-01, which has NO buttons. So by the time he got to the end zone, he'd racked up 3 points, thus winning the game as the stands go WILD!!!!!!
----------------------------------------------------------------------------------
Yes, there's good points and bad for both. But if you want to get people hooked, you need to give them the first fix free. WIFI_MANAGER is about as free and easy it can possibly when you lay down your phone and open the nodemcu from China and then plug it in, connect your phone to it, configure it and then leave it alone, as opposed to unboxing the nodemcu, go out and buy a new router with WPS, switch it all out and then start trying to get the nodemcu connected. (or an edit/recompile ZTo get it connected.

I understand some want it one way and others want it another way. But if set with WIFI_MANAGER as the default and then letting those who already have a WPS router can hit one button (maybe a few times) to change it to their preference.

1 single (or multipress) button on one side vs a full recompile on the other hand.......

Mike A

ascillato, 2018-06-28:
@prabbit237 

hahahahahaha :soccer:

prabbit237, 2018-06-28:
I was about half-asleep when writing that and the typos show. But the main point was that WIFI_MANAGER as the default means "out of the box and connected" with ANY esp8266 hardware and anything that can connect as a client (phone, tablet, computer, etc. You DO have another device to use to configure the esp8266 device with AFTER you get it connected, right?) WIFI_WPS as default means "only connects if you have one of a specific set of esp8266 devices and have a WPS router." 

I see good and bad points for both sides but IMHO, the one that lets someone get up and running the quickest and on the widest range of hardware so that they can spend time actually USING the device and not just trying to get it even connected is the best default. Once the person has had time to check out the software and they decide they like the way it works, they can always tweak it and customize it and recompile it and do whatever it is that they want to with it. But if you have a copy of it that only works with SOME hardware, a person might be more likely to say "heck with it. I'll go try some other firmware instead and not even give it a try."
 
Of course since this isn't being sold, the author can really do anything he wishes with it and "the h*** with what anyone thinks of it" but I'm just putting out my point of view.

Geitde, 2018-07-10:
This will be my last comment here. It does not even matter what I write the stal bot will kill the reports anyway as it did for most other topics.

It is one thing to praise the time developerers spent on a project, but it is another to waste the time of people who want to help. What is the point in reporting anything, if it gets ignored and the reporter has to constantly monitor all reports to keep them alive, while on the other side no one cares about anything, because stalbot will auto bugfix averything. 

Open source development seems to be easy these days. Just take a month vacation and all bug reports are gone.

And yes, I know tasmota is for free. Everyone can do something. Be thankful to the developers. blah blah. And the time and afford people invest in testing and reporting is worth nothing.

This post is just to keep the topic alive! Thanks stalbot!

ascillato, 2018-07-10:
Hi @Geitde

Yes, You are right. The time of everyone is very important. The help given by everyone is very appreciated. Really :+1: . That is why Tasmota is what it is right now. Theo and the rest of the community had spent thousands of hours with this software adding requested features and solving issues. And you are also part of this the community, and of course that your help it is very appreciated.

Sorry to know that you are frustated. The Bot is there because WE CARE and we want to do a follow up to the issues and to solve as many as can be possible. You know that all the time spent here is limited. Not everything can be addressed nor done. **A lot of people care and want to help, but time is limited**. It is our free time that we are spending here trying to solve other people issues and for free. It is time that we are not spending with our family, friends, work, excercise, etc. And by we, I meant everyone included you. So, sometimes, things take longer to be addressed or to start working on.

There were 1248 issues opened the day we put the Bot at work. We did that because that number is impossible to work on and to follow up. With that number, just the last 20 or 30 issues only were the ones that the people of the community were paying attention and willing to solve. The rest were left behind. Now it is 168 opened issues. So it is a better number of issues to do a follow up, to ask and try to solve.

When the Bot started, A lot of issues were tagged, so people retake the issue, continue the thread and a lot of issues were finally solved instead of being left behind and forgotten.

The bot will tag an issue as Stale only after 45 days that nobody writes on it. If after that bot's comment, nobody do nothing, the bot will close the issue after another 15 days. So, The Bot will close an issue after 60 days that nobody continues the thread, ask, give feedback or pay attention to that issue.

Besides, a closed issue can be reopened and also are not locked. Anyone can continue writing there. Also, if anyone performs a search, will see also the closed issues. So your reports will be there for helping others. A lot of people came with a problem or question and after a search they find the answer and leave. So, everything will be there when needed.

If you think that any issue should be reopened, we can do that and also we can pin it telling the bot to not stale it any more. For example an issue with the tag BUG will never have a STALE tag or be closed by the bot. Just ask in any issue you think it is needed, to be pinned and/or reopened. This is work for anyone that wants to help.

In numbers:
* Tasmota has more 1 million downloads
* Tasmota binaries are being downloaded from github more than 2500 times per day (I don't have maddox site numbers but those also add to this)
* Tasmota github is visited by more than 5000 times per day
* Tasmota github has more than 3000 cloning per day
* Tasmota is made mainly by Theo and with the help of 302 persons that I tagged in the 1 millon download issue (some with PR, some with issue solving and some with the wiki - hope I included everyone)
* Tasmota issues are being addressed mainly by 7 active users and by 35 sporadic users.

In conclusion:

We do as much as our free time allow us. Yes, you are right. The Bot closes old issues or issues that are left behind, but also brings attention to those to be retaken and finally solved/answered.
So, If we can encourage more people to help here, I'm sure that we can solve more and more issues, not leaving any other issue unsolved and make Tasmota even more amazing. Please, continue helping to the community. :+1:

ascillato2, 2018-07-14:
@prabbit237

Hi, 

Due to #3221, now WPS is not the default. Closing this issue. Please, reopen if needed. 

Thanks everyone for sharing their opinion.

closed 2018-07-14 by ascillato2