arendst/Tasmota #3047: Sonoff Touch Tasmota pushbutton

ziboo06, 2018-06-23:
Hi everybody,

I flashed my sonoff touch with tasmota and i can't set it in the pushbutton mode. 
I configured the module as a sonoff touch module, put GPIO3 as 09 Switch1and used the command swithmode1 3 through the console.

Can someone help me?
Thx :)

ascillato, 2018-06-23:
Please provide all the information of the troubleshooting template:

> Make sure these boxes are checked [x] before submitting your issue - Thank you!
> 
> - [ ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
> - [ ] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
> - [ ] Searched the problem in the forum (https://forum.digital-pig.com/)
> - [ ] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
> - [ ] Hardware used : 
> - [ ] Provide the output of command ``status 0`` :
> ```
> STATUS 0 OUTPUT HERE
> ```
> 
> **(Please, remember to close the issue when the problem has been addressed)**
>

closed 2018-06-27 by ascillato2

gsaslis, 2020-11-07:
@ziboo06 wondering if you figured this out?