arendst/Tasmota #3048: MQTT over web confuguration seems not working

hagenbuch, 2018-06-23:
Make sure these boxes are checked [x] before submitting your issue - Thank you!

- [x ] Searched the problem in issues (https://github.com/arendst/Sonoff-Tasmota/issues)
- [x] Searched the problem in the wiki (https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting)
- [x] Searched the problem in the forum (https://forum.digital-pig.com/)
- [x] Searched the problem in the discussion group (https://groups.google.com/d/forum/sonoffusers)
- [X] Hardware used : Sonoff HT
- [x] Provide the output of command ``status 0`` :
```
10:22:24 MQT: stat/wohn/STATUS = {"Status":{"Module":4,"FriendlyName":["dellekom"],"Topic":"wohn","ButtonTopic":"sonoff","Power":1,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
10:22:24 MQT: stat/wohn/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Software/System restart","Uptime":"0T00:14:50","StartupUTC":"2018-06-23T09:07:34","Sleep":0,"BootCount":94,"SaveCount":298,"SaveAddress":"FA000"}}
10:22:24 MQT: stat/wohn/STATUS2 = {"StatusFWR":{"Version":"5.14.0","BuildDateTime":"2018.06.21 02:31:46","Boot":4,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
10:22:24 MQT: stat/wohn/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["wlan@vauban.de",""],"TelePeriod":300,"SetOption":["00008009","55818000"]}}
10:22:24 MQT: stat/wohn/STATUS4 = {"StatusMEM":{"ProgramSize":527,"Free":476,"Heap":19,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
10:22:24 MQT: stat/wohn/STATUS5 = {"StatusNET":{"Hostname":"wohn-5884","IPAddress":"192.168.22.174","Gateway":"192.168.22.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.22.1","Mac":"5C:CF:7F:F9:76:FC","Webserver":2,"WifiConfig":2}}
10:22:24 MQT: stat/wohn/STATUS6 = {"StatusMQT":{"MqttHost":"broker.hivemq.com","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_F976FC","MqttUser":"DVES_USER","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
10:22:24 MQT: stat/wohn/STATUS7 = {"StatusTIM":{"UTC":"Sat Jun 23 09:22:24 2018","Local":"Sat Jun 23 10:22:24 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"04:28","Sunset":"20:31"}}
10:22:25 MQT: stat/wohn/STATUS10 = {"StatusSNS":{"Time":"2018-06-23T10:22:24"}}
10:22:25 MQT: stat/wohn/STATUS11 = {"StatusSTS":{"Time":"2018-06-23T10:22:25","Uptime":"0T00:14:51","Vcc":3.178,"POWER":"ON","Wifi":{"AP":1,"SSId":"wlan@vauban.de","RSSI":82,"APMac":"[hidden]"}}}
```

Firt of all, many thanks for tasmota - really great. I managed to install it, am able to command it via web interface, temp and huminity reading works.

I have however had no success whatsoever linking the dice to test.mosquitto.org - it seems there is no way to make tasmota not use credentials. Empty credentials get reolaced by DVES_USER etc. and then test.mosquitto.org does not let me doi anything. (rc 5 means bad credentials)

With a mqtt client on my mac, I am able to publish to temp/random on test.mosquitto.org

Also no luck when using io.adafruit.com or broker.hivemind.com
My settings are:

Host: broker.hivemq.com
Port: 1883
Client: DVES_F976FC
User: DVES_USER
Password: (empty but bullets appear)
Topic: wohn
Full topic: %prefix%/%topic%/

Looks like a connection is established, log shows:

00:00:00 Projekt sonoff dellekom (Topic wohn, Fallback DVES_F976FC, GroupTopic sonoffs) Version 5.14.0-2_3_0
00:00:00 WIF: verbinden mit AP1 wlan@vauban.de in Modus 11N wie wohn-5884...
00:00:03 WIF: verbunden
00:00:03 DNS: initialisiert
00:00:03 HTP: Web-Server aktiv bei wohn-5884.local mit IP-Adresse 192.168.22.174
00:00:05 MQT: Verbindungsversuch...
00:00:05 MQT: verbunden
00:00:05 MQT: tele/wohn/LWT = online (beibehalten)
00:00:05 MQT: cmnd/wohn/POWER = 
00:00:05 MQT: tele/wohn/INFO1 = {"Module":"Sonoff TH","Version":"5.14.0","FallbackTopic":"DVES_F976FC","GroupTopic":"sonoffs"}
00:00:05 MQT: tele/wohn/INFO2 = {"WebServerMode":"Admin","Hostname":"wohn-5884","IPAddress":"192.168.22.174"}
00:00:05 MQT: tele/wohn/INFO3 = {"RestartReason":"Software/System restart"}
00:00:06 MQT: stat/wohn/RESULT = {"POWER":"ON"}
00:00:06 MQT: stat/wohn/POWER = ON
10:29:55 MQT: tele/wohn/STATE = {"Time":"2018-06-23T10:29:55","Uptime":"0T00:00:13","Vcc":3.179,"POWER":"ON","Wifi":{"AP":1,"SSId":"wlan@vauban.de","RSSI":82,"APMac":"[hidden]"}}

-- but then nothing happens

hagenbuch, 2018-06-24:
Thanks for your hint - I would expect that the built-in sensors (temp. and hum.) would try to deliver their data - or at least if I publish the command "power on" to the device url on the broker, the device could switch.

I am able to "control" the broker from a MQTT "terminal" on my mac, so I suspect the link to the device is somehow nonexistanten.

Yes, the device had been able at least to connect to the adafruit servers, but it connected every second or so, 80% of the connections were failing - but since Adafruit does not show me what they're probable missing or what they're expecting I have no clue. 

test.moquitto.org tells me less than that (rc = 5).

Frogmore42, 2018-06-24:
You need to show the log when things are not working as expected.  You can set teleperiod to a smaller value so it will send updates quicker.

hagenbuch, 2018-06-24:
I will try, I might missed that when expecting data from the sensors. But how am I able to operate the switch? I assume by sending "power on" to the address "wohn", right?

Frogmore42, 2018-06-24:
The last log indicates it is staying connected and sending results to MQTT.  Using a non-local MQTT broker (as you are doing) is not generally considered a best practice.  I have a local broker, mosquito.  I run it on a Laptop, but many people run it on a Raspberry PI.

Many people who try and use a non-local broker have difficulty.  There are also security implications that you need to consider.

hagenbuch, 2018-06-24:
I wanted to test a public broker before I go in the depths of installing an own. I am aware about the security problems but I need to communicate to the outside of my LAN anyway: Would I configure a local MQTT broker and need to make it available to the outside world, I would have to open a port AND let the world "research" into my own mosquitto while I might be less able to configure it securely than a public service. Adafruit checks username and password and if I would chose TLS I should be fine, right?

So far, I just wanted to test and could not imagine it bein so hard.

Besides setting up my own mosquitto, is there ANY way to test tasmota? I think there should be - but whch one? Did someover EVER test ist against a public server? I have to suspect not.

ascillato2, 2018-07-10:
Hi

have you managed to solve your issue?

hagenbuch, 2018-07-10:
Dear Adrian,

> have you managed to solve your issue?

Thanks for asking - but no, unfortunately.

I could not put more time into the issue since then. 
I guess I will have to dig down into the MQTT Arduino code of Tasmota.

I discovered another thing - while my Tasmota was not able to find the configured Wifi (might have to do with the router ignoring too many requests), Tasmota switched back to AP mode (which is fine) but then showed two different Wifi names: ESP_4567 and sonff-5688 - that is strange.

I will report and this looks like it will take longer until I get Tasmota really running.

Andreas

Frogmore42, 2018-08-18:
@hagenbuch based on results from other projects, I would not recommend using core 2.3.0 over the internet.  Based on this project, I would not recommend using core 2.4.0 or 2.4.1 for anything.  That leaves you with a choice.

Use a local broker with core 2.3.0.  Many people are doing that with excellent results.

Use a remote broker with core 2.4.2.  This was recently released.  I have heard generally good things about it.  It might help with the connectivity issues you are seeing.

Since you seem to just be evaluating, I would go with the local broker option that is known to work well.  Setting one up for evaluation is literally a 5 min task.

ascillato, 2018-08-28:
Hi

@hagenbuch

Have you managed to solve your issue?

hagenbuch, 2018-08-29:
Dear Adrian,

> Have you managed to solve your issue?

Unfortunately, I could not continue working on this in the meantime (also for holiday reasons..)

Thanks a lot for asking, I will update my status in the forum as soon as I get throught to it..!

Next thing: I will retry to flash a new version as suggested.

All the best
Andreas

closed 2018-09-03 by ascillato2