arendst/Tasmota #3049: Please add AM2315 - Encased I2C Temperature/Humidity Sensor

dr-apple, 2018-06-23:
Hello can add the AM2315 - Encased I2C Temperature/Humidity Sensor ?

Here the infos:

https://www.adafruit.com/product/1293

Thank you cu Danny

dr-apple, 2018-06-23:
And this CO2 Sensor  CCS811 Air Quality Sensor

Info: 

https://learn.adafruit.com/adafruit-ccs811-air-quality-sensor?view=all#arduino-wiring-test

gemu2015, 2018-06-23:
i already implemented CCS811 and made a pull request some time ago =>https://github.com/arendst/Sonoff-Tasmota/pull/2558

SHT3x is already implemented in Tasmota!

hagenbuch, 2018-06-24:
> SHT3x is already implemented in Tasmota!

Ah great - Yes, the website says

    BH1750 Ambient Light Intensity
    BMP180 Pressure and Temperature in group 3a
    BMP280 Pressure and Temperature in group 3a
    BME280 Pressure, Temperature and Humidity in group 3a
    HTU21 Temperature and Humidity in group 3b
    SI7013 Temperature and Humidity in group 3b
    SI7020 Temperature and Humidity in group 3b
    SI7021 Temperature and Humidity in group 3b
    SHT10 Temperature and Humidity in group 3c
    SHT11 Temperature and Humidity in group 3c
    SHT15 Temperature and Humidity in group 3c

.. but I assume this is a hint for the compilation of the code, so far I don't find something like "group 3c" selectable in the dropdown menu.

gemu2015, 2018-06-24:
you must assure that in file sonoff_post.h

these both defines are activ

#define USE_I2C                               // I2C using library wire (+10k code, 0k2 mem, 124 iram)
#define USE_SHT3X 

then compile and flash

then in dropdown select I2C  SCL and SDA on 2 pins you connected the sensor

Crumpy10, 2018-07-05:
Hi,
Can I second the request for AM2315 integration, its a nice little unit that suits well to home automation projects.
Thanks for your time.

closed 2018-09-03 by stale[bot]

JohanViljoen, 2022-06-29:
And here we are, almost four year later, and I, too, have an AM2315 which I would like to read using Tasmota.
Any suggestions? Would one of the existing units already catered for be suitable, or do we need fresh code?

sfromis, 2022-06-29:
Tasmota already supports many good temp/hum sensors, anything special about the AM2315 to select that one?

If you already have it, you could try your luck with the existing support for other AM/AHT sensors.
https://tasmota.github.io/docs/AM2301/
https://tasmota.github.io/docs/AHT1x/