arendst/Tasmota #3050: Max distance for External Button

tenkaibr, 2018-06-23:
I am trying to use an external button and would like to know if there is any limit to install the button. What is the maximum distance? Thank you

ascillato, 2018-06-23:
Hi,

The more length on your wires from your sonoff to your switch will make more unintended turn off and on.

You should keep them as short as possible.

There is no a maximum lenght specified due to the electrical noise in your home will be different than mine. If your wires are next to Mains wires, it will have more electrical noise than if it is alone. If your wires make a spin or a curve, will receive more noise than straight wires. etc. etc.

So the advise is to be as short as possible and you have to test if works.

There are plenty of youtube videos on how to connect an external switch to a sonoff.

Hope this helps.

If this answer your question, please close this issue. thanks.

tenkaibr, 2018-06-23:
Thank you for tour reply!

closed 2018-06-23 by tenkaibr