arendst/Tasmota #3052: Cant get Philips Hue emulation working

Giorgiofox, 2018-06-23:
Hello,
I'm using different esp8266 with both 4.16 and 6.0b Tasmota firmware, and I can get the emulation working: when I start the app on my iPhone it found the hub but I can't create room: it simply ask me again to create a room. 
I've tried also on an android phone, and in that case it say that the hub is "full"

In the configuration I have only WS2812

What I can do?

thanks in advance
Giorgio

Jason2866, 2018-06-23:
Take a look here
https://github.com/arendst/Sonoff-Tasmota/wiki/Theo's-Tasmota-Tips

ascillato2, 2018-07-10:
Hi,

Seems that your question has been addressed. Closing this issue. Please, reopen if needed. Thanks.

closed 2018-07-10 by ascillato2

martin072, 2019-01-31:
Hi, I am experiencing the same issue with the latest release. I don't really understand the link. It refers to a pre-compiled version..I like to compile my own. Is there a fix for this?

Jason2866, 2019-01-31:
@martin072 
Since a while nothing needs to be patched anymore to get a working Alexa with core 2.3.0, core 2.4.2 and core 2.5.0 beta3.  Just download latest development version compile. Enjoy.
Dont use release version. This change is not in there. Change was done after release

martin072, 2019-01-31:
I am running it on a Sonoff S20 socket with the following build versions:

Program Version - 6.4.1.4(sonoff)
Core/SDK version 2_4_2/2.2.1 (cfd48f3)

So I believe I am running all the latest versions, jet as mentioned above when trying to add the hub in de Hue app it asks to add a room which doesn't do anything. The app won't allow you to continue.

Jason2866, 2019-01-31:
Nope, it is too old! Take a look at Tasmota Github.

martin072, 2019-01-31:
Ok, will give it a go. thanks

martin072, 2019-01-31:
Fortunately, getting a lot of error's if I compile the development version

```
/var/folders/x7/r7bv5q3x04s4t572lfr8qjtr0000gn/T/arduino_build_89000/libraries/ESP8266WebServer/Parsing.cpp.o: In function `ESP8266WebServer::_uploadReadByte(WiFiClient&)':
/Users/martin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WebServer/src/ESP8266WebServer.h:160: multiple definition of `ESP8266WebServer::_collectHeader(char const*, char const*)'
/var/folders/x7/r7bv5q3x04s4t572lfr8qjtr0000gn/T/arduino_build_89000/sketch/Parsing.cpp.o:/var/folders/x7/r7bv5q3x04s4t572lfr8qjtr0000gn/T/arduino_build_89000/sketch/Parsing.cpp:268: first defined here
/var/folders/x7/r7bv5q3x04s4t572lfr8qjtr0000gn/T/arduino_build_89000/libraries/ESP8266WebServer/Parsing.cpp.o: In function `ESP8266WebServer::_parseRequest(WiFiClient&)':
/Users/martin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WebServer/src/Parsing.cpp:348: multiple definition of `ESP8266WebServer::_uploadWriteByte(unsigned char)'
/var/folders/x7/r7bv5q3x04s4t572lfr8qjtr0000gn/T/arduino_build_89000/sketch/Parsing.cpp.o:/var/folders/x7/r7bv5q3x04s4t572lfr8qjtr0000gn/T/arduino_build_89000/sketch/Parsing.cpp:353: first defined here
/var/folders/x7/r7bv5q3x04s4t572lfr8qjtr0000gn/T/arduino_build_89000/libraries/ESP8266WebServer/Parsing.cpp.o: In function `ESP8266WebServer::_parseRequest(WiFiClient&)':
/Users/martin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WebServer/src/Parsing.cpp:359: multiple definition of `ESP8266WebServer::_uploadReadByte(WiFiClient&)'
/var/folders/x7/r7bv5q3x04s4t572lfr8qjtr0000gn/T/arduino_build_89000/sketch/Parsing.cpp.o:/var/folders/x7/r7bv5q3x04s4t572lfr8qjtr0000gn/T/arduino_build_89000/sketch/Parsing.cpp:364: first defined here
/var/folders/x7/r7bv5q3x04s4t572lfr8qjtr0000gn/T/arduino_build_89000/libraries/ESP8266WebServer/Parsing.cpp.o: In function `ESP8266WebServer::urlDecode(String const&)':
/Users/martin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WebServer/src/Parsing.cpp:581: multiple definition of `ESP8266WebServer::urlDecode(String const&)'
/var/folders/x7/r7bv5q3x04s4t572lfr8qjtr0000gn/T/arduino_build_89000/sketch/Parsing.cpp.o:/var/folders/x7/r7bv5q3x04s4t572lfr8qjtr0000gn/T/arduino_build_89000/sketch/Parsing.cpp:586: first defined here
/var/folders/x7/r7bv5q3x04s4t572lfr8qjtr0000gn/T/arduino_build_89000/libraries/ESP8266WebServer/Parsing.cpp.o: In function `ESP8266WebServer::_parseArguments(String)':
/Users/martin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WebServer/src/Parsing.cpp:273: multiple definition of `ESP8266WebServer::_parseArguments(String)'
/var/folders/x7/r7bv5q3x04s4t572lfr8qjtr0000gn/T/arduino_build_89000/sketch/Parsing.cpp.o:/var/folders/x7/r7bv5q3x04s4t572lfr8qjtr0000gn/T/arduino_build_89000/sketch/Parsing.cpp:278: first defined here
/var/folders/x7/r7bv5q3x04s4t572lfr8qjtr0000gn/T/arduino_build_89000/libraries/ESP8266WebServer/Parsing.cpp.o: In function `ESP8266WebServer::_parseFormUploadAborted()':
/Users/martin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WebServer/src/Parsing.cpp:611: multiple definition of `ESP8266WebServer::_parseFormUploadAborted()'
/var/folders/x7/r7bv5q3x04s4t572lfr8qjtr0000gn/T/arduino_build_89000/sketch/Parsing.cpp.o:/var/folders/x7/r7bv5q3x04s4t572lfr8qjtr0000gn/T/arduino_build_89000/sketch/Parsing.cpp:616: first defined here
/var/folders/x7/r7bv5q3x04s4t572lfr8qjtr0000gn/T/arduino_build_89000/libraries/ESP8266WebServer/Parsing.cpp.o: In function `ESP8266WebServer::_parseForm(WiFiClient&, String, unsigned int)':
/Users/martin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WebServer/src/Parsing.cpp:368: multiple definition of `ESP8266WebServer::_parseForm(WiFiClient&, String, unsigned int)'
/var/folders/x7/r7bv5q3x04s4t572lfr8qjtr0000gn/T/arduino_build_89000/sketch/Parsing.cpp.o:/var/folders/x7/r7bv5q3x04s4t572lfr8qjtr0000gn/T/arduino_build_89000/sketch/Parsing.cpp:373: first defined here
/var/folders/x7/r7bv5q3x04s4t572lfr8qjtr0000gn/T/arduino_build_89000/libraries/ESP8266WebServer/Parsing.cpp.o: In function `ESP8266WebServer::_parseRequest(WiFiClient&)':
/Users/martin/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WebServer/src/Parsing.cpp:70: multiple definition of `ESP8266WebServer::_parseRequest(WiFiClient&)'
/var/folders/x7/r7bv5q3x04s4t572lfr8qjtr0000gn/T/arduino_build_89000/sketch/Parsing.cpp.o:/var/folders/x7/r7bv5q3x04s4t572lfr8qjtr0000gn/T/arduino_build_89000/sketch/Parsing.cpp:75: first defined here
collect2: error: ld returned 1 exit status
```

martin072, 2019-01-31:
Ok. So that was when compiling with board version 2.4.2, I 've upgraded to 2.5.0 beta 3 and it compiles Ok, But... no joy.. still not able to add the bridge in the Hue app.. same issue.

Current versions
Program Version 6.4.1.13(sonoff)
Core/SDK Version 2_5_0_BETA3/3.0.0-dev(c0f7b44)

Jason2866, 2019-02-01:
Emulation enabled? Friendly Name?
[image]

martin072, 2019-02-01:
Yes all the same as in your picture.  Ah, so there is the answer then.. I am trying to add it in the Hue App..

rmoriz, 2019-03-18:
- Amazon Alexa iOS app won't recognize the "new bridge"
- Hue iOS apps (trie v1 and v2) detect the "new bridge" but won't let me create a new room. Therefore I cannot add the device.

```
Program Version | 6.4.1.22(sonoff)
2019-03-18T21:08:39
2_5_0/3.0.0-dev(c0f7b44)
```

altelch, 2019-03-19:
The HUE emulation isn't a full bridge emulation. It's just the minimal function set to get an Amazon Echo Device recognize it as a bridge and work with it. There is no full support for Rooms/Groups (Hue Apps) or Alexa Apps which require a cloud connected bridge.

Greetings,
    Altelch.

rmoriz, 2019-03-26:
I can confirm after purchasing and installing an Amazon Echo Dot Gen 3 it recognizes the Tasmota running device.